introlab/rtabmap_ros #726: MapData colored point cloud floats in mid air

zkytony, 2022-03-01:
As shown in the image below, the occupancy grid map is well below the colored point cloud in MapData.

[image]

Why would this happen and how to fix it? Let me know what I need to provide. Thanks!

matlabbe, 2022-03-07:
The occupancy grid would be at the same height than the `frame_id` set for rtabmap node. For the point cloud, make sure you have an optical rotation with your camera in your TF tree, otherwise the cloud would appear showing towards the sky (not sure if it is what we are seeing here).

zkytony, 2022-03-07:
I have figured out this issue. First off the grid map shown in this screenshot isn't correct (it was fixed after I set `subscribe_scan` to be false in the rtabmap node, and also decrease the `range_max` parameter to "2" for the `depthimage_to_laserscan` node.
The floating in mid-air is most likely due to odometry drift (but I can't be 100% sure yet). The good thing is after I could build a grid map together with the point cloud, I could use it to localize the robot using rtabmap as well. I learned from [here](https://answers.ros.org/question/302694/rtabmap-localization/?answer=302712#post-id-302712) that relocalization occurs at loop closure works (I don't think this is documented at the ROS wiki page http://wiki.ros.org/rtabmap_ros).

closed 2022-03-07 by zkytony