introlab/rtabmap_ros #729: Warn: Did not receive data since 5 seconds!

UdayRockzz, 2022-03-08:
Hello Everyone, 

I am working on pepper robot with Rtabmap.  pepper robot has an xtion 3D camera and two RGB cameras. I have written the code as below.  

`<?xml version="1.0" encoding="UTF-8"?>

<launch>
  
  
  
   
  
  <arg name="database_path"      default="rtabmap.db"/>
  <arg name="rgb_topic"  	 default="/pepper_robot/camera/front/image_raw"/> 
  <arg name="depth_topic"	 default="/pepper_robot/camera/depth/image_raw"/>
  <arg name="camera_info_topic"  default="/pepper_robot/camera/front/camera_info"/>  
  <arg name="scan"		 default="/pepper_robot/laser"/>      

  
  <group ns="rtabmap"> 

    <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_ros/rgbd_sync" output="screen"> 
          
      <remap from="scan"		 to="$(arg scan)"/>
      <remap from="rgb/image"		 to="$(arg rgb_topic)"/>
      <remap from="rgb/camera_info"	 to="$(arg camera_info_topic)"/>
      <remap from="depth/image" 	 to="$(arg depth_topic)"/>	
      <remap from="rgbd_image"    	 to="rgbd_image"/> 
     

      
        <param name="approx_sync"       value="true"/>
    </node>

    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen"  args="--delete_db_on_start">

      
      <param name="database_path"      		  type="string" value="$(arg database_path)"/>
      <param name="frame_id"            	  type="string" value="base_link"/>
      <param name="odom_frame_id"     		  type="string" value="odom"/>
      <param name="subscribe_depth"   		  type="bool"   value="false"/>
      <param name="subscribe_scan"      	  type="bool"   value="false"/>

      <param name="RGBD/AngularUpdate"            type="string" value="0.05"/>
      <param name="RGBD/LinearUpdate"             type="string" value="0.05"/>
      <param name="RGBD/OptimizeFromGraphEnd"     type="string" value="false"/>
    
      
      
      <param name="Rtabmap/ImagesAlreadyRectified" value="true" />
      <param name="queue_size" 			  type="int" value="10"/>

      
      <param name="Icp/CorrespondenceRatio"    	  type="string" value="0.3"/>
                
    
      
      <remap from="grid_map" to="/map"/>

      
      <param name="Rtabmap/DetectionRate" 	 type="string" value="1"/>

      
      <param name="Reg/Force3DoF" 		 type="string" value="true"/>
	

      
      
      <param name="Kp/DetectorStrategy" type="string" value="0"/>

      
      <param name="Kp/MaxFeatures" type="string" value="400"/>

      
      <param name="SURF/HessianThreshold" type="string" value="100"/>

      
      
      <param name="Reg/Strategy" type="string" value="0"/>

      
      <param name="Vis/MinInliers" type="string" value="15"/>

      
      <param name="Mem/IncrementalMemory" type="string" value="false"/>
    

	
        <param name="Grid/FromDepth"                type="string" value="false"/>
        <param name="GridGlobal/Eroded"             value="true" />
        <param name="GridGlobal/FootprintHeight"    value="1.6" />
        <param name="GridGlobal/FootprintLength"    value="0.6" />
        <param name="GridGlobal/FootprintWidth"     value="0.6" />
        <param name="GridGlobal/MinSize"            value="20.0" />
        <param name="Grid/MaxObstacleHeight"        value="1.4" />
        <param name="Grid/MinClusterSize"           value="15" />
        <param name="Grid/RangeMax"                 value="4.0" />
        <param name="Grid/RayTracing"               value="true" />
        <param name="Grid/NoiseFilteringRadius"     value="0.05"/>
        <param name="Grid/NoiseFilteringMinNeighbors" value="2"/>
        
        <param name="Grid/ClusterRadius"            value="0.5" />
        <param name="Grid/CellSize"                 value="0.05" />
    </node>
    

	   
    <node pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
        <param name="subscribe_depth"             type="bool" value="false"/>
        <param name="subscribe_scan"              type="bool" value="false"/>
        <param name="frame_id"                    type="string" value="base_link"/>
        <param name="queue_size" 	          type="int" value="10"/>

        <remap from="rgb/image"       to="$(arg rgb_topic)"/>
        <remap from="depth/image"     to="$(arg depth_topic)"/>
        <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
     	<remap from="scan"            to="$(arg scan)"/>
    </node>     
  
  </group>
<node pkg="rviz" type="rviz" name="pepper_rviz" args="-d $(find my_robot)/rviz/pepper_rtabmap.rviz"/>
</launch>` 
when I run the code I am getting warning/error as below. 
[image]

I have followed issue 189 but didn't succeed for me. Could anyone tell me where I am doing wrong in  @this?

awesomebytes, 2022-03-09:
Hello @UdayRockzz 

I'd start by checking things like:
```bash
rosnode info /rtabmap
# check what topics is rtabmap subscribing to
```

If the topics you expect to have are there (e.g. rgb and depth camera), check that they have a camera_info topic associated to it. Without it, I think it won't work (or so I recall).

I do not know where the source of your ROS images is (e.g. running your robot? running from a bridge in your computer?), but if it is in the robot, you may need to synchronize the clock of your system and the robot's system to avoid timestamp issues.

I also would recommend you to start with a rosbag with the topics you need from the robot, and then try to run rtabmap with that rosbag until you get something working.

Additionally, having a rosbag, you can post it here (or probably in ROS answers, which I'm pretty sure would be a better place to post this question) so people can help you better. Without data, it's hard to help.

Good luck!

UdayRockzz, 2022-03-09:
Thanks for the reply @awesomebytes. Iam Using [ nao python based SDK](https://github.com/ros-naoqi/pepper_robot) to connect bridge between ros and pepper robot. and then these are the topics I have used. 
[image] it was there in the code which I have sent to you in email. but I dont know how those are missed here. and I will start working with rosbags.

awesomebytes, 2022-03-09:
Showing the launchfile is not the same than checking what the node is subscribed to

UdayRockzz, 2022-03-10:
well, I got it. this is rqt_graph [image] and below are the topics list.

/clicked_point
/cmd_vel
/diagnostics
/initialpose
/joint_angles
/joint_states
/map
/move_base_simple/goal
/pepper_robot/audio
/pepper_robot/bumper
/pepper_robot/camera/bottom/camera_info
/pepper_robot/camera/bottom/image_raw
/pepper_robot/camera/bottom/image_raw/compressed
/pepper_robot/camera/bottom/image_raw/compressed/parameter_descriptions
/pepper_robot/camera/bottom/image_raw/compressed/parameter_updates
/pepper_robot/camera/bottom/image_raw/theora
/pepper_robot/camera/bottom/image_raw/theora/parameter_descriptions
/pepper_robot/camera/bottom/image_raw/theora/parameter_updates
/pepper_robot/camera/bottom/image_rect_color
/pepper_robot/camera/bottom/image_rect_color/compressed
/pepper_robot/camera/bottom/image_rect_color/compressed/parameter_descriptions
/pepper_robot/camera/bottom/image_rect_color/compressed/parameter_updates
/pepper_robot/camera/bottom/image_rect_color/compressedDepth
/pepper_robot/camera/bottom/image_rect_color/compressedDepth/parameter_descriptions
/pepper_robot/camera/bottom/image_rect_color/compressedDepth/parameter_updates
/pepper_robot/camera/bottom/image_rect_color/theora
/pepper_robot/camera/bottom/image_rect_color/theora/parameter_descriptions
/pepper_robot/camera/bottom/image_rect_color/theora/parameter_updates
/pepper_robot/camera/bottom_rectify_color/parameter_descriptions
/pepper_robot/camera/bottom_rectify_color/parameter_updates
/pepper_robot/camera/depth/camera_info
/pepper_robot/camera/depth/image
/pepper_robot/camera/depth/image/compressed
/pepper_robot/camera/depth/image/compressed/parameter_descriptions
/pepper_robot/camera/depth/image/compressed/parameter_updates
/pepper_robot/camera/depth/image/compressedDepth
/pepper_robot/camera/depth/image/compressedDepth/parameter_descriptions
/pepper_robot/camera/depth/image/compressedDepth/parameter_updates
/pepper_robot/camera/depth/image/theora
/pepper_robot/camera/depth/image/theora/parameter_descriptions
/pepper_robot/camera/depth/image/theora/parameter_updates
/pepper_robot/camera/depth/image_raw
/pepper_robot/camera/depth/image_raw/compressed
/pepper_robot/camera/depth/image_raw/compressed/parameter_descriptions
/pepper_robot/camera/depth/image_raw/compressed/parameter_updates
/pepper_robot/camera/depth/image_raw/compressedDepth
/pepper_robot/camera/depth/image_raw/compressedDepth/parameter_descriptions
/pepper_robot/camera/depth/image_raw/compressedDepth/parameter_updates
/pepper_robot/camera/depth/image_raw/theora
/pepper_robot/camera/depth/image_raw/theora/parameter_descriptions
/pepper_robot/camera/depth/image_raw/theora/parameter_updates
/pepper_robot/camera/depth/image_rect
/pepper_robot/camera/depth/image_rect/compressed
/pepper_robot/camera/depth/image_rect/compressed/parameter_descriptions
/pepper_robot/camera/depth/image_rect/compressed/parameter_updates
/pepper_robot/camera/depth/image_rect/compressedDepth
/pepper_robot/camera/depth/image_rect/compressedDepth/parameter_descriptions
/pepper_robot/camera/depth/image_rect/compressedDepth/parameter_updates
/pepper_robot/camera/depth/image_rect/theora
/pepper_robot/camera/depth/image_rect/theora/parameter_descriptions
/pepper_robot/camera/depth/image_rect/theora/parameter_updates
/pepper_robot/camera/depth/image_rect_raw
/pepper_robot/camera/depth/image_rect_raw/compressed
/pepper_robot/camera/depth/image_rect_raw/compressed/parameter_descriptions
/pepper_robot/camera/depth/image_rect_raw/compressed/parameter_updates
/pepper_robot/camera/depth/image_rect_raw/compressedDepth
/pepper_robot/camera/depth/image_rect_raw/compressedDepth/parameter_descriptions
/pepper_robot/camera/depth/image_rect_raw/compressedDepth/parameter_updates
[... 145 lines cut ...]
/rtabmap/octomap_ground
/rtabmap/octomap_obstacles
/rtabmap/octomap_occupied_space
/rtabmap/odom
/rtabmap/proj_map
/rtabmap/rgb/camera_info
/rtabmap/rgb/image
/rtabmap/rgbd_image
/rtabmap/rgbd_image/compressed
/rtabmap/scan_map
/rtabmap/user_data_async
/speech
/statistics
/tf
/tf_static

I checked with rosbag for all the topics.  here are the recorded messages

[image]
[image]
[image]

also, Here are the results when I run `rostopic hz ` for both front camera and depth topics. 
[image] and finally. rviz image
[image]

awesomebytes, 2022-03-10:
You still haven't shared the rosnode info command.

UdayRockzz, 2022-03-14:
These are the list of nodes after launching the launch file,.
[image].

so I have checked nodes with /rtabmap/rtabmap. and here is the list for that.
[image]
[image]
[image]

and here is the node info for /rtabmap/rgbd_sync
[image]

awesomebytes, 2022-03-14:
@UdayRockzz you can see that rtabmap is not subscribed to any existing image topic. You'll need to remap that correctly (using or not using rgbd sync, i don't think you need it, but up to you). Search in the ROS wiki how to remap topics on launchfiles.
Also, you seem to want to use the front (top) camera but forwarding the bottom camera camera_info, that also looks wrong to me.

If you try to follow the errors and carefully check the input and outputs of the nodes you should be able to make it work. You can also check the source code of rtabmap if you still don't know what's going on, a good starting point is to search for the prints of errors/warnings you are seeing.

This is probably your first time using ROS. I would recommend you to take a look again at the ROS tutorials, try to use some tutorial with a simulated robot, and maybe borrow a friend to walk you through how to debug some ROS node.

Good luck! I hope you make good progress soon!

UdayRockzz, 2022-03-18:
Thanks alot for your inputs @awesomebytes. I can slove this issue with correct remaps. And I am testing with bottom camera, so I used bottom camera at that time. and I have another question. Did you use laser sensor for generating map with rtabmap? if so, have you faced any [similar kind ](https://answers.ros.org/question/397634/occupancygridcpp380createlocalmap-cannot-create-local-map-scan-is-empty-node326-gridsensor0/)of issues? and the laser scan works fine when I use with slam_gmapping, but not working with rtabmap.

awesomebytes, 2022-03-18:
I did not use laser scans, I used just rgb + depth/pointcloud.

matlabbe, 2022-04-10:
For RGB-D + Lidar, see http://wiki.ros.org/rtabmap_ros/Tutorials/SetupOnYourRobot#Kinect_.2B-_Odometry_.2B-_2D_laser

hariharan382, 2022-07-19:
I am having perfect input connection,I could see the echo but it throws me error ,time difference between RGB and depth less than 0.01 ,which I have no idea about it ,since rgbdsync will combine and publish at constant rate ,my configuration is d435+laser+odom

matlabbe, 2022-10-10:
> it throws me error
What error?