introlab/rtabmap_ros #732: Use TUM rgbd dataset evaluate rtabmap_ros

Wei0703, 2022-03-16:
Excuse me,

I want to use TUM dataset to evaluate absolute trajectory error (ATE) metric.
I use `$ rtabmap-report --poses rtabmap.db` command to get rtabmap_odom.txt and rtabmap_slam.txt.
And use [TUM dataset tools](https://svncvpr.in.tum.de/cvpr-ros-pkg/trunk/rgbd_benchmark/rgbd_benchmark_tools/src/rgbd_benchmark_tools/) to get ATE result, command like this `$ python evaluate_ate.py groundtruth.txt rtabmap.txt`.
Following picture are results, but I think the results are wrong.
[image]
groundtruth.txt download from TUM rgbd dataset fr3_walking_xyz sequence.
Am I doing something wrong?

matlabbe, 2022-03-17:
Can you run the evaluation tool with "--plot figure.png"? If you want more poses to compare, launch rtabmap node with parameter `Rtabmap/CreateIntermediateNodes` at `true`.

If you test under ros, see also this post using [rgbdslam_datasets.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/rgbdslam_datasets.launch): http://official-rtab-map-forum.206.s1.nabble.com/Test-rtabmap-RGBD-mapping-using-TUM-RGBD-database-tp1095p1099.html

Wei0703, 2022-03-17:
Yes, following picture is my evaluation result.

[image]

I try to use **rgbdslam_datasets.launch**, but loop closure detection window can't display anything.
How do I export more poses to compare.
My rtabmapviz screen like this.

[image]

Thanks a lot!

matlabbe, 2022-04-10:
Which dataset are you testing? I could give a try. It looks like the [rgbd_dataset_freiburg3_sitting_xyz.bag](https://vision.in.tum.de/data/datasets/rgbd-dataset/download#freiburg3_sitting_xyz) dataset. Following those [instructions](https://github.com/introlab/rtabmap_ros/blob/3a025d9e758985a7b79d52312f03bd5d1fb73732/launch/tests/rgbdslam_datasets.launch#L4-L11), I got this:
[image]

However, looking at the database, we don't have all poses in the map (gray is ground truth):
[image]

It is because of `RGBD/LinearUpdate` parameter that is 10 cm by default (rtabmap won't add new nodes if the camera didn't move more than 10 cm).  I [updated](https://github.com/introlab/rtabmap_ros/commit/88ca603b99c885dd898804e370d99131a343b97d) the launch file to keep everything. Here the new results:
[image]

If you want to export all poses from rtabmapviz, make sure to click Pause, then you have to click on Edit->Download Graph and select Global Optimized. You can then do File->Export poses...

Another easy way to get poses and ground truth (after closing launch):
```
$ rtabmap-report --poses ~/.ros/rtabmap.db 

Database: /home/mathieu/.ros/rtabmap.db
   rtabmap.db (363, s=1.000):	error lin=0.019m (max=0.027m, odom=0.019m) ang=0.7deg, slam: avg=36ms (max=76ms) loops=0, odom: avg=0ms (max=0ms), camera: avg=0ms, map=-1MB
```

It is known that camera calibration or depth scale is a little off with this dataset, to automatically scale the poses:
```
$ rtabmap-report --poses --scale ~/.ros/rtabmap.db 

Database: /home/mathieu/.ros/rtabmap.db
   rtabmap.db (363, s=1.007):	error lin=0.018m (max=0.027m, odom=0.018m) ang=0.7deg, slam: avg=36ms (max=76ms) loops=0, odom: avg=0ms (max=0ms), camera: avg=0ms, map=-1MB
```
[rtabmap_gt.txt](https://github.com/introlab/rtabmap_ros/files/8459730/rtabmap_gt.txt)
[rtabmap_slam.txt](https://github.com/introlab/rtabmap_ros/files/8459731/rtabmap_slam.txt)

With evaluation script:
```
$ python evaluate_ate.py --plot figure.png ~/.ros/rtabmap_gt.txt ~/.ros/rtabmap_slam.txt 
0.018371
```
[image]

SYLJSTK, 2023-02-27:
$ rtabmap-report --poses --scale ~/.ros/rtabmap.db 

Database: /home/mathieu/.ros/rtabmap.db
   rtabmap.db (363, s=1.007):	error lin=0.018m (max=0.027m, odom=0.018m) ang=0.7deg, slam: avg=36ms (max=76ms) loops=0, odom: avg=0ms (max=0ms), camera: avg=0ms, map=-1MB

When I run this command, I can't find rtabmap_gt.txt ，How can i get it？

matlabbe, 2023-04-15:
The path is defined here: https://github.com/introlab/rtabmap/blob/0f221ba3cd9913b68ec508af0d914e3f49860430/tools/Report/main.cpp#L934

It should be exported under `~/.ros`