introlab/rtabmap_ros #734: No occupancy map and no features visualized

abylikhsanov, 2022-03-25:
I am using a stereo camera + 2D LiDAR and when running a launch file like this:

```
<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="delete_db_on_start">
      <param name="frame_id" type="string" value="base_footprint"/>
	
      <param name="subscribe_rgb"   type="bool" value="false"/>
      <param name="subscribe_depth" type="bool" value="false"/>
      <param name="subscribe_rgbd"  type="bool" value="false"/>
      <param name="subscribe_stereo"  type="bool" value="true"/>
      <param name="subscribe_scan"  type="bool" value="true"/>
	
      <remap from="scan" to="/scan_front_filtered"/>
      <remap from="left/image_rect" to="/camera/infra1/image_rect_raw" />
      <remap from="right/image_rect" to="/camera/infra2/image_rect_raw" />
      <remap from="left/camera_info" to="/camera/infra1/camera_info" />
      <remap from="right/camera_info" to="/camera/infra2/camera_info" />

      <remap from="odom" to="/scanmatch_odom"/>
      <param name="subscribe_odom_info" type="bool" value="true"/>
      <param name="approx_sync " value="true" />
	

      <param name="Reg/Strategy"       type="string" value="1"/> 
      <param name="Reg/Force3DoF"      type="string" value="true"/>
      <param name="Icp/CorrespondenceRatio"  type="string" value="0.01"/>
      <param name="Icp/VoxelSize"            type="string" value="0.05"/>
      <param name="Grid/Sensor"            type="string" value="false"/>
    </node>
```

I do not see any feautre matching and no occupancy grid created even though the Grid/Sensor is set to False.

What could be wrong?

abylikhsanov, 2022-03-27:
Solved after investigating the setup more thoroughly, need RGBD

closed 2022-03-27 by abylikhsanov

matlabbe, 2022-04-11:
To combine stereo with lidar, you should use rgbd_image interface. Add a stereo_sync node to synchronize your left/right images and camera info, then set `subscribe_rgbd:=true` for rtabmap node.