introlab/rtabmap_ros #736: Cannot use Reg/Strategy=1 due to variance problems

abylikhsanov, 2022-03-28:
Hi,

I have created a map using Reg/Strategy = 1 and it worked fine. After saving and reloading the databaes in localization mode, suddenly I have faced a problem with ICP registration in rtabmap which could not accept loop closure due to variance problems:

```
[ WARN] (2022-03-28 11:42:30.957) RegistrationIcp.cpp:904::computeTransformationImpl() Transform is found (xyz=-0.017121,-0.023627,0.000000 rpy=0.000000,-0.000000,0.006848) but no correspondences has been found!? Variance is unknown!
[ WARN] (2022-03-28 11:42:30.957) Rtabmap.cpp:2788::process() Rejected loop closure 142 -> 635: Cannot compute transform (cor=0 corrRatio=0.000000/0.010000 maxLaserScans=856)
```

If I set the Strategy to 0 obviously it works meaning that the problem lies in the ICP registration. What could be the reason? I have tried both g2o and GTSAM optimizers and both give this warning and covariance in my odometry is just 0.01 in diagonal and I used the same odometry to map the area and it did not complain.

matlabbe, 2022-04-11:
This issue is not related to g2o ot gtsam. It is indeed strange that it worked in slam but not localization. If you move around, will it localize at some point? What are the ICP parameters used?
```
rtabmap-info ~/.ros/rtabmap.db
```

matlabbe, 2022-05-09:
Without any data to look at, it is difficult to see what is the problem.

To completely disable proximity detection with laser scan, you should also explicitly set `RGBD/ProximityPathMaxNeighbors` to 0.

abylikhsanov, 2022-05-24:
@matlabbe How can I reach you via email? The data is not public and I would like to send the link via email

matlabbe, 2022-09-06:
Maybe too late for this issue, but my email is same username than here, at gmail.com

matlabbe, 2022-09-06:
Hi,

sorry being so late, started a new job this summer and moved country... In the shared folder there is no map_6.db. I tried map.db in localization mode with rosbag in second/filtered and it seems fine. However, I had to reprocess the database because my computer doesn't use opencv with all features available. Maybe the difference between map.db and map_6.db is that map_6.db has been created on a different computer, thus rtabmap is maybe be compiled with slightly different version of dependencies that could cause not same parameters used by default.

For example, if I do `rtabmap-info --diff map.db`:

```
rtabmap-info --diff map.db 
Parameters (Yellow=modified, Red=old parameter not used anymore, NA=not in database):
Grid/RangeMax=                     0  (default=5.0)
Grid/RayTracing=                   true  (default=false)
Grid/Sensor=                       false  (default=1)
Icp/CorrespondenceRatio=           0.01  (default=0.1)
Icp/MaxCorrespondenceDistance=     0.05  (default=0.1)
Icp/PointToPlane=                  false  (default=true)
Icp/Strategy=                      0  (default=1)
Kp/DetectorStrategy=               6  (default=8)
Optimizer/Epsilon=                 0.0  (default=0.00001)
Optimizer/Strategy=                1  (default=2)
RGBD/AngularUpdate=                0.01  (default=0.1)
RGBD/CreateOccupancyGrid=          true  (default=false)
RGBD/LinearUpdate=                 0.01  (default=0.1)
RGBD/ProximityPathMaxNeighbors=    10  (default=0)
Reg/Force3DoF=                     true  (default=false)
Reg/Strategy=                      1  (default=0)
Vis/FeatureType=                   6  (default=8)
```

On my computer, I don't have access to GFTT/BRIEF feature (6), so I had to reprocess with GFTT/ORB (8) the database to make visual loop closure detection working. In term of localization accuracy, I put back `Icp/Strategy` to 1 (libpointmatcher instead PCL), enabled `Icp/PointToPlane` and set `Icp/MaxCorrespondenceRatio` to 0.5 (the scans were then matching better on re-localization).

I would make sure that rtabmap on the robot is the same version (with same dependencies) than your desktop computer that created map_6.db. You can do `rtabmap --version` and compare.

PS: I put the answer here instead of replying private email as it is general and maybe too late to be useful for you, but for someone else it can be useful to know.

abylikhsanov, 2022-09-06:
@matlabbe Hi thanks.

I reran the rtabmap reprocess and this is what I ended up:

```
Db/TargetVersion=                  0.20.19  (default=)
Grid/3D=                           true  (default=false)
Grid/RangeMax=                     0  (default=5.0)
Grid/RayTracing=                   true  (default=false)
Grid/Sensor=                       false  (default=1)
Icp/CorrespondenceRatio=           0.01  (default=0.1)
Icp/MaxCorrespondenceDistance=     0.05  (default=0.1)
RGBD/AngularUpdate=                0.01  (default=0.1)
RGBD/CreateOccupancyGrid=          true  (default=false)
RGBD/LinearUpdate=                 0.01  (default=0.1)
RGBD/ProximityPathMaxNeighbors=    10  (default=0)
Reg/Force3DoF=                     true  (default=false)
Reg/Strategy=                      1  (default=0)
```

Then I have used the following launch file to try and localize:

```
        <param name="Mem/CovOffDiagIgnored" type="string" value="false" />
        <param name="RGBD/NeighborLinkRefining" type="string" value="false"/>
        <param name="RGBD/ProximityBySpace"     type="string" value="true"/>
        <param name="RGBD/AngularUpdate"        type="string" value="0.01"/>
        <param name="RGBD/LinearUpdate"         type="string" value="0.01"/>
        <param name="RGBD/OptimizeFromGraphEnd" type="string" value="false"/>
        <param name="Reg/Strategy"       type="string" value="1"/>
        <param name="Reg/Force3DoF"      type="string" value="true"/>
        <param name="Icp/CorrespondenceRatio"  type="string" value="0.01"/>
        <param name="Grid/Sensor"            type="string" value="false"/>
        <param name="Grid/RayTracing" value="true"/>
        <param name="Mem/IncrementalMemory" type="string" value="false"/>
        <param name="RGBD/OptimizeMaxError"                type="double"      value="3"  />
        <param name="Optimizer/Strategy" value="1" />
        <param name="Mem/UseOdomGravity" value="false" />
        <param name="Rtabmap/PublishRAMUsage" value="true" />
        <param name="Kp/DetectorStrategy" value="6" />

        <param name="Kp/MaxFeatures" value="1500" />
        <param name="Rtabmap/DetectionRate" value="1.0" />
        <param name="Vis/MinInliers" value="60" />
        <param name="RGBD/MaxOdomCacheSize" value="10" />
        <param name="SuperPoint/ModelPath" value="$(find genius_launch)/config/superpoint.pt" />
        <param name="Vis/CorNNType" value="1" />
        <param name="PyMatcher/Path" value="/home/abyl/dependencies/rtabmap/corelib/src/python/rtabmap_superglue.py" />
        <param name="PyDetector/Path" value="/home/abyl/dependencies/rtabmap/corelib/src/python/rtabmap_superpoint.py" />
        <param name="PyDetector/Cuda" value="true" />
        <param name="PyMatcher/Cuda" value="true" />
        <param name="SuperPoint/ModelPath" value="$(find genius_launch)/config/superpoint.pt" />
```

However, I still cannot get a loop closure and getting the following warning:

```
[ WARN] (2022-09-06 19:55:58.046) OptimizerG2O.cpp:1157::optimize() Computing marginals: vertex 600 has negative hessian index (-1). Cannot compute last pose covariance.
[ WARN] (2022-09-06 19:55:58.128) RegistrationIcp.cpp:904::computeTransformationImpl() Transform is found (xyz=0.007863,-0.048834,0.000000 rpy=0.000000,-0.000000,0.010339) but no correspondences has been found!? Variance is unknown!
[ WARN] (2022-09-06 19:55:58.128) Rtabmap.cpp:2788::process() Rejected loop closure 460 -> 881: Cannot compute transform (cor=0 corrRatio=0.000000/0.010000 maxLaserScans=1171)
```

matlabbe, 2022-09-06:
Show "rtabmap --version", it was working pretty well on my side yesterday...

abylikhsanov, 2022-09-07:
@matlabbe 

```
RTAB-Map:              0.20.18
PCL:                    1.10.0
With VTK:                7.1.1
OpenCV:                  4.2.0
With OpenCV xfeatures2d:  true
With OpenCV nonfree:      true
With ORB OcTree:          true
With SuperPoint Torch:    true
With Python3:             true
With FastCV:             false
With Madgwick:            true
With PDAL:               false
With TORO:                true
With g2o:                 true
With GTSAM:              false
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI2:             true
With Freenect:           false
With Freenect2:          false
With K4W2:               false
With K4A:                false
With DC1394:              true
With FlyCapture2:        false
With ZED:                false
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:          true
With MYNT EYE S:         false
With DepthAI:            false
With libpointmatcher:     true
With CCCoreLib:          false
With octomap:            false
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FLOAM:              false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false
```

matlabbe, 2022-09-07:
With libpointmacher, you can increase `Icp/MaxCorrespondenceDistance` to 0.5 like what I did in my previous post. On that computer you are using opencv version with xfeatures2d, beware that the other computer you create the map has also the same dependency, otherwise not the same feature will be used by default. For the "limit out of bounds" error, you could increase `Icp/MaxTranslation` to 0.5 (default 0.2).

To reprocess the database with different feature type, here is the command I used:
```
rtabmap-reprocess --Vis/FeatureType 8 --Kp/DetectorStrategy 8 map.db output.db
```