introlab/rtabmap_ros #737: Undefined reference error while building

zippyzoo77, 2022-03-29:
I am trying to update my version of rtabmap to the current one on Ubuntu 20.04 with ros noetic. While following the instructions below from [here](https://github.com/introlab/rtabmap_ros):
`###########
# rtabmap
###########
$ cd rtabmap
$ git pull origin master
$ cd build
$ make #<-- This is where I get the error
$ make install
# Do "sudo make install" if you installed rtabmap in "/usr/local"

###########
# rtabmap_ros
###########
$ roscd rtabmap_ros
$ git pull origin master
$ roscd
$ cd ..
$ catkin_make -j1 --pkg rtabmap_ros`

At the indicated point, I end up with the following error:  
`(base) zippyzoo77@MountainLion:~/rtabmap/build$ make
[  4%] Built target rtabmap_utilite
[  5%] Built target res_tool
[ 42%] Built target rtabmap_core
[ 80%] Built target rtabmap_gui
[ 81%] Linking CXX executable ../../bin/rtabmap
/usr/bin/ld: ../../bin/librtabmap_core.so.0.20.19: undefined reference to `octomath::Pose6D::Pose6D(octomath::Pose6D const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [app/src/CMakeFiles/rtabmap.dir/build.make:243: bin/rtabmap] Error 1
make[1]: *** [CMakeFiles/Makefile2:993: app/src/CMakeFiles/rtabmap.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
`

I have done all the proper steps and have octomap installed, so I really don't know what could be causing this error. Any help on debugging this so I can get rtabmap up and working would be greatly appreciated. Thanks!

matlabbe, 2022-04-11:
Which OctoMap version have you installed? `ros-noetic-octomap`?

matlabbe, 2022-04-11:
Cross-ref: https://answers.ros.org/question/398338/undefined-reference-error-while-building/?answer=398587#post-id-398587

closed 2022-04-11 by matlabbe

matlabbe, 2023-01-27:
On Noetic, another cause is that both `liboctomap-dev` and `ros-noetic-octomap` may be installed at the same time (some ros packages depend on the system version, and other from the noetic version). Cleanup rtabmap's build directory, make sure you sourced `/opt/ros/noetic/setup.bash` and redo cmake/make. Rtabmap's cmake should find the version in `/opt/ros/noetic` first, thus it will work with `ros-noetic-octomap-msgs` used by `rtabmap_ros`, and the undefined reference error won't show up.