introlab/rtabmap_ros #740: error while loading shared libraries: libnvrm.so

zzurc, 2022-04-01:
Hi,

I am having an issue getting rtabmap to work on a Jetson Nano. I plan to use the Jetson Nano, a RealSense D435i RGBD camera, and an RPlidar to do autonomous navigation with ROS Melodic and rtabmap. 

I am following the instructions from here to launch rtabmap: [https://github.com/introlab/rtabmap_ros/tree/master/docker](url)
However, I am getting the following error for a missing file or directory.  I would greatly appreciate it if anyone could direct me to solution to this problem.

Thanks!

'jetson@jetson-bot:~$ sudo docker run -it --rm   --privileged   --env="DISPLAY=$DISPLAY"   --env="QT_X11_NO_MITSHM=1"   --volume="/tmp/.X11-unix:/tmp/.X11-unix:rw"   --env="XAUTHORITY=$XAUTH"   --volume="$XAUTH:$XAUTH"   --runtime=nvidia   --env ROS_MASTER_URI=http://localhost:11311    -v ~/.ros:/root    rtabmap_ros3d   roslaunch rtabmap_ros rtabmap.launch database_path:=/root/rtabmap.db rtabmap_args:="--delete_db_on_start"
... logging to /root/.ros/log/db4318a4-b1e3-11ec-8129-0242ac110002/roslaunch-392faf149335-1.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://392faf149335:37665/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmapviz/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.12
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/expected_update_rate: 0.0
 * /rtabmap/rgbd_odometry/frame_id: camera_link
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/keep_color: False
 * /rtabmap/rgbd_odometry/max_update_rate: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rgbd_odometry/wait_imu_to_init: False
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: /root/rtabmap.db
 * /rtabmap/rtabmap/frame_id: camera_link
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgb: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmapviz/approx_sync: True
 * /rtabmap/rtabmapviz/frame_id: camera_link
 * /rtabmap/rtabmapviz/odom_frame_id: 
 * /rtabmap/rtabmapviz/queue_size: 10
 * /rtabmap/rtabmapviz/subscribe_depth: True
 * /rtabmap/rtabmapviz/subscribe_odom_info: True
 * /rtabmap/rtabmapviz/subscribe_rgb: True
 * /rtabmap/rtabmapviz/subscribe_rgbd: False
 * /rtabmap/rtabmapviz/subscribe_scan: False
 * /rtabmap/rtabmapviz/subscribe_scan_cloud: False
 * /rtabmap/rtabmapviz/subscribe_scan_descriptor: False
 * /rtabmap/rtabmapviz/subscribe_stereo: False
 * /rtabmap/rtabmapviz/wait_for_transform_duration: 0.2

NODES
  /rtabmap/
    rgbd_odometry (rtabmap_ros/rgbd_odometry)
    rtabmap (rtabmap_ros/rtabmap)
    rtabmapviz (rtabmap_ros/rtabmapviz)

auto-starting new master
process[master]: started with pid [41]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to db4318a4-b1e3-11ec-8129-0242ac110002
process[rosout-1]: started with pid [54]
started core service [/rosout]
process[rtabmap/rgbd_odometry-2]: started with pid [74]
process[rtabmap/rtabmap-3]: started with pid [75]
/opt/ros/melodic/lib/rtabmap_ros/rtabmap: error while loading shared libraries: libnvrm.so: cannot open shared object file: No such file or directory
/opt/ros/melodic/lib/rtabmap_ros/rgbd_odometry: error while loading shared libraries: libnvrm.so: cannot open shared object file: No such file or directory
[rtabmap/rgbd_odometry-2] process has died [pid 74, exit code 127, cmd /opt/ros/melodic/lib/rtabmap_ros/rgbd_odometry --delete_db_on_start rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info rgbd_image:=rgbd_image_relay odom:=odom imu:=/imu/data __name:=rgbd_odometry __log:=/root/.ros/log/db4318a4-b1e3-11ec-8129-0242ac110002/rtabmap-rgbd_odometry-2.log].
log file: /root/.ros/log/db4318a4-b1e3-11ec-8129-0242ac110002/rtabmap-rgbd_odometry-2*.log
[rtabmap/rtabmap-3] process has died [pid 75, exit code 127, cmd /opt/ros/melodic/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info rgbd_image:=rgbd_image_relay left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud scan_descriptor:=/scan_descriptor user_data:=/user_data user_data_async:=/user_data_async gps/fix:=/gps/fix tag_detections:=/tag_detections fiducial_transforms:=/fiducial_transforms odom:=odom imu:=/imu/data __name:=rtabmap __log:=/root/.ros/log/db4318a4-b1e3-11ec-8129-0242ac110002/rtabmap-rtabmap-3.log].
log file: /root/.ros/log/db4318a4-b1e3-11ec-8129-0242ac110002/rtabmap-rtabmap-3*.log
process[rtabmap/rtabmapviz-4]: started with pid [76]
/opt/ros/melodic/lib/rtabmap_ros/rtabmapviz: error while loading shared libraries: libnvrm.so: cannot open shared object file: No such file or directory
[rtabmap/rtabmapviz-4] process has died [pid 76, exit code 127, cmd /opt/ros/melodic/lib/rtabmap_ros/rtabmapviz -d ~/.ros/rtabmap_gui.ini rgb/image:=/camera/rgb/image_rect_color depth/image:=/camera/depth_registered/image_raw rgb/camera_info:=/camera/rgb/camera_info rgbd_image:=rgbd_image_relay left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud scan_descriptor:=/scan_descriptor odom:=odom __name:=rtabmapviz __log:=/root/.ros/log/db4318a4-b1e3-11ec-8129-0242ac110002/rtabmap-rtabmapviz-4.log].
log file: /root/.ros/log/db4318a4-b1e3-11ec-8129-0242ac110002/rtabmap-rtabmapviz-4*.log'

matlabbe, 2022-04-11:
See https://github.com/introlab/rtabmap/issues/776#issuecomment-959804848

zzurc, 2022-04-15:
Thank you matlabbe,
I was able to duplicate your results in https://github.com/introlab/rtabmap/issues/776#
I went ahead and followed the instructions in https://github.com/introlab/rtabmap_ros/tree/master/docker so I can run rtabmap with ROS but I am still having issues. I will open another issue for these errors.
Thanks again.

closed 2022-04-19 by zzurc