introlab/rtabmap_ros #741: RTAB Map dies with code -11 when run using two web cameras

IsuruMS, 2022-04-02:
We are running RTAB map with two web cameras with the following launch file.
```
<launch>
    <!-- Choose visualization -->
    <arg name="rviz" default="true" />

    <!-- Run the ROS package stereo_image_proc for image rectification and disparity computation -->
    <group ns="stereo">
        <node pkg="stereo_image_proc" type="stereo_image_proc" name="stereo_image_proc">
            <remap from="left/image_raw"    to="left/image_raw"/>
            <remap from="left/camera_info"  to="left/camera_info"/>
            <remap from="right/image_raw"   to="right/image_raw"/>
            <remap from="right/camera_info" to="right/camera_info"/>
            <param name="disparity_range" value="128"/>
            <param name="approximate_sync"        type="bool" value="True"/>
            <param name="queue_size"        type="int" value="10"/>
        </node>

        
      <!-- Disparity to depth -->
        <node pkg="nodelet" type="nodelet" name="disparity2depth" args="standalone rtabmap_ros/disparity_to_depth">
            <param name="disparity_range" value="128"/>
            <param name="approx_sync"  type="bool" value="true"/>
            <param name="queue_size"        type="int" value="100"/>
        </node>
    </group>
  
    <group ns="rtabmap">
        <node pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" output="screen">
            <remap from="left/image_rect"       to="/stereo/left/image_rect_color"/>
            <remap from="right/image_rect"      to="/stereo/right/image_rect_color"/>
            <remap from="left/camera_info"      to="/stereo/left/camera_info"/>
            <remap from="right/camera_info"     to="/stereo/right/camera_info"/>
            <param name="subscribe_rgbd"  type="bool" value="false"/>
            <!-- <remap from="rgbd_image"            to="/stereo/rgbd_image"/> -->

            <!-- Published topic odom is converted into stereo_odometry -->
            <remap from="odom"                  to="/stereo_odometry"/>

            <param name="approx_sync"  type="bool" value="true"/>
            <param name="frame_id"        type="string" value="base_link"/>
            <param name="odom_frame_id"   type="string" value="odom"/>
            <param name="publish_tf"        type="bool" value="true"/>
            <param name="wait_for_transform"        type="string" value="true"/>
            <param name="queue_size"        type="int" value="100"/>
            
            <param name="Odom/Strategy"      type="string" value="0"/> <!-- 0=Frame-to-Map, 1=Frame=to=Frame -->
            <param name="Vis/EstimationType" type="string" value="1"/> <!-- 0=3D->3D 1=3D->2D (PnP) -->
            <param name="Vis/CorType" value="0"/> <!-- Correspondences: 0=Features Matching, 1=Optical Flow -->
            <!-- maximum features map size, default 2000 -->
            <param name="OdomF2M/MaxSize" type="string" value="1000"/> 
            <!-- maximum features extracted by image, default 1000 -->
            <param name="Vis/MaxFeatures" type="string" value="600"/>
            <param name="Vis/MinInliers"     type="string" value="5"/>
            <param name="Stereo/MaxDisparity"  type="string" value="128"/>
            <param name="Stereo/OpticalFlow"  type="string" value="false"/>

        </node>
        
        <!-- Visual SLAM -->
        <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
            <param name="subscribe_depth"     type="bool" value="true"/>
            <param name="subscribe_rgbd"     type="bool" value="false"/>
            <param name="subscribe_scan" type="bool" value="false"/>
    
            <remap from="rgb/image"       to="/stereo/left/image_rect_color"/>
            <remap from="rgb/camera_info" to="/stereo/left/camera_info"/>
            <remap from="depth/image"     to="/stereo/depth"/>
    
            <remap from="odom" to="/stereo_odometry"/>
    
            <param name="frame_id"    type="string" value="/base_link"/>
            <param name="queue_size"  type="int"    value="2000"/>
            <param name="approx_sync" type="bool"   value="true"/>
    
            <param name="Vis/MinInliers" type="string" value="12"/>
        </node>
    </group>

    <!-- Visualisation RVIZ --> 
    <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(find rtabmap_ros)/launch/config/demo_stereo_outdoor.rviz"/>

  </launch>
```

Our TF transform tree has base_link -> stereo/left, base_link -> stereo/right, odom -> base_link
[image]

After running rostopic hz stereo/depth stereo/left/image_rect_color stereo/left/camera_info we get the following output as well. 
[image]

And finally here is the `rqt_graph`
[image]

We are new to using rtab map. If someone can offer a help it would be much appreciated.

IsuruMS, 2022-04-02:
Here is the error that we are getting in the terminal
[image]

IsuruMS, 2022-04-05:
We finally found the issue. **For any of those who are having rtab map odometry or mapping issues with stereo or RGBD sensors like Kinect 360 build RTAB Map from the source**. 

BTW you can use **two USB web cams** (YOU DON'T EVEN NEED BRANDED ONES LIKE LOGITECH OURS WERE UNBRANDED CHINESE ONES) as a stereo sensor. You can either sync them using a small code or you can use `approx_sync` parameter to `true`. How ever keep in mind that you have to configure TF transformations from base_link to camera_link and camera_link to individual cameras (left and right) as well to make it work. AND **BEWARE** Sometimes stereo setup like this may not give you a highly accurate map or odometry.

You can also try with a kinect 360 which is really cheap these days.

closed 2022-04-05 by IsuruMS

matlabbe, 2022-04-11:
If you use independent mono cameras to create a stereo camera, synchronization between the cameras is important. We recently added "approx_sync_max_interval" parameter to ignore images synchronized with too much delay between them. For example setting it `approx_sync_max_interval=0.001`, you make sure that all left/right images will have at most 1 ms delay between them, otherwise they are ignored. Note sure though how you solved the crash.