introlab/rtabmap_ros #745: rtabmap_ros node crashes 20.04 Noetic

pauldeee, 2022-04-07:
Hi Mathieu,

ROS Noetic
Ubuntu 20.04 LTS

After some time of mapping, 2 - 5 minutes or sometimes longer, I'm confronted with a rtabmap_ros node crash. Here is a BT from install with GTSAM installed via PPA.

```#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff6b0c859 in __GI_abort () at abort.c:79
#2  0x00007ffff6d97911 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff6da338c in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff6da33f7 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff6da36fd in __cxa_rethrow ()
    at /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff198d45d in gtsam::VectorValues gtsam::internal::linearAlgorithms::optimizeBayesTree<gtsam::GaussianBayesTree>(gtsam::GaussianBayesTree const&)
    () at /lib/x86_64-linux-gnu/libgtsam.so.4
#7  0x00007ffff1979f32 in gtsam::GaussianBayesTree::optimize() const ()
    at /lib/x86_64-linux-gnu/libgtsam.so.4
#8  0x00007ffff19a7f72 in gtsam::GaussianFactorGraph::optimize(gtsam::Ordering const&, std::function<std::pair<boost::shared_ptr<gtsam::GaussianConditional>, boost::shared_ptr<gtsam::GaussianFactor> > (gtsam::GaussianFactorGraph const&, gtsam::Ordering const&)> const&) const () at /lib/x86_64-linux-gnu/libgtsam.so.4
#9  0x00007ffff1a82f3b in gtsam::NonlinearOptimizer::solve(gtsam::GaussianFactorGraph const&, gtsam::NonlinearOptimizerParams const&) const ()
    at /lib/x86_64-linux-gnu/libgtsam.so.4
#10 0x00007ffff1a385eb in gtsam::GaussNewtonOptimizer::iterate() ()
    at /lib/x86_64-linux-gnu/libgtsam.so.4
#11 0x00007ffff7620fd3 in rtabmap::OptimizerGTSAM::optimize(int, std::map<int, rtabmap::Transform, std::less<int>, std::allocator<std::pair<int const, rtabmap::--Type <RET> for more, q to quit, c to continue without paging--#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
```

I've experienced this same issue installing GTSAM both from source and PPA. 

Maybe this is worth noting but in the rtabmap install I ran into[ this same cmake error](https://github.com/introlab/rtabmap/issues/215) and was able to successfully build it using [this command](https://github.com/PointCloudLibrary/pcl/issues/5052#issuecomment-984042504)

Any idea what might be wrong here?

pauldeee, 2022-04-10:
After a fresh install of Ubuntu LTS 20.04, ROS Noetic, with rtabmap and GTSAM from source I am no longer running into this issue. However, I still had to use the ```-DBoost_LIBRARY_DIR_RELEASE=/usr/lib/x86_64-linux-gnu``` flag when invoking both cmake in the rtabmap/build directory and with catkin_make in the ros workspace. Which I never needed in previous installs.

closed 2022-04-10 by pauldeee

matlabbe, 2022-04-11:
Thanks for sharing the workaround. I am on Noetic and I don't need setting explicitly `Boost_LIBRARY_DIR_RELEASE`. It seems that docker images don't have the problem too. Otherwise for GTSAM, I use the version 4 from PPA, not the latest from PPA. See also this issue: https://github.com/introlab/rtabmap/issues/823

cheers,
Mathieu

ProfFan, 2022-04-19:
Hi, just want to chime out that this issue has been resolved in `develop` branch of GTSAM :)