introlab/rtabmap_ros #754: some warn when run rtabmap.launch

ysys98, 2022-05-04:
Hi，
The operating system of my computer is Ubuntu 16.04  ，ROS kinetic
I've been using AUSU xtion Pro live for rtabmap slam recently，(I used Kinect + Odometry + Fake 2D laser from Kinect )
When I was running, I encountered some warn：
[image]
[image]
How can I solve this warn？
I did publish the tf from odom to map, and base_ link to camera_ link，
I can build 3D map correctly and make rtabmap.db export 3D map
Another problem I have is that when I run rtabmap again, I removed --delete_db_on_start.
I want to improve the map on the basis of the map I have built
I put the robot in another position, which is far away from the position of the robot at the end of the last drawing map
I still open rviz and rtabmapviz 
however，At the end of the last time, the robot continued to build the map according to its position, 
instead of the new position I put robot in now
I ran openni2 lanunch
and here is my rtabmap launch file
`<?xml version="1.0" encoding="UTF-8"?>
<launch> 
    <arg name="rgb_topic" default="/camera/rgb/image_rect_color"/> 
    <arg name="depth_topic" default="/camera/depth_registered/image_raw"/> 
    <arg name="camera_info_topic" default="/camera/rgb/camera_info"/>
    <arg name="scan" default="/kinect_scan"/>
    
    <node pkg="depthimage_to_laserscan" type="depthimage_to_laserscan" name="depthimage_to_laserscan">
      <remap from="image"       to="/camera/depth_registered/image_raw"/>
      <remap from="camera_info" to="/camera/depth_registered/camera_info"/>
      <remap from="scan"        to="/kinect_scan"/>
      <param name="range_max" type="double" value="4"/>
    </node>
    
    <group ns="rtabmap">
        <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
        
            
            <param name="frame_id"            type="string" value="base_link"/>
            <param name="odom_frame_id"       type="string" value="odom"/>
            <param name="subscribe_depth"     type="bool"   value="true"/>
            <param name="subscribe_scan"      type="bool"   value="true"/>

            
            <remap from="scan"            to="/kinect_scan"/>
            <remap from="rgb/image"       to="/camera/rgb/image_rect_color"/>
            <remap from="depth/image"     to="/camera/depth_registered/image_raw"/>
            <remap from="rgb/camera_info" to="/camera/rgb/camera_info"/>
            <remap from="odom"       to="/odom"/>

            <param name="queue_size" type="int" value="15"/>

            
            <remap from="/rtabmap/grid_map" to="/map"/>

            
            <param name="Rtabmap/DetectionRate" type="string" value="1"/>
            <param name="RGBD/OptimizeFromGraphEnd"     type="string" value="false"/>

            
            <param name="Reg/Force3DoF" type="string" value="true"/>

            
            <param name="Kp/DetectorStrategy" type="string" value="0"/>

            
            <param name="Kp/MaxFeatures" type="string" value="400"/>

            
            <param name="SURF/HessianThreshold" type="string" value="100"/>

            
            
            <param name="Reg/Strategy" type="string" value="0"/>

            
            <param name="Vis/MinInliers" type="string" value="15"/>

            
            <param name="Mem/NotLinkedNodesKept" type="string" value="false"/>
        </node>

        
        <node pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
            <param name="subscribe_depth"             type="bool" value="true"/>
            <param name="subscribe_scan"              type="bool" value="true"/>
            <param name="frame_id"                    type="string" value="base_link"/>
            <param name="queue_size"                  type="int" value="10"/>
            <remap from="odom"            to="/odom"/>
            <remap from="rgb/image"       to="$(arg rgb_topic)"/>
            <remap from="depth/image"     to="$(arg depth_topic)"/>
            <remap from="rgb/camera_info" to="$(arg camera_info_topic)"/>
            <remap from="scan"            to="/kinect_scan"/>
        </node>
    </group>
    <group ns="planner">
        <remap from="openni_points" to="/planner_cloud"/>
        <remap from="map" to="/rtabmap/grid_map"/>
        <remap from="move_base_simple/goal" to="/planner_goal"/>
            
        <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
            <rosparam file="$(find slam)/param/costmap_common_params.yaml" command="load" ns="global_costmap" />
            <rosparam file="$(find slam)/param/costmap_common_params.yaml" command="load" ns="local_costmap" />
            <rosparam file="$(find slam)/param/local_costmap_params.yaml" command="load" />
            <rosparam file="$(find slam)/param/global_costmap_params.yaml" command="load" />
            <rosparam file="$(find slam)/param/base_local_planner_params.yaml" command="load" />
        </node>
   </group>
</launch>`
How can I make the robot continue to build maps in robot's new location? 
Do I need to add an amcl.launch?

matlabbe, 2022-05-10:
For the `Received odom topic with child_Frame_id not set` warning, it it what it said, you are feeding an odometry topic without child_frame_id set.

ysys98, 2022-05-11:
> For the `Received odom topic with child_Frame_id not set` warning, it is what it said, you are feeding an odometry topic without child_frame_id set.

@matlabbe How can I solve this problem? I'm sure that published odom to base_ link  transformation, this is my TF tree
[frames.pdf](https://github.com/introlab/rtabmap_ros/files/8667928/frames.pdf)

matlabbe, 2022-05-11:
this warning is not about TF, it is about the odometry topic. Do `rostopic echo /odometry_topic_used`

ysys98, 2022-05-11:
> this warning is not about TF, it is about the odometry topic. Do `rostopic echo /odometry_topic_used`

@matlabbe Dear matlabbe ，
OK, thank you. I'll do as you say. I have some other questions
I am a beginner
I built the map through rtabmap_ros slam,
[image]
and I switch to Localization mode and restart rtabmap_ ros
roslaunch rtabmap_ros rtabmap.launch localization:=true
rtabmap.launch is my launch file 

[image]
I can't let the robot determine his own position
I did click the "2D Pose Estimate" button
The position of the robot produces elegance
How can I solve this problem

ysys98, 2022-05-11:
> this warning is not about TF, it is about the odometry topic. Do `rostopic echo /odometry_topic_used`

@matlabbe  I understand what you mean. In rostopic echo /odom ，It doesn't exist child_frame_id here，there is no odom to base_ link coordinate transformation, but when I rosrun nao_bringup nao_full_py.launch,it does release odom to base_ link coordinate transformation, here is my TF tree
[frames.pdf](https://github.com/introlab/rtabmap_ros/files/8670417/frames.pdf)

matlabbe, 2022-05-14:
Well, if you cannot modify the code publishing your /odom topic to fix the child_frame_id not set, you may ignore the warning about that. I see also that your odom topic doesn't have covariance set. You can use odom_frame_id parameter instead of using the topic:
```xml
<param name="odom_frame_id" value="odom"/>
```
This will remove the warning and odometry will be taken directly from TF.

In localization mode, when you see that an hypothesis has been rejected, look at the terminal for the reason. Also when relaunching in localization mode, do Edit->Download all clouds in rtabmapviz so that you can see the actual loop closure hypothesis where it is red (to know if it is he same place).

ysys98, 2022-05-14:
> Well, if you cannot modify the code publishing your /odom topic to fix the child_frame_id not set, you may ignore the warning about that. I see also that your odom topic doesn't have covariance set. You can use odom_frame_id parameter instead of using the topic:
> 
> ```
> <param name="odom_frame_id" value="odom"/>
> ```
> 
> This will remove the warning and odometry will be taken directly from TF.
> 
> In localization mode, when you see that an hypothesis has been rejected, look at the terminal for the reason. Also when relaunching in localization mode, do Edit->Download all clouds in rtabmapviz so that you can see the actual loop closure hypothesis where it is red (to know if it is he same place).

Hi, @matlabbe Thanks for your help, When I do Edit->Download all clouds in rtabmapviz, it worked,
but, I also have some trouble,
here is my log info:
[image]
[image]
[image]
I have so many parameters not found 
and so many Unknown parameters
and when I do Edit->Download all clouds
robot might find itself postion
but the warn log says：
[ INFO] [1652546688.359312358]: rtabmap (952): Rate=1.00s, Limit=0.000s, RTAB-Map=0.2972s, Maps update=0.0189s pub=0.0000s (local map=219, WM=219)
[ INFO] [1652546688.387719353]: Odom: quality=329, std dev=0.004898m|0.059190rad, update time=0.169109s
[ INFO] [1652546688.576276847]: Odom: quality=332, std dev=0.005312m|0.053014rad, update time=0.172483s
[ INFO] [1652546688.767422950]: Odom: quality=309, std dev=0.007252m|0.067783rad, update time=0.171086s
[ INFO] [1652546688.970389007]: Odom: quality=273, std dev=0.024636m|0.077719rad, update time=0.180912s
[ INFO] [1652546689.157523252]: Odom: quality=177, std dev=0.006254m|0.051393rad, update time=0.164852s
[ INFO] [1652546689.357883188]: Odom: quality=40, std dev=0.005050m|0.043851rad, update time=0.193654s
[ INFO] [1652546689.496330294]: rtabmap (953): Rate=1.00s, Limit=0.000s, RTAB-Map=0.3104s, Maps update=0.0193s pub=0.0000s (local map=219, WM=219)
[ INFO] [1652546689.594488469]: Odom: quality=344, std dev=0.003464m|0.041757rad, update time=0.215921s
[ WARN] [1652546689.686601255]: The origin for the sensor at (0.41, 0.39, 0.47) is out of map bounds. So, the costmap cannot raytrace for it.
[ INFO] [1652546689.813859097]: Odom: quality=330, std dev=0.006633m|0.048014rad, update time=0.205766s
[ WARN] (2022-05-15 00:44:49.989) OdometryF2M.cpp:561::computeTransform() Registration failed: "Too low inliers after bundle adjustment: 11<20" (guess=xyz=0.000076,-0.000220,0.000000 rpy=0.000000,0.000000,0.000007)
[ WARN] (2022-05-15 00:44:49.990) OdometryF2M.cpp:323::computeTransform() Failed to find a transformation with the provided guess (xyz=0.000076,-0.000220,0.000000 rpy=0.000000,0.000000,0.000007), trying again without a guess.
[ WARN] (2022-05-15 00:44:50.328) OdometryF2M.cpp:551::computeTransform() Trial with no guess still fail.
[ WARN] (2022-05-15 00:44:50.328) OdometryF2M.cpp:561::computeTransform() Registration failed: "Too low inliers after bundle adjustment: 17<20" (guess=xyz=0.000076,-0.000220,0.000000 rpy=0.000000,0.000000,0.000007)
[ INFO] [1652546690.329437489]: Odom: quality=17, std dev=0.000000m|0.000000rad, update time=0.509494s
[ WARN] (2022-05-15 00:44:50.679) OdometryF2M.cpp:557::computeTransform() Registration failed: "Too low inliers after bundle adjustment: 16<20"
[ INFO] [1652546690.680880787]: Odom: quality=16, std dev=0.000000m|0.000000rad, update time=0.331924s
[ WARN] [1652546690.687054211]: The origin for the sensor at (0.41, 0.39, 0.47) is out of map bounds. So, the costmap cannot raytrace for it.
[ WARN] (2022-05-15 00:44:51.079) OdometryF2M.cpp:557::computeTransform() Registration failed: "Too low inliers after bundle adjustment: 15<20"
[ INFO] [1652546691.084406344]: Odom: quality=15, std dev=0.000000m|0.000000rad, update time=0.396592s
[ INFO] [1652546691.527777334]: Odom: quality=24, std dev=0.007517m|0.066416rad, update time=0.429245s
[ WARN] [1652546691.687182729]: The origin for the sensor at (0.41, 0.39, 0.47) is out of map bounds. So, the costmap cannot raytrace for it.
[image]
what should I do？

matlabbe, 2022-05-15:
Did you build also rtabmap from source? The version shown is 0.20.19, while the kinetic binaries are stuck at 0.20.7. The map seems to have been created with 0.20.7, then you try to localize with 0.20.19. It is why a lot of warnings about parameters not existing in the database.

Then for those warnings: `Registration failed: "Too low inliers after bundle adjustment: 15<20"`, you could decrease `Vis/MinInliers` parameter to 10 to accept more localizations.

ysys98, 2022-05-16:
> Did you build also rtabmap from source? The version shown is 0.20.19, while the kinetic binaries are stuck at 0.20.7. The map seems to have been created with 0.20.7, then you try to localize with [...]

@matlabbe Thans for your help！
I follow your advice to build rtabmap from source
and then I sudo apt-get remove ros-kinetic-rtabmap-ros
and  build rtabmap-ros from source
It outputs some error logs：
[image]
[image]
what should I do？

ysys98, 2022-05-17:
> Well, if you cannot modify the code publishing your /odom topic to fix the child_frame_id not set, you may ignore the warning about that. I see also that your odom topic doesn't have covariance set. [...]

@matlabbe Hi, follow this question,
I follow [http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot](url) to build myself robot
 I changed <arg name="rgbd_odometry"     default="false"/>  back
and then <param unless="$(arg rgbd_odometry)" name="odom_frame_id" value="odom"/> worked
when I roslaunch my file and rviz 
I click 2d nav goal 
It has warn log info ：
[ WARN] [1652785301.688729770]: Clearing costmap to unstuck robot (3.000000m).
[ WARN] [1652785312.288919824]: Rotate recovery behavior started.
And robot will not move towards the target
[image]
Thank you very much for your patience in answering my stupid question
here it is my log info：
ros-kinetic@roskinetic-HP-Pavilion-Laptop-15-cw0xxx:~/rtabmap_ws $ roslaunch nao_bringup demo_nao_mapping.launch 
... logging to /home/ros-kinetic/.ros/log/f7aeee74-d5b4-11ec-a0de-802bf9bda405/roslaunch-roskinetic-HP-Pavilion-Laptop-15-cw0xxx-24261.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://roskinetic-HP-Pavilion-Laptop-15-cw0xxx:34211/

SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet_manager/num_worker_threads: 4
 * /camera/depth_rectify_depth/interpolation: 0
 * /camera/depth_registered_rectify_depth/interpolation: 0
 * /camera/disparity_depth/max_range: 4.0
 * /camera/disparity_depth/min_range: 0.5
 * /camera/disparity_registered_hw/max_range: 4.0
 * /camera/disparity_registered_hw/min_range: 0.5
 * /camera/driver/color_depth_synchronization: False
 * /camera/driver/color_mode: 5
 * /camera/driver/depth_camera_info_url: 
 * /camera/driver/depth_frame_id: camera_depth_opti...
 * /camera/driver/depth_mode: 5
 * /camera/driver/depth_registration: True
 * /camera/driver/device_id: #1
 * /camera/driver/enable_reconnect: True
 * /camera/driver/id_manufacturer: 1d27
 * /camera/driver/id_product: 0601
 * /camera/driver/ir_mode: 5
 * /camera/driver/rgb_camera_info_url: 
 * /camera/driver/rgb_frame_id: camera_rgb_optica...
 * /camera/driver/serialnumber_as_name: False
 * /depthimage_to_laserscan/output_frame_id: camera_depth_frame
 * /depthimage_to_laserscan/range_min: 0.45
 * /depthimage_to_laserscan/scan_height: 10
 * /move_base/DWAPlannerROS/acc_lim_theta: 2.0
 * /move_base/DWAPlannerROS/acc_lim_x: 1.0
 * /move_base/DWAPlannerROS/acc_lim_y: 0.0
 * /move_base/DWAPlannerROS/forward_point_distance: 0.325
 * /move_base/DWAPlannerROS/global_frame_id: odom
 * /move_base/DWAPlannerROS/goal_distance_bias: 24.0
 * /move_base/DWAPlannerROS/max_rot_vel: 5.0
 * /move_base/DWAPlannerROS/max_scaling_factor: 0.2
 * /move_base/DWAPlannerROS/max_trans_vel: 0.5
 * /move_base/DWAPlannerROS/max_vel_x: 0.5
 * /move_base/DWAPlannerROS/max_vel_y: 0.0
 * /move_base/DWAPlannerROS/min_rot_vel: 0.4
 * /move_base/DWAPlannerROS/min_trans_vel: 0.1
 * /move_base/DWAPlannerROS/min_vel_x: 0.0
 * /move_base/DWAPlannerROS/min_vel_y: 0.0
 * /move_base/DWAPlannerROS/occdist_scale: 0.5
 * /move_base/DWAPlannerROS/oscillation_reset_dist: 0.05
 * /move_base/DWAPlannerROS/path_distance_bias: 64.0
 * /move_base/DWAPlannerROS/publish_cost_grid_pc: True
[... 774 lines cut ...]
/rtabmap/landmarks
/rtabmap/local_grid_empty
/rtabmap/local_grid_ground
/rtabmap/local_grid_obstacle
/rtabmap/local_path
/rtabmap/local_path_nodes
/rtabmap/localization_pose
/rtabmap/mapData
/rtabmap/mapGraph
/rtabmap/mapPath
/rtabmap/octomap_binary
/rtabmap/octomap_empty_space
/rtabmap/octomap_full
/rtabmap/octomap_global_frontier_space
/rtabmap/octomap_grid
/rtabmap/octomap_ground
/rtabmap/octomap_obstacles
/rtabmap/octomap_occupied_space
/rtabmap/proj_map
/rtabmap/scan_map
/rtabmap/user_data_async
/scan
/speech
/tf
/tf_static

matlabbe, 2022-09-05:
Well, I cannot see the how fully rtabmap is configured, but:

> then I think I should publish transform from base_link to base_footprint and map to odom and I did that
but NAO also can not move towards to the target and also apper warn log：

You should not publish a fixed tf on map->odom, rtabmap will publish it.

For the planning issues, unless it is related to map->base_link TF issue, it would be more a planning issue.

closed 2022-10-09 by ysys98