introlab/rtabmap_ros #755: setting use_sim_time equal to true for bag files

MJA0111, 2022-05-05:
Hi,

I am using a bag file including the rgb image topic, depth topic, camera info and tf topics. I need to set "use_sim_time" to true for simulation clock purposes, but RTABMAP does not publish messages in the map_data topic.
I tried the same thing with the bag file and launch file provided by you in your tutorial "demo_mapping.bag", [here](http://wiki.ros.org/rtabmap_ros#Robot_mapping), and again I receive no message from map topic.
It seems for bag files, the use_sim_time param should be set always to false. I would appreciate it if you could help me to solve this issue. Thanks in advance :)

matlabbe, 2022-05-09:
Just tried the demo and it still work:
[image]

How do you launch the demo? Are there warnings/errors in the terminal?

MJA0111, 2022-05-09:
Hi again!
I think the problem the problem was the --topic in the bag file terminal line. The problem was there were no publishing map in the view_frames. Thanks, it is solved.

closed 2022-05-12 by matlabbe