introlab/rtabmap_ros #756: rtabmap_ros looks for rtabmap installed from ROS PPA rather than the one build from source

adbidwai, 2022-05-06:
I am trying to build rtabmap_ros from source.

I have already build rtabmap from source and installed it. But at the same time I also have rtabmap installed from ROS PPA using `sudo apt install ros-melodic-rtabmap` 

So how do I tell my rtabmap-ros package to refer to the rtabmap which is installed from source and not from the ROS PPA?

Thanks!

MJA0111, 2022-05-06:
Hi,
I think the solution is to overlay the workspaces. Maybe this two links can help you:
[https://answers.ros.org/question/349261/using-at-the-same-time-a-binary-installed-and-a-source-installed-version-of-a-package/](https://answers.ros.org/question/349261/using-at-the-same-time-a-binary-installed-and-a-source-installed-version-of-a-package/)
[http://wiki.ros.org/catkin/Tutorials/workspace_overlaying](http://wiki.ros.org/catkin/Tutorials/workspace_overlaying)

kind regards,
Javad

matlabbe, 2022-05-10:
Other solution is to use only one rtabmap...

closed 2024-07-01 by adbidwai