introlab/rtabmap_ros #759: RGBD Odometry doesnt receive input

MrOCW, 2022-05-15:
[image]
Hi, attempting to run RGBD odometry but I keep getting:
```
/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rgbd_odometry subscribed to (exact sync):
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info
```
although rostopic hz on the respective topics shows that they are indeed being published

This is the launch file:
```
  <node name="rgbd_odometry" type="rgbd_odometry" pkg="rtabmap_ros">
    <remap from="rgb/image"       to="/camera/color/image_raw"/>
    <remap from="depth/image"     to="/camera/aligned_depth_to_color/image_raw"/>
    <remap from="rgb/camera_info" to="/camera/color/camera_info"/>
    <remap from="rgbd_image"       to="rgbd_image"/>
    <remap from="odom"            to="visual_odom"/>
    <remap from="imu"             to="camera/imu_ros"/>

    <param name="frame_id"                    type="string" value="camera_color_optical_frame" TRIED MANY DIFFERENT LINKS HERE/>
    <param name="odom_frame_id"               type="string" value="visual_odom"/>
    <param name="publish_tf"                  type="bool"   value="true"/>
    <param name="ground_truth_frame_id"       type="string" value=""/>
    <param name="ground_truth_base_frame_id"  type="string" value=""/>
    <param name="wait_for_transform_duration" type="double" value="0.2"/>
    <param name="wait_imu_to_init"            type="bool"   value="true"/>
    <param name="approx_sync"                 type="bool"   value="false"/>
    <param name="approx_sync_max_interval"    type="double" value="0"/>
    <param name="config_path"                 type="string" value=""/>
    <param name="queue_size"                  type="int"    value="10"/>
    <param name="subscribe_rgbd"              type="bool"   value="false"/>
    <param name="guess_frame_id"              type="string" value=""/>
    <param name="guess_min_translation"       type="double" value="0"/>
    <param name="guess_min_rotation"          type="double" value="0"/>
    <param name="expected_update_rate"        type="double" value="0"/>
    <param name="max_update_rate"             type="double" value="0"/>
    <param name="keep_color"                  type="bool"   value="false"/>
    <param name="publish_null_when_lost"      type="bool"   value="false"/>
  </node>
```

matlabbe, 2022-05-15:
The `frame_id` should be the base frame of the robot (e.g., `base_link`).

You can also do:
```
rostopic hz /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info
```
to compare are topic rates at the same time.

If they are indeed all published at 30Hz, it could be that they are not having the exact same stamps, so `approx_sync` could be set to `true`. Note that realsense streams should be exactly synchronized as I remember (looking at those [examples](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping)).

MrOCW, 2022-05-18:
it seems to be a ROS latency issue between host and client. works fine if performed on the edge device

closed 2022-05-18 by MrOCW