introlab/rtabmap_ros #762: Rtabmap performance with zed 2i camera

brayanpa, 2022-05-22:
Hi, I am looking for the best sensor for depth and odometry, I have seen that the ZED 2i promises a lot, is this a good option for rtabmap? If not, what cameras would you recommend?

matlabbe, 2022-09-06:
Is it for indoor or outdoor? Outdoor I would say any stereo cameras can work relatively the same (specs to look at: large FOV and Global shutter, large baseline is preferred if outdoor). For indoor, best for depth are those cameras with TOF sensor (or LiDAR camera like iOS refers too). An example is the Kinect for Azure, which is pretty great. For odometry, I prefer stereo cameras as timsesync between a TOF sensor and a  camera is not always perfectly exact like the left/right cameras of a stereo camera.

brayanpa, 2022-09-14:
Thanks Mathieu. the environment is indoor. 
In the end I selected two realsense D435 cameras and a T265 camera for odometry.

closed 2022-09-14 by brayanpa