introlab/rtabmap_ros #764: odometry: waiting imu (/rtabmap/imu) to initialize orientation (wait_imu_to_init=true)

MymengA, 2022-05-23:
environment: Ubuntu 20.04, ROS noetic / realsense435i / vins-fusion / rtabmap

HELLO！
I want to use VINS-Fusion to provide external /odom input for RTABMAP for mapping.

My command to 
start the camera is: **roslaunch realsense2_camera rs_camera_vins.launch**
`<launch>
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="camera"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>

  <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="infra_width"        default="640"/>
  <arg name="infra_height"       default="480"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="true"/>
  <arg name="enable_accel"        default="true"/>

  <arg name="enable_pointcloud"         default="false"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_COLOR"/>
  <arg name="pointcloud_texture_index"  default="0"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="true"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="-2"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default="linear_interpolation"/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="emitter_enable"   		default="false"/>

<rosparam>
  /camera/stereo_module/emitter_enabled: 0
</rosparam>

<rosparam if="$(arg emitter_enable)">
  /camera/stereo_module/emitter_enabled: 1
[... 169 lines cut ...]
   <arg name="rtabmapviz"        value="$(arg rtabmapviz)"/>
   <arg name="rviz"              value="$(arg rviz)"/>
</include>

<include unless="$(arg depth_mode)" file="$(find rtabmap_ros)/launch/rtabmap.launch">
   <arg name="rtabmap_args"            value="--delete_db_on_start --Optimizer/GravitySigma 0.3 --Odom/Strategy $(arg odom_strategy) --OdomVINS/ConfigPath /home/meng/vins_fusion_ws/src/VINS-Fusion/config/realsense_d435i/realsense_stereo_config.yaml" />
   <arg name="left_image_topic"        value="/camera/infra1/image_rect_raw"/>
   <arg name="right_image_topic"       value="/camera/infra2/image_rect_raw"/>
   <arg name="left_camera_info_topic"  value="/camera/infra1/camera_info"/>
   <arg name="right_camera_info_topic" value="/camera/infra2/camera_info"/>
   <arg name="stereo"                  value="true"/>
   <arg name="frame_id"                value="camera_link"/>
   <arg name="imu_topic"               value="/rtabmap/imu"/>
   
   <arg name="wait_imu_to_init"        value="true"/>
   <arg name="rtabmapviz"              value="$(arg rtabmapviz)"/>
   <arg name="rviz"                    value="$(arg rviz)"/>
</include>

</launch>`

After start RTABMAP
My error is **[ WARN] [1653304384.207397999]: odometry: waiting imu (/rtabmap/imu) to initialize orientation (wait_imu_to_init=true)**

Can you help me solve this problem？

matlabbe, 2022-09-06:
If you are using external VINS-Fusion odometry, you don't need to start rtabmap_ros/stereo_odometry node, just remap vins-fusion output odom topic to rtabmap node. That example is using VINS-fusion internally.

tulingcheng86, 2024-03-20:
> environment: Ubuntu 20.04, ROS noetic / realsense435i / vins-fusion / rtabmap
> 
> HELLO！ I want to use VINS-Fusion to provide external /odom input for RTABMAP for mapping.
> 
> My command to start the camera is: **roslaunch realsense2_camera rs_camera_vins.launch** `
> 
> /camera/stereo_module/emitter_enabled: 0 /camera/stereo_module/emitter_enabled: 1
> ```
>   <arg name="enable_pointcloud"        value="$(arg enable_pointcloud)"/>
>   <arg name="pointcloud_texture_stream" value="$(arg pointcloud_texture_stream)"/>
>   <arg name="pointcloud_texture_index"  value="$(arg pointcloud_texture_index)"/>
>   <arg name="enable_sync"              value="$(arg enable_sync)"/>
>   <arg name="align_depth"              value="$(arg align_depth)"/>
> 
>   <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
>   <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
>   <arg name="enable_fisheye"           value="$(arg enable_fisheye)"/>
> 
>   <arg name="depth_width"              value="$(arg depth_width)"/>
>   <arg name="depth_height"             value="$(arg depth_height)"/>
>   <arg name="enable_depth"             value="$(arg enable_depth)"/>
> 
>   <arg name="color_width"              value="$(arg color_width)"/>
>   <arg name="color_height"             value="$(arg color_height)"/>
>   <arg name="enable_color"             value="$(arg enable_color)"/>
> 
>   <arg name="infra_width"              value="$(arg infra_width)"/>
>   <arg name="infra_height"             value="$(arg infra_height)"/>
>   <arg name="enable_infra1"            value="$(arg enable_infra1)"/>
>   <arg name="enable_infra2"            value="$(arg enable_infra2)"/>
> 
>   <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
>   <arg name="depth_fps"                value="$(arg depth_fps)"/>
>   <arg name="infra_fps"                value="$(arg infra_fps)"/>
>   <arg name="color_fps"                value="$(arg color_fps)"/>
>   <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
>   <arg name="accel_fps"                value="$(arg accel_fps)"/>
>   <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
>   <arg name="enable_accel"             value="$(arg enable_accel)"/>
> 
>   <arg name="publish_tf"               value="$(arg publish_tf)"/>
>   <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>
> 
>   <arg name="filters"                  value="$(arg filters)"/>
>   <arg name="clip_distance"            value="$(arg clip_distance)"/>
>   <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
>   <arg name="initial_reset"            value="$(arg initial_reset)"/>
>   <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
>   <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
>   <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
>   <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
>   <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
> </include>
> ```
> 
> `
> start vins-fusion: **rosrun vins vins_node ~/vins_fusion_ws/src/VINS-Fusion/config/realsense_d435i/realsense_stereo_config.yaml**
> 
> ```
> `
> %YAML:1.0
> #common parameters
> #support: 1 imu 1 cam; 1 imu 2 cam: 2 cam; 
> imu: 0         
> num_of_cam: 2  
> 
> imu_topic: "/camera/imu"
> image0_topic: "/camera/infra1/image_rect_raw"
> image1_topic: "/camera/infra2/image_rect_raw"
> output_path: "/home/dji/output/"
[... 108 lines cut ...]
> <include unless="$(arg depth_mode)" file="$(find rtabmap_ros)/launch/rtabmap.launch">
>    <arg name="rtabmap_args"            value="--delete_db_on_start --Optimizer/GravitySigma 0.3 --Odom/Strategy $(arg odom_strategy) --OdomVINS/ConfigPath /home/meng/vins_fusion_ws/src/VINS-Fusion/config/realsense_d435i/realsense_stereo_config.yaml" />
>    <arg name="left_image_topic"        value="/camera/infra1/image_rect_raw"/>
>    <arg name="right_image_topic"       value="/camera/infra2/image_rect_raw"/>
>    <arg name="left_camera_info_topic"  value="/camera/infra1/camera_info"/>
>    <arg name="right_camera_info_topic" value="/camera/infra2/camera_info"/>
>    <arg name="stereo"                  value="true"/>
>    <arg name="frame_id"                value="camera_link"/>
>    <arg name="imu_topic"               value="/rtabmap/imu"/>
>    
>    <arg name="wait_imu_to_init"        value="true"/>
>    <arg name="rtabmapviz"              value="$(arg rtabmapviz)"/>
>    <arg name="rviz"                    value="$(arg rviz)"/>
> </include>
> 
> </launch>
> ```
> 
> `
> 
> After start RTABMAP My error is **[ WARN] [1653304384.207397999]: odometry: waiting imu (/rtabmap/imu) to initialize orientation (wait_imu_to_init=true)**
> 
> Can you help me solve this problem？

do you slove this problem?