introlab/rtabmap_ros #765: I wanna know where i can get key-frames

kdhRGT, 2022-05-25:
Hello author!

now I am enjoying your program for testing SLAM.

I have a question.

1. where can I get the key-frame node? I found keyframe nodes in "corewrapper.cpp" but I can't find where it comes 

```
void CoreWrapper::publishGlobalPath(const ros::Time & stamp)
{
	if((globalPathPub_.getNumSubscribers() || globalPathNodesPub_.getNumSubscribers()) && rtabmap_.getPath().size())
	{
		Transform pose = uValue(rtabmap_.getLocalOptimizedPoses(), rtabmap_.getPathCurrentGoalId(), Transform());
		if(!pose.isNull() && rtabmap_.getPathCurrentGoalIndex() < rtabmap_.getPath().size())
		{
			// transform the global path in the goal referential
			Transform t = pose * rtabmap_.getPath().at(rtabmap_.getPathCurrentGoalIndex()).second.inverse();

			nav_msgs::Path path;
			rtabmap_ros::Path pathNodes;
			path.header.frame_id = pathNodes.header.frame_id = mapFrameId_;
			path.header.stamp = pathNodes.header.stamp = stamp;
			path.poses.resize(rtabmap_.getPath().size());
			pathNodes.nodeIds.resize(rtabmap_.getPath().size());
			pathNodes.poses.resize(rtabmap_.getPath().size());
			int oi = 0;
			for(std::vector<std::pair<int, Transform> >::const_iterator iter=rtabmap_.getPath().begin(); 
                         iter!=rtabmap_.getPath().end(); ++iter)
			{
				path.poses[oi].header = path.header;
				rtabmap_ros::transformToPoseMsg(t*iter->second, path.poses[oi].pose);
				pathNodes.poses[oi] = path.poses[oi].pose;
				pathNodes.nodeIds[oi] = iter->first;
				++oi;
			}
			Transform goalLocalTransform = Transform::getIdentity();
			if(!goalFrameId_.empty() && goalFrameId_.compare(frameId_) != 0)
			{
				Transform localT = rtabmap_ros::getTransform(frameId_, goalFrameId_, stamp, tfListener_, waitForTransform_?waitForTransformDuration_:0.0);
				if(!localT.isNull())
				{
					goalLocalTransform = localT.inverse().to3DoF();
				}
			}

			if(!rtabmap_.getPathTransformToGoal().isIdentity() || !goalLocalTransform.isIdentity())
			{
				path.poses.resize(path.poses.size()+1);
				path.poses[path.poses.size()-1].header = path.header;
				pathNodes.nodeIds.resize(pathNodes.nodeIds.size()+1);
				pathNodes.poses.resize(pathNodes.poses.size()+1);
				Transform p = t * rtabmap_.getPath().back().second*rtabmap_.getPathTransformToGoal() * goalLocalTransform;
				rtabmap_ros::transformToPoseMsg(p, path.poses[path.poses.size()-1].pose);
				pathNodes.poses[pathNodes.poses.size()-1] = path.poses[path.poses.size()-1].pose;
				pathNodes.nodeIds[pathNodes.nodeIds.size()-1] = 0;
			}
                       globalPathNodesPub_.publish(pathNodes);
			if(globalPathPub_.getNumSubscribers())
			{
				globalPathPub_.publish(path);
			}
			if(globalPathNodesPub_.getNumSubscribers())
			{
				globalPathNodesPub_.publish(pathNodes);
			}
		}
	}
}
```
I wanna know where is getting key-frames from rtabmap! because I wanna know which RGB-D camera image frame is used for key-frame. if I can know which frame is used as key-frame, I will use that frame timestamp!

thank you so much for your effort!

best regards

matlabbe, 2022-09-06:
That global path is used during planning, so I don't think it is what you are looking for. If you open the database with `rtabmap-databaseViewer`, you can see the ID and timestamp of each node in the graph. That could be useful to know which frames have been used for the nodes.

If you want to have this info in ROS from a service, call `rosservice call /rtabmap/get_map_data 1 1 1`, you will get all nodes with ID and corresponding timestamp.

kdhRGT, 2022-10-20:
Thank you so much!

closed 2022-10-20 by kdhRGT

kuacboss, 2024-03-12:
I really enjoyed the answer above! Is there a way to check the coordinates of the map point (point cloud) of the keframe displayed in rtabmap-databaseViewer? Can I check additional matched map points?
[image]

matlabbe, 2024-03-14:
We can mouse over the features in the keyframes to know their 2D pixel coordinate in the image frame (GraphicsView). Based on your suggestion, I [just added](https://github.com/introlab/rtabmap/commit/a6d9425bce9c3e971b9cd9341de9b6490d77c887) a menu option to show the depth and map coordinate of a pixel. The graph view should be opened to get the map coordinate, otherwise only depth is shown if the frame has depth. If the map coordinate is not shown (only depth), move back and forth the image slider to refresh.
[image]

kdhRGT, 2024-03-14:
@matlabbe Thank you so much for your attention to this problem.

Now i am really enjoying the rtabmap it is really good slam algorithm