introlab/rtabmap_ros #770: TUM RGBD dataset on rtabmap_ros showing blank screen

AbhiKhoyani, 2022-06-12:
Hii Mathieu,

I was following rgbdslam_datasets.launch for loading TUM RGBD dataset and followed all the steps mentioned initially in that file and your other responses in issue also but I'm not getting any frames in rviz and rtabmapviz. 
Using this launch file: https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/rgbdslam_datasets.launch

Here, I've uploaded the snapshot of rviz.
[image]

My environment is I'm running Linux on WSL, with ros noetic, python3 and for GUI using Xserver launch.
I tried demo_robot_mapping.launch with demo_mapping.bag and it's working perfectly in rviz as well as rtabmapviz so I'm sure there must not be environment problem. For TUM, I tried it on both of the below file with compressed and decompressed, with and without python renamed world->kinect to world->kinect_gt but no luck.
$ wget http://vision.in.tum.de/rgbd/dataset/freiburg3/rgbd_dataset_freiburg3_long_office_household.bag
$ wget https://vision.in.tum.de/rgbd/dataset/freiburg1/rgbd_dataset_freiburg1_xyz.bag

any hint/help is appreciated.   
-Abhi

matlabbe, 2022-10-09:
I don't know about WSL, but on native ubuntu just re-did the instructions in https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/rgbdslam_datasets.launch and it works as expected:
[image]

SYLJSTK, 2023-02-13:
I have the same situation with you. Have you solved it?

wz8530, 2023-04-28:
Hi, @matlabbe, I have the same problem.
I use this [rgbdslam_datesets.launch](https://github.com/introlab/rtabmap_ros/blob/0.20.23-melodic/launch/tests/rgbdslam_datasets.launch).
[image]

I can confirm that the subscription topic was successful.. Do you have any suggestions to solve this problem?

matlabbe, 2023-04-30:
Are you on noetic or melodic?

On noetic, it seems working following those [instructions](https://github.com/introlab/rtabmap_ros/blob/0bd2190f77a5be191b1cebd438dea4c61d960c99/rtabmap_examples/launch/rgbdslam_datasets.launch#L5-L12):
[image]

wz8530, 2023-05-04:
> Are you on noetic or melodic?
> 
> On noetic, it seems working following those [instructions](https://github.com/introlab/rtabmap_ros/blob/0bd2190f77a5be191b1cebd438dea4c61d960c99/rtabmap_examples/launch/rgbdslam_datasets.launch#L5-L12): [image]

Hi, @matlabbe , I'm sorry for replying so late. I installed `ros-melodic` on my computer. After restarting the computer, it seems to be working normally, which is very strange.
[image]

closed 2023-05-15 by matlabbe