introlab/rtabmap_ros #774: issues running zed2 + rtab ros2

an99990, 2022-06-15:
Hi, thank you for your work.
I am trying to run the mapping using the ZED2 + ROS2, and strangely I am getting issues when i launch using zed2 launch but not zed.launch.py

steps to reproduce 

`ros2 launch rtabmap_ros rtabmap.launch.py     rtabmap_args:="--delete_db_on_start"     rgb_topic:=/zed2/zed_node/rgb/image_rect_color     depth_topic:=/zed2/zed_node/depth/depth_registered     camera_info_topic:=/zed2/zed_node/rgb/camera_info     frame_id:=base_link     approx_sync:=false     wait_imu_to_init:=true     imu_topic:=/zed2/zed_node/imu/data     qos:=1`

`ros2 launch zed_wrapper zed2.launch.py`

```
[rgbd_odometry-1] [ WARN] (2022-06-15 13:37:59.413) MsgConversion.cpp:1729::getTransform() (can transform base_link -> zed2_imu_link?) Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist. canTransform returned after 0.201961 timeout was 0.2. (wait_for_transform=0.200000)
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] [ERROR] [1655314679.618928454] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed2_imu_link" to frame "base_link", TF not available at time 1654803063.837124
[rgbd_odometry-1] [ WARN] (2022-06-15 13:37:59.618) MsgConversion.cpp:1729::getTransform() (can transform base_link -> zed2_imu_link?) Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist. canTransform returned after 0.202198 timeout was 0.2. (wait_for_transform=0.200000)
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[... 43 lines cut ...]
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[rgbd_odometry-1]          at line 156 in /tmp/binarydeb/ros-galactic-tf2-0.17.2/src/buffer_core.cpp
[rgbd_odometry-1] Warning: Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist
[... cut ...]
[rgbd_odometry-1] [ERROR] [1655314680.029478996] [rtabmap.rgbd_odometry]: Could not transform IMU msg from frame "zed2_imu_link" to frame "base_link", TF not available at time 1654803063.971044
[rgbd_odometry-1] [ WARN] (2022-06-15 13:38:00.029) MsgConversion.cpp:1729::getTransform() (can transform base_link -> zed2_imu_link?) Invalid frame ID "base_link" passed to canTransform argument target_frame - frame does not exist. canTransform returned after 0.202002 timeout was 0.2. (wait_for_transform=0.200000)
```

an99990, 2022-06-28:
changed zed2.launch.py line.37 to base_frame = 'base_link'

closed 2022-06-28 by an99990

matlabbe, 2022-10-10:
zed2.launch.py seems using base_link as default https://github.com/stereolabs/zed-ros2-wrapper/blob/ff945337c38442a44a93c2fcb0edf000b2e25f00/zed_wrapper/launch/zed2.launch.py#L33