introlab/rtabmap_ros #775: How to use RTAB-MAP map navigation in ROS2?

miku54, 2022-06-27:
System: Ubuntu20.04 Galactic
The map for RTAB-MAP is actually alredy is integrated with nav2, but how do I use this map in nav2? I tried this scheme https://github.com/introlab/rtabmap_ros/issues/696#issuecomment-1013790139
But still can not achieve the navigation effect.

matlabbe, 2022-10-10:
Just tried again the example here: https://github.com/introlab/rtabmap_ros/issues/696#issuecomment-1013790139

In rviz, click on Nav2 goal, set the goal, then you will see the robot planning to go there:
[image]

miku54, 2022-10-20:
The robot can indeed reach the target point defined by me, but the map of RTAB-MAP is not displayed. Do I need to do other configuration work?

matlabbe, 2022-10-20:
Can you show a screenshot of what you are seeing? The occupancy grid as above is the map from rtabmap.

miku54, 2022-10-24:
The situation when debugging rtabmap navigation is shown in the figure
[image]
[image]
[image]

matlabbe, 2022-10-24:
The gray map seems the map from rtabmap.

miku54, 2022-10-24:
> 灰色地图似乎是来自 rtabmap 的地图。

If the gray is a map from rtabmap, sometimes when I use a real robot to debug, the gray area is small or not, why is this?
btw, is there only a grey map without a 3d stereo map?

matlabbe, 2022-10-24:
To see the 3d point cloud, you need to add MapCloud display in rviz. This should also be the turtlebot config with 3d camera.

If the gray map is small, maybe it is because the sensor doesn't see far. The local grids are also limited to 4 meters range by default. You can ahow a screenshot to see if this is expected.

miku54, 2022-10-24:
> 要查看 3d 点云，需要在 rviz 中添加 MapCloud 显示。这也应该是带有 3d 相机的 turtlebot 配置。
> 
> 如果灰度图很小，可能是因为传感器看的不远。默认情况下，本地网格也限制在 4 米范围内。您可以通过屏幕截图查看这是否符合预期。

You are right, the 3D point cloud of the object is only displayed when I control the robot to walk into the object.
There is another doubt, rtab-map navigation in ros1 can use the 3d point cloud & 2d map saved by rtab-map mapping, but I found that the 3d point cloud saved by rtab-map mapping cannot be loaded in ROS2, this is New feature?

matlabbe, 2022-11-26:
rtabmap library is independent of ROS, so a database created in ROS1 should be usable in ROS2 (or vice-versa)