introlab/rtabmap_ros #779: FATAL error occurs.. related to covariance

jaykorea, 2022-07-17:
Hello.

I failed to apply realsense t265 to my robot cause of drifting data.

So I changed the sensor to D455(Imu inside) with wheel odometry

As I run not localization mode, everything works fine, however in localization mode, Fatal error occurs seeing below.

```
[FATAL] (2022-07-17 22:20:18.216) OptimizerG2O.cpp:958::optimize() Condition (optimizer.verifyInformationMatrices(true)) not met! [This error can be caused by (1) bad covariance matrix set in odometry messages (see requirements in g2o::OptimizableGraph::verifyInformationMatrices() function) or that (2) PCL and g2o hadn't been built both with or without "-march=native" compilation flag (if one library is built with this flag and not the other, this is causing Eigen to not work properly, resulting in segmentation faults).]
terminate called after throwing an instance of 'UException'
  what():  [FATAL] (2022-07-17 22:20:18.216) OptimizerG2O.cpp:958::optimize() Condition (optimizer.verifyInformationMatrices(true)) not met! [This error can be caused by (1) bad covariance matrix set in odometry messages (see requirements in g2o::OptimizableGraph::verifyInformationMatrices() function) or that (2) PCL and g2o hadn't been built both with or without "-march=native" compilation flag (if one library is built with this flag and not the other, this is causing Eigen to not work properly, resulting in segmentation faults).]

```

I'm using robot localization package following the realsense tutorial

my pkg param is like below

```
<launch>

    <include file="$(find realsense2_camera)/launch/d400.launch">
      <arg name="camera" value="d400"/>
      <arg name="serial_no_camera" value="213522254416"/>
      <arg name="initial_reset" value="false"/>
      <arg name="align_depth" value="true"/>
      <arg name="enable_sync" value="true"/>
      <arg name="clip_distance" value="-2"/>
      <arg name="linear_accel_cov" value="1.0"/>
      <arg name="unite_imu_method" value="linear_interpolation"/>
      <arg name="enable_gyro"   value="true"/>
      <arg name="enable_accel"  value="true"/>

      <!-- <arg name="filters" value="disparity,spatial,temporal,decimation"/> -->
    </include>

    <node pkg="imu_filter_madgwick" type="imu_filter_node" name="ImuFilter">
        <param name="use_mag" type="bool" value="false" />
        <param name="publish_tf" type="bool" value="false" />
        <param name="world_frame" type="string" value="enu" />
        <remap from="/imu/data_raw" to="/d400/imu"/>
    </node>

    <include file="$(find robot_localization)/launch/ukf_template.launch"/>
    <param name="/ukf_se/frequency" value="300"/>
    <param name="/ukf_se/base_link_frame" value="base_footprint"/>
    <param name="/ukf_se/odom0" value="odom_md"/>
    <rosparam param="/ukf_se/odom0_config">[true,true,true,
                                            true,true,true,
                                            true,true,true,
                                            true,true,true,
                                            true,true,true]
    </rosparam>
    <param name="/ukf_se/odom0_relative" value="true"/>
    <param name="/ukf_se/odom0_pose_rejection_threshold" value="10000000"/>
    <param name="/ukf_se/odom0_twist_rejection_threshold" value="10000000"/>

    <param name="/ukf_se/imu0" value="/imu/data"/>
    <rosparam param="/ukf_se/imu0_config">[false, false, false,
                                           true,  true,  true,
                                           true,  true,  true,
                                           true,  true,  true,
                                           true,  true,  true]
    </rosparam>
    <param name="/ukf_se/imu0_differential" value="true"/>
    <param name="/ukf_se/imu0_relative" value="false"/>
    <param name="/ukf_se/use_control" value="false"/>
    <!-- <param name="/ukf_se/odom0_config" value="{true,true,true,}"/> -->    

    <!-- <node pkg="tf2_ros" type="static_transform_publisher" name="d400_link_to_base_link" args="0.008 -0.0115 0.059 0 0 0 base_link d400_link"/> -->

</launch>
```
Relating to this problem, I've checked my covariance data.

```
covariance: [0.0006229568214647916, -8.083730846463338e-23, -2.3526478256779684e-27, 7.603975234514498e-32, -1.9686793224112045e-31, 1.0084986900882368e-26, -8.083730846463338e-23, 0.0006229568214647914, 6.593697615547681e-27, 9.816306764103932e-32, -4.042227063101472e-32, -2.9704252787807974e-27, -2.352672574167219e-27, 6.593697567362882e-27, 1.1034054753282324e-06, 3.649317321204718e-33, -4.804595389038422e-33, 2.460199139855491e-28, 2.083755590102779e-33, 4.88594046384911e-32, -4.6055065061564126e-32, 1.0814084149036437e-06, 3.242877461853746e-19, -1.8387080226128324e-21, -1.9686793224112045e-31, -4.042219882283864e-32, 6.731164903353323e-32, 3.242877461854316e-19, 1.0814084149036494e-06, 5.288115321342584e-21, 9.277193333936052e-27, -2.9720030005912395e-27, 2.9662673689168533e-28, -1.838708013874778e-21, 5.2881153300805475e-21, 1.9999397964668092e-09]

```

It does not seem to poor, localization works quite well.

Plz give me some tip to solve this error..

thank you

jaykorea, 2022-07-20:
Solved this problem by downgrade rtabmap version to 0.20.16 

Thank you

matlabbe, 2022-10-15:
As explained in the error message, this error could be also caused by compilation flags:
`OptimizerG2O.cpp:958::optimize() Condition (optimizer.verifyInformationMatrices(true)) not met! [This error can be caused by (1) bad covariance matrix set in odometry messages (see requirements in g2o::OptimizableGraph::verifyInformationMatrices() function) or that (2) PCL and g2o hadn't been built both with or without "-march=native" compilation flag (if one library is built with this flag and not the other, this is causing Eigen to not work properly, resulting in segmentation faults).]`
That fatal error check is also there in 0.20.16, maybe it was a compilation issue? closing the issue.

closed 2022-10-15 by matlabbe

huma0605, 2023-05-27:
I am seeing the same issue. Everything was working fine when localization = false. But when we set it to be true, we saw the same g2o error. Below is my package versions
```
Package: ros-noetic-rtabmap-ros
Version: 0.21.1-4focal.20230416.011814
```
```
Package: ros-noetic-pcl-ros
Version: 1.7.4-1focal.20230216.065759
```