introlab/rtabmap_ros #782: CMakeFiles/rtabmap_common.dir/src/MsgConversion.cpp.o] Error 1

an99990, 2022-07-21:
the latest commit done gives me issues when building rtabmap_ros.

env:
ros2 galactic
rtabmap branch : master
rtabmap_ros: ros2
steps:

colcon build

error :

```
make[2]: *** [CMakeFiles/rtabmap_common.dir/build.make:63: CMakeFiles/rtabmap_common.dir/src/MsgConversion.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:340: CMakeFiles/rtabmap_common.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< rtabmap_ros [28.3s, exited with code 2]
```

matlabbe, 2022-07-22:
What was the error, make sure you have this commit: https://github.com/introlab/rtabmap_ros/commit/f32d80c82cb9983cb36c32d6a554666fb45043c3

an99990, 2022-07-22:
hi @matlabbe , thank you for answering.
I am on the latest commit of branch **ros2**.

```
 rtabmap::CameraModel> >, std::less<int>, std::allocator<std::pair<const int, rtabmap::CameraModel> > >::const_iterator'} to 'const std::error_code&'
  398 |   operator!=(const error_code& __lhs, const error_code& __rhs) noexcept
      |              ~~~~~~~~~~~~~~~~~~^~~~~
/opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/system_error:404:3: note: candidate: 'bool std::operator!=(const std::error_code&, const std::error_condition&)'
  404 |   operator!=(const error_code& __lhs, const error_condition& __rhs) noexcept
      |   ^~~~~~~~
/opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/system_error:404:32: note:   no known conversion for argument 1 from 'std::map<int, rtabmap::CameraModel>::const_iterator' {aka 'std::_Rb_tree<int, std::pair<const int, rtabmap::CameraModel>, std::_Select1st<std::pair<const int, rtabmap::CameraModel> >, std::less<int>, std::allocator<std::pair<const int, rtabmap::CameraModel> > >::const_iterator'} to 'const std::error_code&'
  404 |   operator!=(const error_code& __lhs, const error_condition& __rhs) noexcept
      |              ~~~~~~~~~~~~~~~~~~^~~~~
/opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/system_error:410:3: note: candidate: 'bool std::operator!=(const std::error_condition&, const std::error_code&)'
  410 |   operator!=(const error_condition& __lhs, const error_code& __rhs) noexcept
      |   ^~~~~~~~
/opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/system_error:410:37: note:   no known conversion for argument 1 from 'std::map<int, rtabmap::CameraModel>::const_iterator' {aka 'std::_Rb_tree<int, std::pair<const int, rtabmap::CameraModel>, std::_Select1st<std::pair<const int, rtabmap::CameraModel> >, std::less<int>, std::allocator<std::pair<const int, rtabmap::CameraModel> > >::const_iterator'} to 'const std::error_condition&'
  410 |   operator!=(const error_condition& __lhs, const error_code& __rhs) noexcept
      |              ~~~~~~~~~~~~~~~~~~~~~~~^~~~~
/opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/system_error:415:3: note: candidate: 'bool std::operator!=(const std::error_condition&, const std::error_condition&)'
  415 |   operator!=(const error_condition& __lhs,
      |   ^~~~~~~~
/opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/system_error:415:37: note:   no known conversion for argument 1 from 'std::map<int, rtabmap::CameraModel>::const_iterator' {aka 'std::_Rb_tree<int, std::pair<const int, rtabmap::CameraModel>, std::_Select1st<std::pair<const int, rtabmap::CameraModel> >, std::less<int>, std::allocator<std::pair<const int, rtabmap::CameraModel> > >::const_iterator'} to 'const std::error_condition&'
  415 |   operator!=(const error_condition& __lhs,
      |              ~~~~~~~~~~~~~~~~~~~~~~~^~~~~
In file included from /opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/bits/locale_facets.h:48,
                 from /opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/locale:40,
                 from /opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/iomanip:43,
                 from /opt/conda/include/rosidl_runtime_cpp/traits.hpp:19,
                 from /opt/conda/include/builtin_interfaces/msg/detail/time__traits.hpp:10,
                 from /opt/conda/include/builtin_interfaces/msg/time.hpp:9,
                 from /opt/conda/include/rclcpp/time.hpp:18,
                 from /ros_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:31,
                 from /ros_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/bits/streambuf_iterator.h:233:5: note: candidate: 'template<class _CharT, class _Traits> bool std::operator!=(const std::istreambuf_iterator<_CharT, _Traits>&, const std::istreambuf_iterator<_CharT, _Traits>&)'
  233 |     operator!=(const istreambuf_iterator<_CharT, _Traits>& __a,
      |     ^~~~~~~~
/opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/bits/streambuf_iterator.h:233:5: note:   template argument deduction/substitution failed:
/ros_ws/src/rtabmap_ros/src/MsgConversion.cpp:1552:36: note:   'std::map<int, rtabmap::CameraModel>::const_iterator' {aka 'std::_Rb_tree<int, std::pair<const int, rtabmap::CameraModel>, std::_Select1st<std::pair<const int, rtabmap::CameraModel> >, std::less<int>, std::allocator<std::pair<const int, rtabmap::CameraModel> > >::const_iterator'} is not derived from 'const std::istreambuf_iterator<_CharT, _Traits>'
 1552 |   iter!=info.localBundleModels.end();
      |                                    ^
In file included from /opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/future:39,
                 from /opt/conda/include/tf2_ros/async_buffer_interface.h:40,
                 from /opt/conda/include/tf2_ros/buffer.h:35,
                 from /ros_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:32,
                 from /ros_ws/src/rtabmap_ros/src/MsgConversion.cpp:28:
/opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/thread:311:3: note: candidate: 'bool std::operator!=(std::thread::id, std::thread::id)'
  311 |   operator!=(thread::id __x, thread::id __y) noexcept
      |   ^~~~~~~~
/opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/thread:311:25: note:   no known conversion for argument 1 from 'std::map<int, rtabmap::CameraModel>::const_iterator' {aka 'std::_Rb_tree<int, std::pair<const int, rtabmap::CameraModel>, std::_Select1st<std::pair<const int, rtabmap::CameraModel> >, std::less<int>, std::allocator<std::p [... cut ...]
/opt/conda/x86_64-conda-linux-gnu/include/c++/10.3.0/bits/stl_map.h:1508:5: note:   template argument deduction/substitution failed:
/ros_ws/src/rtabmap_ros/src/MsgConversion.cpp:1552:36: note:   'std::map<int, rtabmap::CameraModel>::const_iterator' {aka 'std::_Rb_tree<int, std::pair<const int, rtabmap::CameraModel>, std::_Select1st<std::pair<const int, rtabmap::CameraModel> >, std::less<int>, std::allocator<std::pair<const int, rtabmap::CameraModel> > >::const_iterator'} is not derived from 'const std::map<_Key, _Tp, _Compare, _Allocator>'
 1552 |   iter!=info.localBundleModels.end();
      |                       ...
```

matlabbe, 2022-07-22:
Did you update the code? It looks like you are still on this version:
https://github.com/introlab/rtabmap_ros/blob/df7ee6ee8d06e7a6d64ce249a29c646c50540071/src/MsgConversion.cpp#L1552

and not this one:
https://github.com/introlab/rtabmap_ros/blob/f32d80c82cb9983cb36c32d6a554666fb45043c3/src/MsgConversion.cpp#L1552

for line 1552

matlabbe, 2022-07-22:
I triggered again the last ros2 commit just to make sure: https://github.com/introlab/rtabmap_ros/runs/7467723866?check_suite_focus=true

an99990, 2022-07-22:
so im currently trying to get rtabmap build inside a docker. A fresh build outside the docker works fine. but inside the docker i get this message : 

```
                 from /ros_ws/src/rtabmap_ros/src/rviz/MapGraphDisplay.cpp:37:
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:1744:26: warning: extra ';' [-Wpedantic]
 1744 |         PCL_ADD_UNION_RGB;
      |                          ^
      |                          -
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:1743:7: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
 1743 |       {
      |       ^
In file included from /opt/conda/include/pcl-1.11/pcl/point_types.h:349,
                 from /ros_ws/install/rtabmap/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/Signature.h:32,
                 from /ros_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:47,
                 from /ros_ws/src/rtabmap_ros/src/rviz/MapGraphDisplay.cpp:37:
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:195:12: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  195 |     struct { \
      |            ^
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:213:3: note: in expansion of macro 'PCL_ADD_UNION_POINT4D'
  213 |   PCL_ADD_UNION_POINT4D \
      |   ^~~~~~~~~~~~~~~~~~~~~
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:1787:5: note: in expansion of macro 'PCL_ADD_POINT4D'
 1787 |     PCL_ADD_POINT4D;
      |     ^~~~~~~~~~~~~~~
In file included from /opt/conda/include/pcl-1.11/pcl/point_types.h:349,
                 from /ros_ws/install/rtabmap/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/Signature.h:32,
                 from /ros_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:47,
                 from /ros_ws/src/rtabmap_ros/src/rviz/InfoDisplay.cpp:29:
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:195:12: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  195 |     struct { \
      |            ^
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:213:3: note: in expansion of macro 'PCL_ADD_UNION_POINT4D'
  213 |   PCL_ADD_UNION_POINT4D \
      |   ^~~~~~~~~~~~~~~~~~~~~
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:291:5: note: in expansion of macro 'PCL_ADD_POINT4D'
  291 |     PCL_ADD_POINT4D; // This adds the members x,y,z which can also be accessed using the point (which is float[4])
      |     ^~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/rtabmap_plugins.dir/build.make:118: CMakeFiles/rtabmap_plugins.dir/src/nodelets/point_cloud_xyzrgb.cpp.o] Error 1
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:243:7: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  243 |       { \
      |       ^
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:267:3: note: in expansion of macro 'PCL_ADD_UNION_RGB'
  267 |   PCL_ADD_UNION_RGB \
      |   ^~~~~~~~~~~~~~~~~
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:322:5: note: in expansion of macro 'PCL_ADD_RGB'
  322 |     PCL_ADD_RGB;
      |     ^~~~~~~~~~~
In file included from /opt/conda/include/pcl-1.11/pcl/point_types.h:349,
                 from /ros_ws/install/rtabmap/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/Signature.h:32,
                 from /ros_ws/src/rtabmap_ros/include/rtabmap_ros/MsgConversion.h:47,
                 from /ros_ws/src/rtabmap_ros/src/rviz/InfoDisplay.cpp:29:
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:272:5: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  272 |     { \
      |     ^
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:365:5: note: in expansion of macro 'PCL_ADD_INTENSITY'
  365 |     PCL_ADD_INTENSITY;
      |     ^~~~~~~~~~~~~~~~~
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:365:22: warning: extra ';' [-Wpedantic]
  365 |     PCL_ADD_INTENSITY;
      |                      ^
      |                      -
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:278:5: warning: ISO C++ prohibits anonymous structs [-Wpedantic]
  278 |     { \
      |     ^
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:391:5: note: in expansion of macro 'PCL_ADD_INTENSITY_8U'
  391 |     PCL_ADD_INTENSITY_8U;
      |     ^~~~~~~~~~~~~~~~~~~~
/opt/conda/include/pcl-1.11/pcl/impl/point_types.hpp:391:25: warning: extra ';' [-Wpedantic]
  391 |     PCL_ADD_INTENSITY_8U;
      |                         ^
[... 405 lines cut ...]
/opt/conda/include/OGRE/OgreVector3.h:2:62: note: '#pragma message: /opt/conda/include/OGRE/OgreVector3.h is deprecated, migrate to Ogre.h'
    2 | #pragma message( __FILE__ " is deprecated, migrate to Ogre.h")
      |                                                              ^
make[2]: *** [CMakeFiles/rtabmap_plugins.dir/build.make:76: CMakeFiles/rtabmap_plugins.dir/src/nodelets/rgbd_odometry.cpp.o] Error 1
make[2]: *** [CMakeFiles/rtabmap_sync.dir/build.make:146: CMakeFiles/rtabmap_sync.dir/src/impl/CommonDataSubscriberRGBDX.cpp.o] Error 1
In file included from /opt/conda/include/rviz_common/frame_manager_iface.hpp:45,
                 from /opt/conda/include/rviz_common/display_context.hpp:42,
                 from /opt/conda/include/rviz_common/ros_topic_display.hpp:46,
                 from /opt/conda/include/rviz_common/message_filter_display.hpp:38,
                 from /ros_ws/build/rtabmap_ros/src/rviz/../../../../src/rtabmap_ros/src/rviz/InfoDisplay.h:41,
                 from /ros_ws/build/rtabmap_ros/src/rviz/moc_InfoDisplay.cpp:9:
[... cut ...]
make[2]: *** [CMakeFiles/rtabmap_sync.dir/build.make:90: CMakeFiles/rtabmap_sync.dir/src/impl/CommonDataSubscriberDepth.cpp.o] Error 1
make[2]: *** [CMakeFiles/rtabmap_sync.dir/build.make:188: CMakeFiles/rtabmap_sync.dir/src/CoreWrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:960: CMakeFiles/rtabmap_sync.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:662: CMakeFiles/rtabmap_plugins.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< rtabmap_ros [1min 38s, exited with code 2]
```

matlabbe, 2022-07-25:
Are there multiple version of PCL installed in docker? It seems using headers from `/opt/conda/include/pcl-1.11`. However, what you shown are mostly warnings, not errors. You may try to find the error in the log.

an99990, 2022-07-26:
when launching rtabmap inside the docker, i am unable to see the pose graph. Could this be related to PCL ?
Also my goal is to see the tag_detections, i have followed that you mentioned in this forum to "trick" the tag_detection. http://official-rtab-map-forum.206.s1.nabble.com/Using-RTAB-Map-for-graph-based-mapping-td8431.html#a8436 

Now i would like to see the tags/landmarks in the graph, theres a landmark topic but it doesnt publish anythin. I was able to publish to tag_detection, how can i see the tags/landmarks ?

please help with this inquire 
thank you @matlabbe

an99990, 2022-07-27:
hi i was ust able to see a landmark !
What is the labels topic ? how is it different from the landmarks ?

thank you !

matlabbe, 2022-07-27:
Labels are attached to an existing node in the graph. They can be useful to send goals afterwards without having to know the id of a node. See this video for example (at 0:37): https://www.youtube.com/watch?v=A487ybS7E4E&t=37s

Landmarks are unique features in the environment (e.g., apriltag, aruco tags), they are used for loop closure detection / re-localization and graph optimization.

an99990, 2022-07-27:
hi thank you for your answer.
Is is possible to add labels to landmarks ? I see in the video that you manually add the label via a ros service. Is there a way to label the landmark, for example changing the landmarks msg type to add a string field for the label ?

thank you

matlabbe, 2022-07-28:
One way I did it for another project is to keep a "map" between landmark id and a label. You could subscribe to mapGraph topic to get landmark ids with their position, then have a node republishing the corresponding labels at the landmark locations. What are the landmarks you are using?

an99990, 2022-07-28:
to add semantic information to a map. 
thank you very much for your answers !

matlabbe, 2022-07-28:
landmarks are more used for graph optimization and re-location. For semantic or general information to add to the map, you may use the user data approach (you can store anything in nodes): http://official-rtab-map-forum.206.s1.nabble.com/Detected-Object-Location-Mark-on-RTABMAP-tp4571p4594.html

an99990, 2022-07-29:
Is the userdata approach still add node to the pose graph to help loop closure and graph optimization ? Can both approach be used or would it be futile ?

matlabbe, 2022-11-27:
The userdata approach is not used to help loop closure detection of graph optimization, it is just data attached to graph.