introlab/rtabmap_ros #783: rtabmap_ros catkin_make error in ROS NEOTIC

gowtham120, 2022-07-22:
Hi everyone,
I have a building problem when build the last one step for the source, in the Install RTAB-Map ros-pkg in your src folder of your Catkin workspace. The command lines are:

followed these steps for installing rtabmap libraries from https://github.com/introlab/rtabmap_ros#build-from-source (STEP-2)

$ cd ~
$ git clone https://github.com/introlab/rtabmap.git rtabmap
$ cd rtabmap/build
$ cmake ..  [<---double dots included]
$ make -j6
$ sudo make install

i installed the rtabmap libraries inside the ~/catkin_ws/devel as mentioned in #118  

while doing catkin make -j4 or -j1 like below i am getting an error (STEP-3)

$ cd ~/catkin_ws
$ git clone https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
$ catkin_make -j4

THIS IS THE ERROR THAT I AM GETTING

make[2]: *** No rule to make target '/opt/ros/noetic/lib/x86_64-linux-gnu/librtabmap_core.so', needed by '/home/gowtham/catkin_ws/devel/lib/rtabmap_ros/stereo_sync'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:18221: rtabmap_ros/CMakeFiles/rtabmap_stereo_sync.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j1" failed

thanks in advance

matlabbe, 2022-07-22:
Remove binaries and rebuild: `sudo apt remove ros-noetic-rtabmap`

gowtham120, 2022-07-23:
> Remove binaries and rebuild: `sudo apt remove ros-noetic-rtabmap`

i executed the above command and make catkin_make -j1, and i got this 

make[2]: *** No rule to make target '/opt/ros/noetic/lib/x86_64-linux-gnu/librtabmap_core.so', needed by '/home/gowtham/catkin_ws/devel/lib/rtabmap_ros/stereo_sync'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:18221: rtabmap_ros/CMakeFiles/rtabmap_stereo_sync.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j1" failed

gowtham120, 2022-07-23:
@matlabbe  i should do the step 2 from here https://github.com/introlab/rtabmap_ros#build-from-source  
                    inside the /home folder or inside the ~/catkin_ws/devel ?

matlabbe, 2022-07-25:
Step 2 you should clone outside your catkin workspace, make/install it like any other applications. You can install it in `~/catkin_ws/devel` if you want. For the error, I would clean the build directory of catkin to start fresh: `rm -r ~/catkin_ws/build`

gowtham120, 2022-07-26:
@matlabbe  thank you for your help and support, This didn't work for me, But i found one of you comment for the exact problem( #91 )  where you asked to remove the CMakeCache.txt from catkin_ws/build directory, then catkin_make again. and that worked for me 

thanks again

we also looking forward for support for another error in mapping and navigation on rtabmap with tb3 and intel realsense d435i #785 

> Step 2 you should clone outside your catkin workspace, make/install it like any other applications. You can install it in `~/catkin_ws/devel` if you want. For the error, I would clean the build directory of catkin to start fresh: `rm -r ~/catkin_ws/build`