introlab/rtabmap_ros #784: How to build rtabmap_ros with octomap?

adbidwai, 2022-07-22:
Hi,

[This](https://www.google.com/search?channel=fs&client=ubuntu&q=rtabmpa_ros+) guide says that the topics pertaining to octomap like `octomap_full`, `octomap_binary`, etc. are only available with I have `rtabmap_ros` build with `octomap`. How exactly am I supposed to do this? In the documentation of rtabmap_ros I could not find a way to build it from soruce with octomap.

Can anyone help me if I am missing anything?

Thanks and cheers!

KhairulM, 2022-09-07:
you also need to install octomap and octomap-mapping, something like this
```bash
sudo apt install ros-$ROS-DISTRO-octomap ros-$ROS-DISTRO-octomap-mapping
```
then try to build rtabmap again

matlabbe, 2022-09-07:
Normally those lines in rtabmap_ros build [instructions](https://github.com/introlab/rtabmap_ros#build-from-source):
```
$ sudo apt install ros-$ROS_DISTRO-rtabmap ros-$ROS_DISTRO-rtabmap-ros
$ sudo apt remove ros-$ROS_DISTRO-rtabmap ros-$ROS_DISTRO-rtabmap-ros
```
should install octomap. Octomap dependency is already included in rtabmap binaries if you don't want to rebuild rtabmap.

closed 2022-09-07 by adbidwai