introlab/rtabmap_ros #785: Rtabros error While navigation with turtlebo3 waffle and d435i

javidaslam, 2022-07-22:
hello ROS developers,

i am following this tutorial to perform mapping and navigation with turtlebot3 using this documentation from here https://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot  

but now i removed the depth camera in tb3 and added intel realsense depth camera 435i and started mapping i am getting an error i have no idea and also i installed open cv in my pc 

please help,

THIS IS THE ERROR I AM GETTING 

[ INFO] [1658489832.502982799, 58.821000000]: 
/rtabmap/rtabmap subscribed to (approx sync):
   /odom \
   /rtabmap/rgbd_image \
   /scan
[ WARN] [1658489832.860281056, 58.961000000]: The time difference between rgb and depth frames is high (diff=0.052000s, rgb=58.783000s, depth=58.835000s). You may want to set approx_sync_max_interval lower than 0.01s to reject spurious bad synchronizations or use approx_sync=false if streams have all the exact same timestamp.
[ INFO] [1658489833.022358893, 59.023000000]: rtabmap 0.20.19 started...
[ WARN] [1658489833.226275458, 59.154000000]: The time difference between rgb and depth frames is high (diff=0.042000s, rgb=58.960000s, depth=59.002000s). You may want to set approx_sync_max_interval lower than 0.01s to reject spurious bad synchronizations or use approx_sync=false if streams have all the exact same timestamp.

[ INFO] [1658489833.243801686, 59.164000000]: rtabmapviz: Reading parameters from the ROS server...

[FATAL] (2022-07-22 17:07:13.315) Memory.cpp:4460::createSignature() Condition (data.depthOrRightRaw().empty() || ( ( data.depthOrRightRaw().type() == CV_16UC1 || data.depthOrRightRaw().type() == CV_32FC1 || data.depthOrRightRaw().type() == CV_8UC1) && ( (data.imageRaw().empty() && data.depthOrRightRaw().type() != CV_8UC1) || (data.depthOrRightRaw().rows <= data.imageRaw().rows && data.depthOrRightRaw().cols <= data.imageRaw().cols)))) not met! [image=(640/480, type=16, [accepted=0,16]) depth=(1280/720, type=2 [accepted=2(depth mm),5(depth m),0(stereo)]). For stereo, left and right images should be same size. For RGB-D, depth can be X times smaller than RGB (where X is an integer).]
terminate called after throwing an instance of 'UException'

  what():  [FATAL] (2022-07-22 17:07:13.315) Memory.cpp:4460::createSignature() Condition (data.depthOrRightRaw().empty() || ( ( data.depthOrRightRaw().type() == CV_16UC1 || data.depthOrRightRaw().type() == CV_32FC1 || data.depthOrRightRaw().type() == CV_8UC1) && ( (data.imageRaw().empty() && data.depthOrRightRaw().type() != CV_8UC1) || (data.depthOrRightRaw().rows <= data.imageRaw().rows && data.depthOrRightRaw().cols <= data.imageRaw().cols)))) not met! [image=(640/480, type=16, [accepted=0,16]) depth=(1280/720, type=2 [accepted=2(depth mm),5(depth m),0(stereo)]). For stereo, left and right images should be same size. For RGB-D, depth can be X times smaller than RGB (where X is an integer).]

[ WARN] [1658489833.649045516, 59.378000000]: The time difference between rgb and depth frames is high (diff=0.022000s, rgb=59.153000s, depth=59.175000s). You may want to set approx_sync_max_interval lower than 0.01s to reject spurious bad synchronizations or use approx_sync=false if streams have all the exact same timestamp.
[rtabmap/rtabmap-3] process has died [pid 7373, exit code -6, cmd /home/javid/catkin_ws/devel/lib/rtabmap_ros/rtabmap -d move_base:=/move_base scan:=/scan odom:=/odom rgbd_image:=rgbd_image grid_map:=/map __name:=rtabmap __log:=/home/javid/.ros/log/48b0bb06-0998-11ed-94ed-a4db30b041fa/rtabmap-rtabmap-3.log].

log file: /home/javid/.ros/log/48b0bb06-0998-11ed-94ed-a4db30b041fa/rtabmap-rtabmap-3*.log

[ INFO] [1658489834.290566258, 59.857000000]: rtabmapviz: Parameters read = 356
[ INFO] [1658489834.290607591, 59.857000000]: rtabmapviz: Parameters successfully read.
[ INFO] [1658489836.472156054, 61.620000000]: /rtabmap/rtabmapviz: subscribe_depth = false
[ INFO] [1658489836.472197651, 61.620000000]: /rtabmap/rtabmapviz: subscribe_rgb = false
[ INFO] [1658489836.472215946, 61.620000000]: /rtabmap/rtabmapviz: subscribe_stereo = false
[ INFO] [1658489836.472237622, 61.620000000]: /rtabmap/rtabmapviz: subscribe_rgbd = false (rgbd_cameras=1)
[ INFO] [1658489836.472257174, 61.620000000]: /rtabmap/rtabmapviz: subscribe_odom_info = false
[ INFO] [1658489836.472273562, 61.620000000]: /rtabmap/rtabmapviz: subscribe_user_data = false
[ INFO] [1658489836.472304763, 61.620000000]: /rtabmap/rtabmapviz: subscribe_scan = true
[ INFO] [1658489836.472328850, 61.620000000]: /rtabmap/rtabmapviz: subscribe_scan_cloud = false
[ INFO] [1658489836.472351726, 61.620000000]: /rtabmap/rtabmapviz: subscribe_scan_descriptor = false
[ INFO] [1658489836.472373644, 61.620000000]: /rtabmap/rtabmapviz: queue_size    = 10
[ INFO] [1658489836.472390883, 61.620000000]: /rtabmap/rtabmapviz: approx_sync   = true
[ INFO] [1658489836.472435033, 61.620000000]: Setup scan callback
[ INFO] [1658489836.484427310, 61.634000000]: 
/rtabmap/rtabmapviz subscribed to (approx sync):
   /odom \
   /scan
[ INFO] [1658489836.484552546, 61.634000000]: rtabmapviz started.
[ WARN] [1658489836.758219940, 61.803000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.
[ WARN] [1658489842.846704810, 66.833000000]: Timed out waiting for transform from base_footprint to map to become available before running costmap, tf error: canTransform: target_frame map does not exist.. canTransform returned after 0.1 timeout was 0.1.

PLEASE HELP

jaykorea, 2022-07-22:
It often happens because of approx_sync param

read about approx_sync param description on rtabmap ros wiki 

Or, It could happen cause of bad tf tree transformation of your sensor link.

Be sure that, Rtabmap requires specific frame coordination

Thx

hariharan382, 2022-07-25:
tf seems to be fine for me I could see every tf except the map(which will be published by rtabmap), I tried approx sync true / false as well same error

hariharan382, 2022-07-25:
my topic hz is 
======================================================================
/odom                 30.0    0.0         0.091       0.01783   50000 
/rtabmap/rgbd_image   19.69   0.002       330.8       1.48      50000 
/scan                 29.41   0.0         0.089       0.01748   50000

gowtham120, 2022-07-27:
> hello ROS developers,
>
> i am following this tutorial to perform mapping and navigation with turtlebot3 using this documentation from here [...]

check whether the depth image resolution is smaller than rgb image

matlabbe, 2022-07-27:
Your depth image is not registered to RGB image. RGB is 640/480 and depth is 1280/720. Do you launch the D435i with `aligned_depth:=true` ?
```
roslaunch realsense2_camera rs_camera.launch align_depth:=true
```
Normally RGB and depth should have the same resolution. However, we accept lower depth resolution while being still registered to color camera, but just decimated (like half or quarter the size of RGB frame).