introlab/rtabmap_ros #786: rtabmap odom frame does not transform to ned frame

jaykorea, 2022-07-22:
Hello, developers

I've checked out that rtabmap odometry source has ENU coordination frame.

For my ekf algorithm to integration rtabmap odom data with my own wheel odom, ENU to NED frame transformation is needed.

I set base_frame to my robot base_footprint set NED frame, however rtabmap does not provide transformed odom data to NED.

How can I get rtabmap NED coordination data to source ekf package?

Thanks

matlabbe, 2022-07-28:
Hi, you may need to do what mavros does when converting ENU to NED for the px4. I didn't search their code, but I know they do that conversion somewhere.

jaykorea, 2022-08-06:
> 안녕하세요, px4에 대해 ENU를 NED로 변환할 때 mavros가 수행하는 작업을 수행해야 할 수도 있습니다. 나는 그들의 코드를 검색하지 않았지만 그들이 어딘가에서 그 변환을 수행한다는 것을 알고 있습니다.

I've found out that I should call rtabmap service like below

rosserivce call rtabmap/reste_odom_to_pose 0.0 0.0. 0.0 0.0 0.0 6.28

to align the rtabmap odometry to robot_localization odometry pose

Should I change imu frame to set rtabmap odometry frame to north west up frame?

matlabbe, 2022-11-06:
Normally, you should not have to make sure that all odometry inputs are started in same frame. When using multiple odomery input sources, I think robot_localization recommends to fuse velocities and not position/orientation, or use `_differential` mode: http://docs.ros.org/en/noetic/api/robot_localization/html/configuring_robot_localization.html#the-differential-parameters