introlab/rtabmap_ros #787: RTABMAP does not subscribe to /fiducial_transforms

MJA0111, 2022-07-23:
Hi,

Thanks for nice replies in the previous issues. I am rerunning the launch files and bag files that I mentioned [here](http://official-rtab-map-forum.206.s1.nabble.com/How-to-improve-mapping-accuracy-based-on-ArUco-identification-code-td8393.html#a8497).

But surprisingly, rtabmap does not subscribe to /fiducial_transforms topic. I reinstalled rtabmap from both source and binary, and I also tried other PCs, but it didn't work out. I would appreciate it if you could help me to find the bug.

I mentioned below the command line's result that shows there is not subscription to the topic:

-------------
rosnode info /rtabmap/rtabmap
Node [/rtabmap/rtabmap]
Publications: 
 * /rosout [rosgraph_msgs/Log]
 * /rtabmap/cloud_ground [sensor_msgs/PointCloud2]
 * /rtabmap/cloud_map [sensor_msgs/PointCloud2]
 * /rtabmap/cloud_obstacles [sensor_msgs/PointCloud2]
 * /rtabmap/global_path [nav_msgs/Path]
 * /rtabmap/global_path_nodes [rtabmap_ros/Path]
 * /rtabmap/goal_out [geometry_msgs/PoseStamped]
 * /rtabmap/goal_reached [std_msgs/Bool]
 * /rtabmap/grid_map [nav_msgs/OccupancyGrid]
 * /rtabmap/grid_prob_map [nav_msgs/OccupancyGrid]
 * /rtabmap/info [rtabmap_ros/Info]
 * /rtabmap/labels [visualization_msgs/MarkerArray]
 * /rtabmap/landmarks [geometry_msgs/PoseArray]
 * /rtabmap/local_grid_empty [sensor_msgs/PointCloud2]
 * /rtabmap/local_grid_ground [sensor_msgs/PointCloud2]
 * /rtabmap/local_grid_obstacle [sensor_msgs/PointCloud2]
 * /rtabmap/local_path [nav_msgs/Path]
 * /rtabmap/local_path_nodes [rtabmap_ros/Path]
 * /rtabmap/localization_pose [geometry_msgs/PoseWithCovarianceStamped]
 * /rtabmap/mapData [rtabmap_ros/MapData]
 * /rtabmap/mapGraph [rtabmap_ros/MapGraph]
 * /rtabmap/mapOdomCache [rtabmap_ros/MapGraph]
 * /rtabmap/mapPath [nav_msgs/Path]
 * /rtabmap/octomap_binary [octomap_msgs/Octomap]
 * /rtabmap/octomap_empty_space [sensor_msgs/PointCloud2]
 * /rtabmap/octomap_full [octomap_msgs/Octomap]
 * /rtabmap/octomap_global_frontier_space [sensor_msgs/PointCloud2]
 * /rtabmap/octomap_grid [nav_msgs/OccupancyGrid]
 * /rtabmap/octomap_ground [sensor_msgs/PointCloud2]
 * /rtabmap/octomap_obstacles [sensor_msgs/PointCloud2]
 * /rtabmap/octomap_occupied_space [sensor_msgs/PointCloud2]
 * /rtabmap/proj_map [nav_msgs/OccupancyGrid]
 * /rtabmap/scan_map [sensor_msgs/PointCloud2]
 * /tf [tf2_msgs/TFMessage]

Subscriptions: 
 * /clock [unknown type]
 * /rtabmap/global_pose [unknown type]
 * /rtabmap/goal [unknown type]
 * /rtabmap/goal_node [unknown type]
 * /rtabmap/gps/fix [unknown type]
 * /rtabmap/imu [unknown type]
 * /rtabmap/initialpose [unknown type]
 * /rtabmap/odom [nav_msgs/Odometry]
 * /rtabmap/republish_node_data [unknown type]
 * /rtabmap/rgbd_image [rtabmap_ros/RGBDImage]
 * /rtabmap/tag_detections [unknown type]
 * /rtabmap/user_data_async [unknown type]
 * /tf [tf2_msgs/TFMessage]
 * /tf_static [tf2_msgs/TFMessage]

Services: 
 * /rtabmap/add_link
 * /rtabmap/backup
 * /rtabmap/cancel_goal
 * /rtabmap/cleanup_local_grids
 * /rtabmap/detect_more_loop_closures
 * /rtabmap/get_grid_map
 * /rtabmap/get_map
 * /rtabmap/get_map_data
 * /rtabmap/get_map_data2
 * /rtabmap/get_node_data
 * /rtabmap/get_nodes_in_radius
 * /rtabmap/get_plan
 * /rtabmap/get_plan_nodes
 * /rtabmap/get_prob_map
 * /rtabmap/get_proj_map
 * /rtabmap/global_bundle_adjustment
 * /rtabmap/list_labels
 * /rtabmap/load_database
 * /rtabmap/octomap_binary
 * /rtabmap/octomap_full
 * /rtabmap/pause
 * /rtabmap/publish_map
 * /rtabmap/remove_label
 * /rtabmap/reset
 * /rtabmap/resume
 * /rtabmap/rtabmap/get_loggers
 * /rtabmap/rtabmap/list
 * /rtabmap/rtabmap/load_nodelet
 * /rtabmap/rtabmap/log_debug
 * /rtabmap/rtabmap/log_error
 * /rtabmap/rtabmap/log_info
 * /rtabmap/rtabmap/log_warning
 * /rtabmap/rtabmap/set_logger_level
 * /rtabmap/rtabmap/unload_nodelet
 * /rtabmap/set_goal
 * /rtabmap/set_label
 * /rtabmap/set_mode_localization
 * /rtabmap/set_mode_mapping
 * /rtabmap/trigger_new_map
 * /rtabmap/update_parameters

contacting node http://Javad:36341/ ...
Pid: 11638
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound (34059 - 127.0.0.1:41966) [16]
    * transport: TCPROS
 * topic: /tf
    * to: /rtabmap/rtabmap
    * direction: outbound
    * transport: INTRAPROCESS
 * topic: /tf
    * to: /rtabmap/rgbd_odometry
    * direction: outbound (34059 - 127.0.0.1:41994) [17]
    * transport: TCPROS
 * topic: /tf
    * to: /rtabmap/rviz
    * direction: outbound (34059 - 127.0.0.1:41996) [18]
    * transport: TCPROS
 * topic: /rtabmap/info
    * to: /rtabmap/rviz
    * direction: outbound (34059 - 127.0.0.1:42036) [23]
    * transport: TCPROS
 * topic: /rtabmap/mapGraph
    * to: /rtabmap/rviz
    * direction: outbound (34059 - 127.0.0.1:42038) [20]
    * transport: TCPROS
 * topic: /tf
    * to: /rtabmap/rtabmap (http://Javad:36341/)
    * direction: inbound
    * transport: INTRAPROCESS
 * topic: /tf
    * to: /rtabmap/rgbd_odometry (http://Javad:44831/)
    * direction: inbound (34240 - Javad:48993) [21]
    * transport: TCPROS
 * topic: /tf_static
    * to: /rtabmap/link1_broadcaster (http://Javad:40025/)
    * direction: inbound (33450 - Javad:34165) [22]
    * transport: TCPROS
 * topic: /rtabmap/rgbd_image
    * to: /rtabmap/rgbd_sync (http://Javad:41699/)
    * direction: inbound (56616 - Javad:38067) [28]
    * transport: TCPROS
 * topic: /rtabmap/odom
    * to: /rtabmap/rgbd_odometry (http://Javad:44831/)
    * direction: inbound (34336 - Javad:48993) [29]
    * transport: TCPROS

------
Here I upload the pic in the rviz, that there are no green lines for landmarks.

[image]

matlabbe, 2022-07-28:
Hi, 
If you look at the code:
https://github.com/introlab/rtabmap_ros/blob/4530c42d06b8560c98b5500feb512ebe0ef36e72/src/CoreWrapper.cpp#L820-L822

rtabmap_ros should be built with fiducial_msgs pck to work:
https://github.com/introlab/rtabmap_ros/blob/4530c42d06b8560c98b5500feb512ebe0ef36e72/CMakeLists.txt#L31

Install `ros-noetic-fiducial-msgs` and rebuild rtabmap_ros.

MJA0111, 2022-08-01:
Hi,

Thank you for the reply.
Fudicial_msgs is already installed, but when I tried removing the line ( #ifdef WITH_FIDUCIAL_MSGS ) both for callback function and this line, it worked! 

So the source code must be always used to remove these two lines.

Kind regards,
Javad

matlabbe, 2022-11-27:
The binaries are not built with that message support, we have to rebuild. Then if fiducial_msgs is found, the define will be there to include that subscriber:
https://github.com/introlab/rtabmap_ros/blob/671c80dc3f0164f1ab95da96db74104182e8ec85/CMakeLists.txt#L311

closed 2022-11-27 by matlabbe