introlab/rtabmap_ros #796: Run RTAB Map on TUM RGBD dataset

sagardhatrak, 2022-08-10:
How can we run RUN RTABMAP on ROS2 foxy using tum rgbd dataset?

I am getting such error after executing command :

ros2 launch rtabmap_ros rgbdslam_datasets.launch

getting following error:

[INFO] [launch]: All log files can be found below /home/sagar/.ros/log/2022-08-10-19-21-00-067673-sagar-VirtualBox-10590
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: The launch file may have a syntax error, or its format is unknown

matlabbe, 2022-11-20:
This is a ROS1 launch file. Not sure if we can port it as TUM RGBD dataset rosbag format is ROS1 format.

You can however process offline the TUM dataset images using rtabmap standalone application or that CLI tool:
```bash
$ rtabmap-rgbd_dataset 

Usage:
rtabmap-rgbd_dataset [options] path
  path               Folder of the sequence (e.g., "~/rgbd_dataset_freiburg3_long_office_household")
                        containing least rgb_sync and depth_sync folders. These folders contain
                        synchronized images using associate.py tool (use tool version from
                        https://gist.github.com/matlabbe/484134a2d9da8ad425362c6669824798). If 
                        "groundtruth.txt" is found in the sequence folder, they will be saved in the database.
  --output           Output directory. By default, results are saved in "path".
  --output_name      Output database name (default "rtabmap").
  --skip #           Skip X frames.
  --quiet            Don't show log messages and iteration updates.
RTAB-Map options:
   --help                         Show usage.
   --version                      Show version of rtabmap and its dependencies.
   --params                       Show all parameters with their default value and description. 
                                  If a database path is set as last argument, the parameters in the 
                                  database will be shown in INI format.
   --"parameter name" "value"     Overwrite a specific RTAB-Map's parameter :
                                    --SURF/HessianThreshold 150
                                   For parameters in table format, add ',' between values :
                                    --Kp/RoiRatios 0,0,0.1,0
Logger options:
   --nolog              Disable logger
   --logconsole         Set logger console type
   --logfile "path"     Set logger file type
   --logfilea "path"    Set logger file type with appending mode if the file already exists
   --udebug             Set logger level to debug
   --uinfo              Set logger level to info
   --uwarn              Set logger level to warn
   --uerror             Set logger level to error
   --logtime "bool"     Print time when logging
   --logwhere "bool"    Print where when logging
   --logthread "bool"   Print thread id when logging

Example:

   $ rtabmap-rgbd_dataset \
       --Rtabmap/PublishRAMUsage true\
       --Rtabmap/DetectionRate 2\
       --RGBD/LinearUpdate 0\
       --Mem/STMSize 30\
       ~/rgbd_dataset_freiburg3_long_office_household
```

matlabbe, 2022-11-20:
Well, I've found this useful script to convert ros1 bags to ros2: https://docs.openvins.com/dev-ros1-to-ros2.html

I added an example launch file to use TUM RGBD dataset with rtabmap_ros on ROS2:  https://github.com/introlab/rtabmap_ros/blob/ros2/launch/ros2/rgbdslam_datasets.launch.py

Follow instructions at the top. You should have ROS1 and ROS2 installed and switch between then depending on the indicated section in the instructions.

closed 2022-12-15 by sagardhatrak