introlab/rtabmap_ros #798: Ros2 foxy, play .db file and without odometry and map

alvaropascuaI, 2022-08-12:
Is there any way to be able to launch a recorded .db file of the river topics themselves, of the depth cameras, visible..., and without odometry and the map, I have seen that it is possible without odometry, but at the time When selecting the database, it asks me for tables that are generated by recording with the rtabmap itself.

matlabbe, 2022-11-06:
Not sure to understand the question. If you want to play a `*.db` in ROS2, the [data_player](https://github.com/introlab/rtabmap_ros/blob/ros2/src/DbPlayerNode.cpp) node has not been ported to ros2 yet.

alvaropascuaI, 2022-11-07:
Hi, thanks for your answer, but i resolved, now i will launch 2 cameras and then reconstruction in rtabmap, you know how i can do it ?

matlabbe, 2022-11-26:
If you compare [ros1 multiple camera example](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_two_kinects.launch) with [ros2 examples](https://github.com/introlab/rtabmap_ros/tree/ros2/launch/ros2), you should be able to convert the ros1 example.

alvaropascuaI, 2022-12-01:
Hi, i try to launch demo_two_kinects.launch but with 2 realsense but in terminal i launch cameras realsense i have next error, i dont understand what topic he need

[image]

matlabbe, 2022-12-04:
You are feeding a `sensor_msgs/CompressedImage` topic to a `rtabmap_ros/RGBDImage` topic subscriber.

alvaropascuaI, 2022-12-12:
I know, but how i can generate this topic `rtabmap_ros/RGBDImage` ?

matlabbe, 2022-12-13:
there is a node called [rgbd_sync](http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Frgbd_sync).

alvaropascuaI, 2022-12-13:
I can launch file of  demo_two_kinects.launch, with my topics what i want but rtabmap viewer dont open and in terminal only write warnings
[image]

matlabbe, 2022-12-15:
What is the TF tree?
```
rosrun tf2_tools view_frames.py
```
You set `frame_id` of odometry node to `camera_link`, but there is no TF between `camera_link` and the frame set in the topic.

alvaropascuaI, 2022-12-16:
I launch the commando that you told me and generate this [file](https://github.com/introlab/rtabmap_ros/files/10243586/frames.pdf), but i dont understand you for change frame_id of odometry, how i can do it?

matlabbe, 2022-12-16:
Rgbd-odometry and rtabmap nodes have `frame_id` parameters to chamge the base frame. It should be the root frame of your robot, e.g., `base_link`.

Your tf tree is wrong, you should have only one root. You will have to fix it before using rtabmap.

alvaropascuaI, 2022-12-20:
How i can change topic that generate rtabmap/Mapdata topic?

matlabbe, 2022-12-21:
> How i can change topic that generate rtabmap/Mapdata topic?

I don't understand your question... _How can I change a topic to generate a topic?_ ??

alvaropascuaI, 2022-12-21:
If i launch file opensource topic of  rtabmap/Mapdata generate a map cloud, i've been looking and module that publish that topic in rtabmap and have the following suscribers topics and i cant understand why if i have the same topics in ROS 2 dont generate nothing

[image]

matlabbe, 2022-12-26:
These are subscribers, if these topics are not published, the problem is upstream on nodes that are publishing that data. Check which nodes are publishing the topics you need.