introlab/rtabmap_ros #804: Jetson Nano, RTABMAP, Ros Melodic, Jetpack 4.6, D435i

Pardeepanesar, 2022-09-04:
Dear whom it may concern I am currently experiencing issues when attempting to download the RTABMAP-Ros pkg on my 2gb Jetson nano. When following the instruction guide as stated on the readme.md file specifically for building with jetson nano issues are experienced.

The issue I am experiencing is that when building from source as stated and building the stand alone RTAB-Map libraries it is unable to complete the "make -j6" as shown in 2. When running the "make -j6" terminal command the building process for the node takes over an hour and only manages to get to 8%. The CPU usage bar is fully green implying it is using all CPU power available. 

Therefore I am currently stuck as I would like to utilise the RTABMAP Ros pkg with the 435i to utilise the obstacle_detection nodelets.

Any help would be greatly appreciated as I am unsure on how best to approach the problem. I also attempted to follow the installation guide highlighted in issue post: https://github.com/introlab/rtabmap/issues/427#issuecomment-608052821 and this has been unable to solve my installation issues. 

Any help on this would be greatly appreciated

matlabbe, 2022-10-24:
It is because there is not enough RAM. Try:
```
make -j1
```
and increase the swap memory is needed.

Brac24, 2023-02-22:
Yes that worked for me when I did it. Although it took about 14 hours to build so keep that in mind. There might be other dependencies as well that might cause the build to fail and you'll have to rerun it.

I recently was able to figure out how to cross build docker images on x86. These docker images have the l4t filesystem provided by the Jetpack images from Nvidia just not the cuda libraries. So when the docker image builds it is also compiling rtabmap and rtabmap_ros from source as if it was getting built on the Jetson nano. So if you have a desktop computer with more memory and better CPU that might help you out but the problem is the learning curve with docker and setting it up for cross building which takes some time to understand.

An issue I'm having right now is that rtabmap builds correctly but rtabmap_ros does not. I'm getting the same error as this https://github.com/ros-industrial/yak/issues/51

Not sure if it's a cmake issue as it was in that GitHub issue. I tried the build on the ros2 branch of the rtabmap_ros repo on Sunday.

matlabbe, 2023-02-26:
Reading your first sentence, I was about to suggest docker. We provide already built arm64 images: https://hub.docker.com/r/introlab3it/rtabmap_ros, tags explained here: https://github.com/introlab/rtabmap_ros/tree/master/docker

For the build error, not sure what is the problem. You can see our docker build logs here: https://github.com/introlab/rtabmap_ros/actions/workflows/docker.yml

For CUDA, rtabmap doesn't really use it anyway, unless you want to use SIFT GPU or SURF GPU, or SuperPoint. I have an example here for amd64 though: https://github.com/introlab/rtabmap/tree/master/archive/2022-IlluminationInvariant#docker

cheers,
Mathieu

Brac24, 2023-04-15:
Hey @matlabbe just tried the rtabmap ros arm64 image and was getting the following error: exec user process caused: exec format error. I found this link for possible solutions https://stackoverflow.com/questions/42494853/standard-init-linux-go178-exec-user-process-caused-exec-format-error.

I tried running the following commands:
sudo docker pull --platform linux/arm64 introlab3it/rtabmap_ros:foxy
sudo docker run --platform linux/arm64 -it --rm --network=host introlab3it/rtabmap_ros:foxy

I also tried pulling using the digest value to try to make sure I was pulling the arm64 image. I was trying to run this on Jetson Nano.

MrDarknessWolf, 2023-05-09:
question i managed to build this way, first rtab map THEN build the ros2 wrapper with stepper order, but when trying to launch it keeps crashing the RtabVizualizer  after trying to open, some say is the open cv version being 2 times but idk.. im confused

matlabbe, 2023-05-15:
@Brac24 It looks like osrf doesn't provide arm64 images for ros2: https://hub.docker.com/r/osrf/ros/tags?page=1&name=humble . The rtabmap image is based on osrf image, not sure why arm64 images of rtabmap are still built. I'll need to look more into that.