introlab/rtabmap_ros #810: Multi-camera marker detection

brayanpa, 2022-09-14:
Hi, I'm working with a robot with two cameras realsense d435. It is working very well in high feature environments. But right now I'm testing in a building where all the walls look the same, the solution I found was to use apriltags to identify some doors and corridors. But I'm having some trouble trying to get multiple cameras to work with marker detection. Is this feature planned to be implemented soon? Or should I find a different solution?

Thank you

matlabbe, 2022-09-14:
Two options:
* Use apriltag ros package to do detection on each camera, send tag detected topic to rtabmap_ros
* Wait for https://github.com/introlab/rtabmap/issues/898, currently working on it.

brayanpa, 2022-09-16:
Thanks for your quick response. I've been trying to run the latest version of rtabmap (in which you included support for multi-camera marker detector). But I am having several problems compiling rtabmap_ros with this version. Any suggestions on how I can launch this latest version of rtabmap in ROS Galactic.

Thank you.

matlabbe, 2022-10-24:
What are the problems?

Ideally, to avoid recompiling also `cv_bridge` and all other packages depending on it, rebuild the **same** OpenCV version (with [opencv_contrib](https://github.com/opencv/opencv_contrib)) than the system binaries on your system. See comment [here](https://github.com/introlab/rtabmap_ros#build-from-source) about opencv.

closed 2022-11-30 by brayanpa