introlab/rtabmap_ros #811: TypeError when i try to run Rtabmap for Oak-D in ros2 foxy

Mothish97, 2022-09-15:
I am trying to run the Rtabmap with Oak-D camera and the camera and the madgwick filter for IMU are working fine but when I try to run the rtabmap, I'm getting a type error

Command: 

ros2 launch rtabmap_ros rtabmap.launch.py     args:='--delete_db_on_start'     stereo:=true      left_image_topic:= /stereo_inertial_publisher/left/image_rect     right_image_topic:= /stereo_inertial_publisher/right/image_rect     left_camera_info_topic:=/stereo_inertial_publisher/left/camera_info     right_camera_info_topic:=/stereo_inertial_publisher/right/camera_info     imu_topic:=/stereo_inertial_publisher/imu/data     frame_id:=oak-d_frame     approx_sync:=true     approx_sync_max_interval:=0.001     wait_imu_to_init:=true

Error:
[INFO] [launch]: All log files can be found below /home/deadeye/.ros/log/2022-09-15-15-38-08-397575-deadeye-ROG-Strix-G513QE-G513QE-26143
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:226> exception=TypeError("Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)")>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 2 more times]
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 444, in execute
    self._perform_substitutions(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/actions/node.py", line 399, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 160, in evaluate_parameters
    output_params.append(evaluate_parameter_dict(context, param))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch_ros/utilities/evaluate_parameters.py", line 95, in evaluate_parameter_dict
    raise TypeError(
TypeError: Allowed value types are bytes, bool, int, float, str, Sequence[bool], Sequence[int], Sequence[float], Sequence[str]. Got <class 'NoneType'>.If the parameter is meant to be a string, try wrapping it in launch_ros.parameter_descriptions.ParameterValue(value, value_type=str)

matlabbe, 2022-09-30:
There is an issue with `rtabmap.launch.py` on foxy, but it works on ROS Galactic. I'll suggest to move to galactic.

matlabbe, 2022-10-01:
Well, just tried latest [rtabmap.launch.py](https://github.com/introlab/rtabmap_ros/commits/master/launch/ros2/rtabmap.launch.py) on foxy and no problems...