introlab/rtabmap_ros #812: update the map while passing through the same point

hsynklceng, 2022-09-16:
Hi @matlabbe ,

we're trying to map an environment where people move around. Old pictures are not deleted when we go through the same region again. How can I remove people who have been added to the map later?

matlabbe, 2022-10-24:
The occupancy grid would be able to clear dynamic obstacles. However, for the 3D point cloud used for visualization, the dynamic obstacles won't be removed. If you need a 3D point cloud with dynamic obstacles removed, you may generate an OctoMap and in rviz show the related PointCloud2 from octomap. Note that to better clear dynamic obstacles, set `Grid/RayTracing=true`.

hsynklceng, 2022-10-25:
Hi @matlabbe ,

thank you for answer and advise.

closed 2022-10-25 by hsynklceng