introlab/rtabmap_ros #816: no transform from base_footprint to map using turtlebot3_navigation.launch

PaulBremner, 2022-09-22:
I am trying to run:
 $ export TURTLEBOT3_MODEL=waffle
 $ roslaunch turtlebot3_gazebo turtlebot3_world.launch

 $ export TURTLEBOT3_MODEL=waffle
 $ roslaunch rtabmap_ros demo_turtlebot3_navigation.launch

and I get an error that there is no transform between /map and /base_footprint and that target_frame map does not exist. The map frame does not show up on my tf_frame graph. I've changed nothing so I'm confused as to why this doesn't just work out of the box.

I am running ROS Melodic on Ubuntu 18.04 on a VMWare virtual machine.

Snip of the error messages

[image]

tosa-no-onchan, 2022-10-14:
I think.
Check an each publish rate of 3 topics.
$ rostopic hz /camera/rgb/image_raw
$ rostopic hz /camera/depth/image_raw
$ rostopic hz /camera/rgb/camera_info
timeout was 0.1 -> 10[Hz]
or "approx sync"
approx_sync=true
&lt;param name="approx_sync"      type="bool" value="true"/&gt;

matlabbe, 2022-10-23:
Cannot reproduce it on Noetic. Are there other errors? Maybe a virtual machine issue... as @tosa-no-onchan said, check if the topics are published with `rostopic hz`.