introlab/rtabmap_ros #818: Question for rtabmapviz

jaykorea, 2022-09-30:
Is rtabmapviz also opensource??

then, where can I find the source..

I want to import and export 2d map with command line or source..

If u don't mind, plz tell me where to modficate else, Is there any way to import & export 2d map with command line.

Sincerely, Thanks again to your awesome slam package!!

matlabbe, 2022-09-30:
You cannot import a 2d map. To export the 2d map, if you are on ros, I'll suggest to use [map_saver](http://wiki.ros.org/map_server#map_saver) node.

For the first question, yes, rtabmapviz is a ros wrapper of the MainWindow [here](https://github.com/introlab/rtabmap/blob/master/guilib/src/MainWindow.cpp).

jaykorea, 2022-10-02:
> 2D 지도는 가져올 수 없습니다. 2d 맵을 내보내려면 ros에 있다면 [map_saver](http://wiki.ros.org/map_server#map_saver) 노드를 사용하는 것이 좋습니다.
> 
> 첫 번째 질문의 경우 예, rtabmapviz는 [여기](https://github.com/introlab/rtabmap/blob/master/guilib/src/MainWindow.cpp) 에서 MainWindow의 ros 래퍼입니다 .

Sorry for my ignorance.

You mean is it impossible to import 2d map via command line? I think I can import 2d map via rtabmapviz , right?

matlabbe, 2022-10-23:
Without modifying anything, you can import a 2D occupancy grid using `rtabmap-databaseViewer` -> File-> "Import optimized 2d map". However, as explained [here](https://github.com/introlab/rtabmap_ros/issues/213), this only works with a grid previously exported from the same tool.