introlab/rtabmap_ros #819: apriltag and apriltag_ros difference between rtabmap and rtabmap_ros

tugbakara, 2022-10-01:
I serached in rtabmap , rtabmap_ros github issues and also I looked at smilar launch files (there exist only [one](https://github.com/antoan/nemesis_core/blob/a7ccec361f04221c1555b6b65c2b8b61507c901c/nemesis_navigation/launch/nav.launch)) and in rtabmap forum but there is something that is not clear about detection algorithms, if I use external apriltag_ros package to use with rtbmap I know there is rtabmap_ros apriltag_ros detection launch BUT if I use that launch file I cannot set RGBD/MarkerDetection to true that's why ( I think RGBD/MarkerDetection relates to Marker/... params) I cannot use Marker/MaxRange param. I am not sure are the related to each other or If I use example launch file of apriltag_ros which is rtabmap_ros package, setting Marker/MaxRange some value without setting RGBD/MarkerDetection true?

matlabbe, 2022-10-01:
If you are using external `aprilttag_ros` to detect markers, don't set `RGBD/MarkerDetection` and `Marker/` parameters are not used with external detected markers. If you want to filter the detections by range, you should filter the detection topics before sending them to `rtabmap` node.

tugbakara, 2022-10-01:
> If you are using external `aprilttag_ros` to detect markers, don't set `RGBD/MarkerDetection` and `Marker/` parameters are not used with external detected markers. If you want to filter the detections by range, you should filter the detection topics before sending them to `rtabmap` node.

What do you mean when you said **filter**?
How can I do that?

tugbakara, 2022-10-03:
I install rtabmap and rtabmap_ros  master branches but I didn't see any flag related to aruco markers adn when I try to use RGBD/MarkerDetection param to use it with april tags rtabmap gaves me an error that's rtabmap itself wasn't build with aruco? Is there any wayto install it? It's not written anywhere so is there any missing steps? Also I have Nvidia Jetson AGX.
Have a nice day @matlabbe !

Srijal97, 2022-10-21:
@tugbakara 
1) By filter, they mean that you can create a node which subscribes to the apriltag /tag_detection topic, check the pose value of each marker that is detected and remove the detections which are farther away than the range you want. You can then publish this "filtered" data to a new topic, say /tag_detection_filtered, and send that to rtabmap
2) I am looking to do something similar, but I have not really done it yet. But if you search for "apriltag" or "aruco" under the rtabmap (not rtabmap_ros) github issues AND also the rtabmap forums, there are several helpful discussions. E.g:
https://github.com/introlab/rtabmap/issues/377
http://official-rtab-map-forum.206.s1.nabble.com/Using-Apriltags-to-optimize-the-graph-td6369.html#a6386
which may help. I will be testing it out an ROS Noetic on a Jetson Orin, and will keep this thread updated with how that goes.

tugbakara, 2022-10-21:
Hi @Srijal97,  I solved that issue and open a pull request for it : [link](https://github.com/AprilRobotics/apriltag_ros/pull/136) by the way I have a question about percentage of detection. Is there any fork or implementation about it?

Srijal97, 2022-10-21:
> Hi @Srijal97, I solved that issue and open a pull request for it : [link](https://github.com/AprilRobotics/apriltag_ros/pull/136) by the way I have a question about percentage of detection. Is there any fork or implementation about it?

What exactly do you mean by the percentage of detection?

matlabbe, 2022-10-23:
To use external tags with rtabmap_ros, no recompilation is needed on rtabmap library side. There are two types of tag message supported depending of the upstream ros package you are using for the tag detection (`tag_detections` is available by default with `rtabmap_ros` binaries):
https://github.com/introlab/rtabmap_ros/blob/44bbaa2cefd3e976648baa8cbd8ba7df4b90b5af/src/CoreWrapper.cpp#L817-L822

The rtabmap library has also its own internal way to detect Apriltags or Aruco tags, which can be enabled by setting `RGBD/MarkerDetection` to true (then setting corresponding parameters under `Marker/****` group: `rtabmap --params | grep Marker/`). This approach is mainly used for RTAB-Map standalone app, Android and iOS apps. It requires to build rtabmap library with OpenCV aruco module (which is not available with OpenCV binaries). Apriltags can be used with this library by setting `Marker/Dictionary` to corresponding apriltag dictionary. For this approach, as openCV  needs to be rebuild from source, if you want to use under ROS, you should carefully rebuild same OpenCV version, see comment [here](https://github.com/introlab/rtabmap_ros/#build-from-source):

> On Melodic/Noetic, build from source with xfeatures2d module (and nonfree module if needed) the same OpenCV version already installed on the system. You will then avoid breaking cv_bridge with rtabmap_ros. If you want to install a more recent OpenCV version, I recommend to uninstall libopencv* libraries (with all ros packages depending on it) and rebuild all those ros packages in your catkin workspace (to make sure cv_bridge is linked on the OpenCV version you just compiled).

closed 2022-10-24 by tugbakara