introlab/rtabmap_ros #820: rtabmapviz parameters not found

Huynh-Anh, 2022-10-03:
For some reason i get a bunch of parameters not found when i launch the node

I installed rtabmap ros with source AND then i tried via apt
```
roslaunch rtabmap_ros rtabmap.launch args:="--delete_db_on_start" \
rgb_topic:=/stereo_inertial_publisher/color/image \
camera_info_topic:=/stereo_inertial_publisher/color/camera_info \
depth_topic:=/stereo_inertial_publisher/stereo/depth \
imu_topic:=/stereo_inertial_publisher/imu \
frame_id:=oak-d_frame \
approx_sync:=true \
wait_imu_to_init:=true
```
logs :
```
[ WARN] [1664831864.979014596]: rtabmapviz: Parameter ORB/PatchSize not found
[ WARN] [1664831864.984602077]: rtabmapviz: Parameter ORB/ScaleFactor not found
[ WARN] [1664831864.990836564]: rtabmapviz: Parameter ORB/ScoreType not found
[ WARN] [1664831864.996508253]: rtabmapviz: Parameter ORB/WTA_K not found
[ WARN] [1664831865.003033127]: rtabmapviz: Parameter Optimizer/Epsilon not found
[ WARN] [1664831865.007849279]: rtabmapviz: Parameter Optimizer/GravitySigma not found
[ WARN] [1664831865.015556668]: rtabmapviz: Parameter Optimizer/Iterations not found
[ WARN] [1664831865.021055512]: rtabmapviz: Parameter Optimizer/LandmarksIgnored not found
[ WARN] [1664831865.026621269]: rtabmapviz: Parameter Optimizer/PriorsIgnored not found
[ WARN] [1664831865.030369831]: rtabmapviz: Parameter Optimizer/Robust not found
[ WARN] [1664831865.035200723]: rtabmapviz: Parameter Optimizer/Strategy not found
[ WARN] [1664831865.041116612]: rtabmapviz: Parameter Optimizer/VarianceIgnored not found
[ WARN] [1664831865.047235665]: rtabmapviz: Parameter PyDetector/Cuda not found
[ WARN] [1664831865.052290505]: rtabmapviz: Parameter PyDetector/Path not found
[ WARN] [1664831865.057357348]: rtabmapviz: Parameter PyMatcher/Cuda not found
[ WARN] [1664831865.063597317]: rtabmapviz: Parameter PyMatcher/Iterations not found
[ WARN] [1664831865.073213114]: rtabmapviz: Parameter PyMatcher/Model not found
[ WARN] [1664831865.080075928]: rtabmapviz: Parameter PyMatcher/Path not found
[ INFO] [1664831865.085753880]: odometry: Subscribing to IMU topic /stereo_inertial_publisher/imu
[ WARN] [1664831865.087893028]: rtabmapviz: Parameter PyMatcher/Threshold not found
[ WARN] [1664831865.094161081]: rtabmapviz: Parameter RGBD/AngularSpeedUpdate not found
[ WARN] [1664831865.100407935]: rtabmapviz: Parameter RGBD/AngularUpdate not found
[ WARN] [1664831865.114193009]: rtabmapviz: Parameter RGBD/CreateOccupancyGrid not found
[ WARN] [1664831865.120635714]: rtabmapviz: Parameter RGBD/Enabled not found
[ WARN] [1664831865.126265734]: rtabmapviz: Parameter RGBD/GoalReachedRadius not found
[ WARN] [1664831865.136836162]: rtabmapviz: Parameter RGBD/GoalsSavedInUserData not found
[ WARN] [1664831865.144091817]: rtabmapviz: Parameter RGBD/LinearSpeedUpdate not found
[ WARN] [1664831865.148914334]: rtabmapviz: Parameter RGBD/LinearUpdate not found
[ WARN] [1664831865.153923572]: rtabmapviz: Parameter RGBD/LocalBundleOnLoopClosure not found
[ INFO] [1664831865.157912642]: RGBDOdometry: approx_sync    = true
[ INFO] [1664831865.158010534]: RGBDOdometry: queue_size     = 10
[ INFO] [1664831865.158085599]: RGBDOdometry: subscribe_rgbd = false
[ INFO] [1664831865.158183799]: RGBDOdometry: rgbd_cameras   = 1
[ INFO] [1664831865.158272569]: RGBDOdometry: keep_color     = false
[ WARN] [1664831865.161573886]: rtabmapviz: Parameter RGBD/LocalImmunizationRatio not found
[ WARN] [1664831865.167210659]: rtabmapviz: Parameter RGBD/LocalRadius not found
[ WARN] [1664831865.175746913]: rtabmapviz: Parameter RGBD/LoopClosureIdentityGuess not found
[ WARN] [1664831865.182969209]: rtabmapviz: Parameter RGBD/LoopClosureReextractFeatures not found
[ WARN] [1664831865.188428085]: rtabmapviz: Parameter RGBD/LoopCovLimited not found
[ WARN] [1664831865.192922394]: rtabmapviz: Parameter RGBD/MarkerDetection not found
[ WARN] [1664831865.199307772]: rtabmapviz: Parameter RGBD/MaxLocalRetrieved not found
[ WARN] [1664831865.206261628]: rtabmapviz: Parameter RGBD/MaxLoopClosureDistance not found
[ WARN] [1664831865.212131534]: rtabmapviz: Parameter RGBD/MaxOdomCacheSize not found
[ WARN] [1664831865.224876299]: rtabmapviz: Parameter RGBD/NeighborLinkRefining not found
[ WARN] [1664831865.230429449]: rtabmapviz: Parameter RGBD/NewMapOdomChangeDistance not found
[ WARN] [1664831865.235246681]: rtabmapviz: Parameter RGBD/OptimizeFromGraphEnd not found
[ WARN] [1664831865.241954201]: rtabmapviz: Parameter RGBD/OptimizeMaxError not found
[ WARN] [1664831865.247584814]: rtabmapviz: Parameter RGBD/PlanAngularVelocity not found
[ WARN] [1664831865.252998343]: rtabmapviz: Parameter RGBD/PlanLinearVelocity not found
[ WARN] [1664831865.258843972]: rtabmapviz: Parameter RGBD/PlanStuckIterations not found
[ WARN] [1664831865.268080601]: rtabmapviz: Parameter RGBD/ProximityAngle not found
[ WARN] [1664831865.278081462]: rtabmapviz: Parameter RGBD/ProximityBySpace not found
[ WARN] [1664831865.282786285]: rtabmapviz: Parameter RGBD/ProximityByTime not found
[ WARN] [1664831865.291980355]: rtabmapviz: Parameter RGBD/ProximityGlobalScanMap not found
[ WARN] [1664831865.297653008]: rtabmapviz: Parameter RGBD/ProximityMaxGraphDepth not found
[... 116 lines cut ...]
[ WARN] [1664831872.313302067]: rtabmapviz: Parameter Vis/MaxDepth not found
[ WARN] [1664831872.526526327]: rtabmapviz: Parameter Vis/MaxFeatures not found
[ WARN] [1664831872.774264507]: rtabmapviz: Parameter Vis/MeanInliersDistance not found
[ WARN] [1664831873.078979052]: rtabmapviz: Parameter Vis/MinDepth not found
[ WARN] [1664831873.353767656]: rtabmapviz: Parameter Vis/MinInliers not found
[ WARN] [1664831873.664904104]: rtabmapviz: Parameter Vis/MinInliersDistribution not found
[ WARN] [1664831873.930498346]: rtabmapviz: Parameter Vis/PnPFlags not found
[ WARN] [1664831874.262626942]: rtabmapviz: Parameter Vis/PnPRefineIterations not found
[ WARN] [1664831874.576287838]: rtabmapviz: Parameter Vis/PnPReprojError not found
[ WARN] [1664831874.987141550]: rtabmapviz: Parameter Vis/RefineIterations not found
[ WARN] [1664831875.326713023]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /stereo_inertial_publisher/color/image \
   /stereo_inertial_publisher/stereo/depth \
   /stereo_inertial_publisher/color/camera_info
[ WARN] [1664831875.385191468]: rtabmapviz: Parameter Vis/RoiRatios not found
[ WARN] [1664831875.769758079]: rtabmapviz: Parameter Vis/SubPixEps not found
[ WARN] [1664831876.190527129]: rtabmapviz: Parameter Vis/SubPixIterations not found
[ WARN] [1664831876.629998944]: rtabmapviz: Parameter Vis/SubPixWinSize not found
[ WARN] [1664831877.115772164]: rtabmapviz: Parameter g2o/Baseline not found
[ WARN] [1664831877.607231601]: rtabmapviz: Parameter g2o/Optimizer not found
[ WARN] [1664831878.128869678]: rtabmapviz: Parameter g2o/PixelVariance not found
[ WARN] [1664831878.607496923]: rtabmapviz: Parameter g2o/RobustKernelDelta not found
[ WARN] [1664831879.192801186]: rtabmapviz: Parameter g2o/Solver not found
```

Huynh-Anh, 2022-10-05:
@matlabbe do you have an idea on why i would get so many warnings ? I am using an OAK-D camera plugged in a raspberry pi. and my pc is running rtabmap and subscribed to the topics from the pi.
I can see the topics, but even after all those warnings, i dont see anything happening. it still waits for the topics..

```
roslaunch rtabmap_ros rtabmap.launch args:="--delete_db_on_start" \
stereo:=true \
left_image_topic:=/stereo_publisher/left/image \
right_image_topic:=/stereo_publisher/right/image \
left_camera_info_topic:=/stereo_publisher/left/camera_info \
right_camera_info_topic:=/stereo_publisher/right/camera_info \
imu_topic:=/stereo_inertial_publisher/imu \
frame_id:=oak-d_frame \
approx_sync:=true \
approx_sync_max_interval:=0.001 \
wait_imu_to_init:=true
```

```
[ WARN] [1664982723.435917338]: rtabmapviz: rtabmap's parameters seem not all there yet! continuing with those there if some...
[ WARN] [1664982724.504323405]: rtabmapviz: Parameter BRIEF/Bytes not found
[ WARN] [1664982725.445268922]: rtabmapviz: Parameter BRISK/Octaves not found
[ WARN] [1664982726.000672114]: rtabmapviz: Parameter BRISK/PatternScale not found
[ WARN] [1664982727.309989646]: rtabmapviz: Parameter BRISK/Thresh not found
[ WARN] [1664982728.275127613]: rtabmapviz: Parameter Bayes/FullPredictionUpdate not found
[ WARN] [1664982729.082287823]: rtabmapviz: Parameter Bayes/PredictionLC not found
[ INFO] [1664982729.083640796]: /rtabmap/rtabmap(maps): map_filter_radius          = 0.000000
[ INFO] [1664982729.083850420]: /rtabmap/rtabmap(maps): map_filter_angle           = 30.000000
[ INFO] [1664982729.083986372]: /rtabmap/rtabmap(maps): map_cleanup                = true
[ INFO] [1664982729.084079364]: /rtabmap/rtabmap(maps): map_always_update          = false
[ INFO] [1664982729.084201237]: /rtabmap/rtabmap(maps): map_empty_ray_tracing      = true
[ INFO] [1664982729.084295598]: /rtabmap/rtabmap(maps): cloud_output_voxelized     = true
[ INFO] [1664982729.084414855]: /rtabmap/rtabmap(maps): cloud_subtract_filtering   = false
[ INFO] [1664982729.084518634]: /rtabmap/rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[ WARN] [1664982729.818694535]: rtabmapviz: Parameter Bayes/VirtualPlacePriorThr not found
[ WARN] [1664982730.785965924]: rtabmapviz: Parameter Db/TargetVersion not found
[ WARN] [1664982731.626069897]: rtabmapviz: Parameter DbSqlite3/CacheSize not found
[ WARN] [1664982732.166581243]: rtabmapviz: Parameter DbSqlite3/InMemory not found
[ WARN] [1664982733.109373197]: rtabmapviz: Parameter DbSqlite3/JournalMode not found
[ WARN] [1664982733.854469438]: rtabmapviz: Parameter DbSqlite3/Synchronous not found
[ WARN] [1664982734.361101937]: rtabmapviz: Parameter DbSqlite3/TempStore not found
[ WARN] [1664982735.030156058]: rtabmapviz: Parameter FAST/CV not found
[ WARN] [1664982735.608316484]: rtabmapviz: Parameter FAST/Gpu not found
[ WARN] [1664982736.478841017]: rtabmapviz: Parameter FAST/GpuKeypointsRatio not found
[ WARN] [1664982737.369899942]: rtabmapviz: Parameter FAST/GridCols not found
[ WARN] [1664982737.569175728]: rtabmapviz: Parameter FAST/GridRows not found
[ WARN] [1664982738.565254630]: rtabmapviz: Parameter FAST/MaxThreshold not found
[ WARN] [1664982739.127377780]: rtabmapviz: Parameter FAST/MinThreshold not found
[ WARN] [1664982739.357977037]: rtabmapviz: Parameter FAST/NonmaxSuppression not found
[ WARN] [1664982739.413550241]: rtabmapviz: Parameter FAST/Threshold not found
[ WARN] [1664982739.970317894]: rtabmapviz: Parameter FREAK/NOctaves not found
[ WARN] [1664982740.945755912]: rtabmapviz: Parameter FREAK/OrientationNormalized not found
[ WARN] [1664982741.482460927]: rtabmapviz: Parameter FREAK/PatternScale not found
[ WARN] [1664982742.196003734]: rtabmapviz: Parameter FREAK/ScaleNormalized not found
[ WARN] [1664982742.865577064]: rtabmapviz: Parameter GFTT/BlockSize not found
[ WARN] [1664982743.644656733]: rtabmapviz: Parameter GFTT/K not found
[ WARN] [1664982744.557189331]: rtabmapviz: Parameter GFTT/MinDistance not found
[ WARN] [1664982745.601387510]: rtabmapviz: Parameter GFTT/QualityLevel not found
[ WARN] [1664982746.503590679]: rtabmapviz: Parameter GFTT/UseHarrisDetector not found
[ WARN] [1664982747.340883707]: rtabmapviz: Parameter GMS/ThresholdFactor not found
[ WARN] [1664982747.828505595]: rtabmapviz: Parameter GMS/WithRotation not found
[ WARN] [1664982748.448661042]: rtabmapviz: Parameter GMS/WithScale not found
[ INFO] [1664982748.448977323]: Odometry: frame_id               = oak-d_frame
[ INFO] [1664982748.449281876]: Odometry: odom_frame_id          = odom
[ INFO] [1664982748.449551745]: Odometry: publish_tf             = true
[ INFO] [1664982748.449741790]: Odometry: wait_for_transform     = true
[ INFO] [1664982748.449919504]: Odometry: wait_for_transform_duration  = 0.200000
[ INFO] [1664982748.450191516]: Odometry: initial_pose           = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[ INFO] [1664982748.450276579]: Odometry: ground_truth_frame_id  = 
[ INFO] [1664982748.450306803]: Odometry: ground_truth_base_frame_id = 
[ INFO] [1664982748.450330257]: Odometry: config_path            = 
[... 316 lines cut ...]
[ WARN] [1664982987.506652746]: rtabmapviz: Parameter Vis/GridRows not found
[ WARN] [1664982988.474755824]: rtabmapviz: Parameter Vis/InlierDistance not found
[ WARN] [1664982989.157531110]: rtabmapviz: Parameter Vis/Iterations not found
[ WARN] [1664982990.136135200]: rtabmapviz: Parameter Vis/MaxDepth not found
[... cut ...]
[ WARN] [1664983001.606867528]: rtabmapviz: Parameter Vis/SubPixWinSize not found
[ WARN] [1664983002.517489438]: rtabmapviz: Parameter g2o/Baseline not found
[ WARN] [1664983003.435693147]: rtabmapviz: Parameter g2o/Optimizer not found
[ WARN] [1664983004.270654557]: rtabmapviz: Parameter g2o/PixelVariance not found
[ WARN] [1664983005.097783106]: rtabmapviz: Parameter g2o/RobustKernelDelta not found
[ WARN] [1664983006.197659391]: rtabmapviz: Parameter g2o/Solver not found
[ INFO] [1664983006.197785033]: rtabmapviz: Parameters read = 2
```

matlabbe, 2022-10-05:
Can you try to explicitly set the `ROS_IP` on both machines? Can you also show the full log when launching rtabmap.launch? Normally rtabmapviz should wait some seconds before trying to get params from rosparam server. If rtabmap node is up and rosparam fails, it means there is a problem with communication between the two computers. Generally setting `ROS_IP` explicitly on both computers may fix communication issues (in particular when hostname of computers cannot be reliably resolved by both computers).

Huynh-Anh, 2022-10-05:
is there a way to specify a log path, i have to many logs that i cant get the exact one ..

Huynh-Anh, 2022-10-05:
ok this is the log from the .ros folder
```
[roslaunch][INFO] 2022-10-05 15:21:07,317: Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2022-10-05 15:21:07,324: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2022-10-05 15:21:07,325: roslaunch starting with args ['/opt/ros/melodic/bin/roslaunch', 'rtabmap_ros', 'rtabmap.launch', 'args:=--delete_db_on_start', 'stereo:=true', 'left_image_topic:=/stereo_publisher/left/image', 'right_image_topic:=/stereo_publisher/right/image', 'left_camera_info_topic:=/stereo_publisher/left/camera_info', 'right_camera_info_topic:=/stereo_publisher/right/camera_info', 'frame_id:=oak-d_frame', 'approx_sync:=true']
[roslaunch][INFO] 2022-10-05 15:21:07,325: roslaunch env is {'XDG_SESSION_TYPE': 'x11', 'LC_PAPER': 'fr_CA.UTF-8', 'SHELL': '/bin/bash', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop', 'MANDATORY_PATH': '/usr/share/gconf/ubuntu.mandatory.path', 'TERMINATOR_UUID': 'urn:uuid:e0ddcc3d-2036-400c-8e44-1216a33448b7', 'CLUTTER_IM_MODULE': 'xim', 'TEXTDOMAIN': 'im-config', 'XMODIFIERS': '@im=ibus', 'GTK2_MODULES': 'overlay-scrollbar', 'XDG_RUNTIME_DIR': '/run/user/1000', 'LC_ADDRESS': 'fr_CA.UTF-8', 'PYTHONPATH': '/home/huynh-anh/catkin_ws/devel/lib/python2.7/dist-packages:/opt/ros/melodic/lib/python2.7/dist-packages', 'ROS_ROOT': '/opt/ros/melodic/share/ros', 'XDG_SESSION_ID': '2', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', 'DESKTOP_SESSION': 'ubuntu', 'GTK_MODULES': 'gail:atk-bridge', 'LC_NAME': 'fr_CA.UTF-8', 'XDG_MENU_PREFIX': 'gnome-', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:', 'LC_NUMERIC': 'fr_CA.UTF-8', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'XDG_CURRENT_DESKTOP': 'ubuntu:GNOME', 'USER': 'huynh-anh', 'XDG_VTNR': '2', 'CMAKE_PREFIX_PATH': '/home/huynh-anh/catkin_ws/devel:/opt/ros/melodic', 'XAUTHORITY': '/run/user/1000/gdm/Xauthority', 'LANGUAGE': 'en_CA:en', 'SESSION_MANAGER': 'local/huynhanh-XPS-13-9350:@/tmp/.ICE-unix/1734,unix/huynhanh-XPS-13-9350:/tmp/.ICE-unix/1734', 'LC_MEASUREMENT': 'fr_CA.UTF-8', 'QT_QPA_PLATFORMTHEME': 'appmenu-qt5', 'LD_LIBRARY_PATH': '/home/huynh-anh/catkin_ws/devel/lib:/opt/ros/melodic/lib', 'QT_ACCESSIBILITY': '1', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'XDG_SESSION_DESKTOP': 'ubuntu', 'GDMSESSION': 'ubuntu', 'TERMINATOR_DBUS_NAME': 'net.tenshu.Terminator20x1a6021154d881d', 'PKG_CONFIG_PATH': '/home/huynh-anh/catkin_ws/devel/lib/pkgconfig:/opt/ros/melodic/lib/pkgconfig', 'ROS_PACKAGE_PATH': '/home/huynh-anh/catkin_ws/src:/opt/ros/melodic/share', '_': '/opt/ros/melodic/bin/roslaunch', 'GTK_IM_MODULE': 'ibus', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/etc/xdg', 'ROS_MASTER_URI': 'http://192.168.1.154:11311', 'COLORTERM': 'truecolor', 'LC_TIME': 'fr_CA.UTF-8', 'QT4_IM_MODULE': 'xim', 'GJS_DEBUG_OUTPUT': 'stderr', 'ROS_IP': '192.168.1.197', 'HOME': '/home/huynh-anh', 'DISPLAY': ':1', 'LANG': 'en_CA.UTF-8', 'ROS_LOG_FILENAME': '/home/huynh-anh/.ros/log/d81e1128-44e2-11ed-81b6-dca632d50a38/roslaunch-huynhanh-XPS-13-9350-25966.log', 'LC_MONETARY': 'fr_CA.UTF-8', 'ROS_PYTHON_VERSION': '2', 'USERNAME': 'huynh-anh', 'LC_IDENTIFICATION': 'fr_CA.UTF-8', 'VTE_VERSION': '5202', 'TERMINATOR_DBUS_PATH': '/net/tenshu/Terminator2', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'DEFAULTS_PATH': '/usr/share/gconf/ubuntu.default.path', 'PAPERSIZE': 'letter', 'ROS_ETC_DIR': '/opt/ros/melodic/etc/ros', 'ROS_DISTRO': 'melodic', 'ROS_HOSTNAME': '192.168.1.197', 'ROSLISP_PACKAGE_DIRECTORIES': '/home/huynh-anh/catkin_ws/devel/share/common-lisp', 'QT_IM_MODULE': 'ibus', 'LOGNAME': 'huynh-anh', 'XDG_SEAT': 'seat0', 'PATH': '/opt/ros/melodic/bin:/home/huynh-anh/bin:/home/huynh-anh/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/huynh-anh/.local/share/JetBrains/Toolbox/scripts', 'SSH_AGENT_PID': '1820', 'TERM': 'xterm-256color', 'WINDOWPATH': '2', ' [... cut ...]
[roslaunch][INFO] 2022-10-05 15:24:32,560: shutting down processing monitor <ProcessMonitor(ProcessMonitor-1, stopped daemon 139702603896576)>
[roslaunch.pmon][INFO] 2022-10-05 15:24:32,561: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 139702603896576)>
[roslaunch][INFO] 2022-10-05 15:24:32,561: ... shutting down processing monitor complete
[roslaunch][INFO] 2022-10-05 15:24:32,561: done
[roslaunch.pmon][INFO] 2022-10-05 15:24:32,562: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 139702603896576)>
[rospy.core][INFO] 2022-10-05 15:24:32,563: signal_shutdown [atexit]
```

Huynh-Anh, 2022-10-05:
```
... logging to /home/huynh-anh/.ros/log/e550e928-44e3-11ed-848c-74dfbf308a97/roslaunch-huynhanh-XPS-13-9350-26197.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.197:45389/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmapviz/
 * /rtabmap/stereo_odometry/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: oak-d_frame
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: False
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgb: False
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap/subscribe_stereo: True
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmapviz/approx_sync: True
 * /rtabmap/rtabmapviz/frame_id: oak-d_frame
 * /rtabmap/rtabmapviz/odom_frame_id: 
 * /rtabmap/rtabmapviz/queue_size: 10
 * /rtabmap/rtabmapviz/subscribe_depth: False
 * /rtabmap/rtabmapviz/subscribe_odom_info: True
 * /rtabmap/rtabmapviz/subscribe_rgb: False
 * /rtabmap/rtabmapviz/subscribe_rgbd: False
 * /rtabmap/rtabmapviz/subscribe_scan: False
 * /rtabmap/rtabmapviz/subscribe_scan_cloud: False
 * /rtabmap/rtabmapviz/subscribe_scan_descriptor: False
 * /rtabmap/rtabmapviz/subscribe_stereo: True
 * /rtabmap/rtabmapviz/wait_for_transform_duration: 0.2
 * /rtabmap/stereo_odometry/approx_sync: True
 * /rtabmap/stereo_odometry/approx_sync_max_interval: 0.0
 * /rtabmap/stereo_odometry/config_path: 
[... 435 lines cut ...]
[ INFO] [1664998249.346225113]: /rtabmap/rtabmapviz: subscribe_scan_descriptor = false
[ INFO] [1664998249.346337030]: /rtabmap/rtabmapviz: queue_size    = 10
[ INFO] [1664998249.346440695]: /rtabmap/rtabmapviz: approx_sync   = true
[ INFO] [1664998249.346667091]: Setup stereo callback
[ INFO] [1664998250.474413299]: Setting RTAB-Map parameter "Mem/IncrementalMemory"="true"
[ INFO] [1664998251.070577892]: Setting RTAB-Map parameter "Mem/InitWMWithAllNodes"="false"tabmapviz)
    stereo_odometry (rtabmap_ros/stereo_odometry)
/rtabmap/rtabmapviz subscribed to (approx sync):
   /rtabmap/odom \
   /stereo_publisher/left/image \
   /stereo_publisher/right/image \
   /stereo_publisher/left/camera_info \
   /stereo_publisher/right/camera_info \
   /rtabmap/odom_info
[ INFO] [1664998261.858239873]: rtabmapviz started.
[ WARN] [1664998266.858236180]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/rtabmap/rtabmapviz subscribed to (approx sync):
   /rtabmap/odom \
   /stereo_publisher/left/image \
   /stereo_publisher/right/image \
   /stereo_publisher/left/camera_info \
   /stereo_publisher/right/camera_info \
   /rtabmap/odom_info

```

matlabbe, 2022-10-06:
The rtabmap node seems frozen after:
```
[ INFO] [1664998234.839069868]: rtabmap: gen_depth  = false
```
rtabmapviz did wait 5 seconds, but it is like rtabmap node never continued after that line. Which rtabmap version are you using? Can you try launching `rtabmap.launch` alone on your host computer (with ROS_MASTER localhost), just to see if rtabmap can be correctly initialized. Normally rtabmap and stereo_odometry nodes should also output `Did not receive data since 5 seconds!` like rtabmapviz.

matlabbe, 2022-10-06:
The ros master is not even starting... what happens if you do only `roscore`?

Huynh-Anh, 2022-10-07:
@matlabbe tried again with my IP but the parameters do not seem to load. Do you have an idea on how i could resolved this ?
thank you

matlabbe, 2022-10-07:
It feels like a communication problem between the computers. These warnings are just on visualization side, rtabmap should be working. If you do "Edit->Preferences", it should trigger another try to get the parameters from rosparam. Maybe it is just rtabmap node takes more longer to initialize when the rosmaster is on your other computer. On your log above locally, rtabmap is up in 3 seconds. Normally it should also do it as fast even if ros master is remote, don't don't why it is stuck so long on your previous log.

an99990, 2022-10-10:
if i put rtabmapviz = false,
i dont get the warnings but rtabmap just hangs 

```
roslaunch rtabmap_ros rtabmap.launch args:="--delete_db_on_start" rgb_topic:=/stereo_inertial_publisher/color/image camera_info_topic:=/stereo_inertial_publisher/color/camera_info depth_topic:=/stereo_inertial_publisher/stereo/depth imu_topic:=/stereo_inertial_publisher/imu frame_id:=oak-d_frame approx_sync:=true wait_imu_to_init:=true
... logging to /home/huynh-anh/.ros/log/d8584b62-48e9-11ed-a7ff-bf1d1f317710/roslaunch-huynhanh-XPS-13-9350-6372.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.197:42779/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rgbd_odometry/
 * /rtabmap/rtabmap/

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /rtabmap/rgbd_odometry/approx_sync: True
 * /rtabmap/rgbd_odometry/approx_sync_max_interval: 0.0
 * /rtabmap/rgbd_odometry/config_path: 
 * /rtabmap/rgbd_odometry/expected_update_rate: 0.0
 * /rtabmap/rgbd_odometry/frame_id: oak-d_frame
 * /rtabmap/rgbd_odometry/ground_truth_base_frame_id: 
 * /rtabmap/rgbd_odometry/ground_truth_frame_id: 
 * /rtabmap/rgbd_odometry/guess_frame_id: 
 * /rtabmap/rgbd_odometry/guess_min_rotation: 0.0
 * /rtabmap/rgbd_odometry/guess_min_translation: 0.0
 * /rtabmap/rgbd_odometry/keep_color: False
 * /rtabmap/rgbd_odometry/max_update_rate: 0.0
 * /rtabmap/rgbd_odometry/odom_frame_id: odom
 * /rtabmap/rgbd_odometry/publish_tf: True
 * /rtabmap/rgbd_odometry/queue_size: 10
 * /rtabmap/rgbd_odometry/subscribe_rgbd: False
 * /rtabmap/rgbd_odometry/wait_for_transform_duration: 0.2
 * /rtabmap/rgbd_odometry/wait_imu_to_init: True
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: oak-d_frame
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_odom_info: True
 * /rtabmap/rtabmap/subscribe_rgb: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
[... 28 lines cut ...]
[ INFO] [1665441665.779918106]: Odometry: publish_tf             = true
[ INFO] [1665441665.779929103]: Odometry: wait_for_transform     = true
[ INFO] [1665441665.779953803]: Odometry: wait_for_transform_duration  = 0.200000
[ INFO] [1665441665.780002361]: Odometry: initial_pose           = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[ INFO] [1665441665.780024757]: Odometry: ground_truth_frame_id  = 
[ INFO] [1665441665.780050340]: Odometry: ground_truth_base_frame_id = 
[ INFO] [1665441665.780072903]: Odometry: config_path            = 
[ INFO] [1665441665.780095296]: Odometry: publish_null_when_lost = true
[ INFO] [1665441665.780118017]: Odometry: guess_frame_id         = 
[ INFO] [1665441665.780140862]: Odometry: guess_min_translation  = 0.000000
[ INFO] [1665441665.780164618]: Odometry: guess_min_rotation     = 0.000000
[ INFO] [1665441665.780176756]: Odometry: guess_min_time         = 0.000000
[ INFO] [1665441665.780186379]: Odometry: expected_update_rate   = 0.000000 Hz
[ INFO] [1665441665.780195796]: Odometry: max_update_rate        = 0.000000 Hz
[ INFO] [1665441665.780205634]: Odometry: wait_imu_to_init       = true
[ INFO] [1665441665.780243815]: Odometry: stereoParams_=0 visParams_=1 icpParams_=0
[ INFO] [1665441693.344812754]: rtabmap: frame_id      = oak-d_frame
[ INFO] [1665441693.345031162]: rtabmap: map_frame_id  = map
[ INFO] [1665441693.345147185]: rtabmap: use_action_for_goal  = false
[ INFO] [1665441693.345309924]: rtabmap: tf_delay      = 0.050000
[ INFO] [1665441693.345440743]: rtabmap: tf_tolerance  = 0.100000
[ INFO] [1665441693.345517199]: rtabmap: odom_sensor_sync   = false
[ INFO] [1665441695.604855075]: rtabmap: gen_scan  = false
[ INFO] [1665441695.604891863]: rtabmap: gen_depth  = false
```

an99990, 2022-10-15:
so now it is just never receving any messages, even if i receive them.
HOw do i set their timestamps ? @matlabbe thank you for your help
```
                  topic                        rate    min_delta   max_delta   std_dev   window
======================================================================= f=========================
/stereo_inertial_publisher/stereo/depth        0.4388   1.178       3.563       0.9043    8     
/stereo_inertial_publisher/color/camera_info   14.2     7.868e-06   2.365       0.1978    8     

                   topic                        rate   min_delta   max_delta   std_dev   window
===============================================================================================
/stereo_inertial_publisher/color/camera_info   14.15   7.868e-06   2.365       0.1967    297   

                   topic                        rate   min_delta   max_delta   std_dev   window
===============================================================================================
/stereo_inertial_publisher/color/camera_info   14.04   7.868e-06   2.365       0.194     314   

                   topic                        rate   min_delta   max_delta   std_dev   window
===============================================================================================
/stereo_inertial_publisher/color/camera_info   14.31   7.868e-06   2.365       0.1897    331   

                   topic                        rate    min_delta   max_delta   std_dev   window
================================================================================================
/stereo_inertial_publisher/stereo/depth        0.4403   0.2763      4.212       1.223     10    
/stereo_inertial_publisher/color/camera_info   14.67    7.868e-06   2.365       0.1848    10    

                   topic                        rate   min_delta   max_delta   std_dev   window
===============================================================================================
/stereo_inertial_publisher/color/camera_info   15.44   7.868e-06   2.365       0.1788    387   

                   topic                        rate   min_delta   max_delta   std_dev   window
===============================================================================================
/stereo_inertial_publisher/color/camera_info   15.19   7.868e-06   2.365       0.1795    401   

                   topic                        rate   min_delta   max_delta   std_dev   window
===============================================================================================
/stereo_inertial_publisher/color/camera_info   14.99   7.868e-06   2.365       0.1797    408 
```

matlabbe, 2022-10-23:
> so now it is just never receving any messages, even if i receive them.

From the log above, the topic rates are inconsistent. You should fix this before using rtabmap.

an99990, 2022-10-24:
is it possible to fix this by using the rgbd_sync argument ?

matlabbe, 2022-11-26:
No, `rgbd_sync` won't be able to synchronize if frequency is too different between the input topics.