introlab/rtabmap_ros #822: Weird TF behaviour if odom --> frame_id already exists.

rakshith95, 2022-10-07:
Hello @matlabbe,  
I'm seeing some possibly weird `tf` behaviour when the `odom --> base_footprint` transform is given by some other node. I've attached the launch file I'm using.  
[stereo_rtabmap_inria.launch.txt](https://github.com/introlab/rtabmap_ros/files/9732266/stereo_rtabmap_inria.launch.txt)

I run it as two cases:
1.  with `vo_frame_id:=odom_random` and there is no node which publishes `odom`-->`base_link` and the tf_tree is as below: 
[image]

2. with `vo_frame_id:=odom_random` and there is a node which publishes `odom`-->`base_link`; tf_tree below:
[image]

I don't understand why, in the second case, there is no edge connecting `odom_random` and `base_footprint`.  That there is a tf between a frame called `odom` --> `base_footprint` seems to prevent the `/stereo_odometry` node from sending the tf from `vo_frame_id` --> `frame_id` which in this case is `odom_random` --> `base_footprint`.

Thank you!

matlabbe, 2022-10-07:
You cannot have two parents for same child transform. To not break your tree, set `odom_guess_frame_id` to `odom`. You will then get `map`->`odom_random`->`odom`->`base_footprint`.

rakshith95, 2022-10-10:
@matlabbe this seems to fix the tf tree, but the point cloud is still mis-aligned (the projected plane does not align with the ground) when I start the mapping, which is not the case when there is no existing `odom`-->`base_footprint`.  
[image]

rakshith95, 2022-10-11:
If I start rtabmap after the other nodes have started, it seems to be aligned fine, so I just added a 'sleep' to my script to make sure it starts last. Not really sure why the problem occurs in the first place though.

closed 2022-10-11 by rakshith95