introlab/rtabmap_ros #824: how to use rgbd topic from rgbd_sync in rtabmap launch

an99990, 2022-10-15:
Hi
I am trying to remotely map an area. I saw that to avoid issues with synchronization, i can use the rgbd synchronization. The output topic rgbd image is then published.
BUt how do i feed this topic to the rtabmap launch ?
I tried to remap the topics but it doesnt work 
theres a rgbd_topic argument but when i put the rgbd topic is doesnt remap ..
please help

heres how the launch file looks like

```
<!-- -->
<launch>
  <!-- Convenience launch file to launch odometry, rtabmap and rtabmapviz nodes at once -->

  <!-- For stereo:=false
        Your RGB-D sensor should be already started with "depth_registration:=true".
        Examples:
           $ roslaunch freenect_launch freenect.launch depth_registration:=true 
           $ roslaunch openni2_launch openni2.launch depth_registration:=true -->
           
  <!-- For stereo:=true
        Your camera should be calibrated and publishing rectified left and right 
        images + corresponding camera_info msgs. You can use stereo_image_proc for image rectification.
        Example:
           $ roslaunch rtabmap_ros bumblebee.launch -->
     
  <!-- Choose between depth and stereo, set both to false to do only scan -->      
  <arg name="stereo"                    default="false"/>
  <arg     if="$(arg stereo)" name="depth"  default="false"/>
  <arg unless="$(arg stereo)" name="depth"  default="true"/>
  <arg name="subscribe_rgb"  default="$(arg depth)"/>
 
  <!-- Choose visualization -->
  <arg name="rtabmapviz"              default="false" /> 
  <arg name="rviz"                    default="false" />
  
  <!-- Localization-only mode -->
  <arg name="localization"            default="false"/>
  
  <!-- sim time for convenience, if playing a rosbag -->
  <arg name="use_sim_time"            default="false"/>
  <param if="$(arg use_sim_time)" name="use_sim_time" value="true"/>
  
  <!-- Corresponding config files -->
  <arg name="cfg"                     default="" /> <!-- To change RTAB-Map's parameters, set the path of config file (*.ini) generated by the standalone app -->
  <arg name="gui_cfg"                 default="~/.ros/rtabmap_gui.ini" />
  <arg name="rviz_cfg"                default="$(find rtabmap_ros)/launch/config/rgbd.rviz" />
  
  <arg name="frame_id"                default="camera_link"/>     <!-- Fixed frame id, you may set "base_link" or "base_footprint" if they are published -->
  <arg name="odom_frame_id"           default=""/>                <!-- If set, TF is used to get odometry instead of the topic -->
  <arg name="odom_frame_id_init"      default=""/>                <!-- If set, TF map->odom is published even if no odometry topic has been received yet. The frame id should match the one in the topic. -->
  <arg name="map_frame_id"            default="map"/>
  <arg name="ground_truth_frame_id"   default=""/>     <!-- e.g., "world" -->
  <arg name="ground_truth_base_frame_id" default=""/>  <!-- e.g., "tracker", a fake frame matching the frame "frame_id" (but on different TF tree) -->
  <arg name="namespace"               default="rtabmap"/>
  <arg name="database_path"           default="~/.ros/rtabmap.db"/>
  <arg name="queue_size"              default="10"/>
  <arg name="wait_for_transform"      default="0.2"/>
  <arg name="args"                    default=""/>              <!-- delete_db_on_start, udebug -->
  <arg name="rtabmap_args"            default="$(arg args)"/>   <!-- deprecated, use "args" argument -->
  <arg name="gdb"                     default="false"/>         <!-- Launch nodes in gdb for debugging (apt install xterm gdb) -->
  <arg     if="$(arg gdb)" name="launch_prefix" default="xterm -e gdb -q -ex run --args"/>
  <arg unless="$(arg gdb)" name="launch_prefix" default=""/>
  <arg name="clear_params"            default="true"/>
  <arg name="output"                  default="screen"/>        <!-- Control node output (screen or log) -->
  <arg name="publish_tf_map"          default="true"/>

  <!-- if timestamps of the input topics are synchronized using approximate or exact time policy-->
  <arg     if="$(arg stereo)" name="approx_sync"  default="false"/>
  <arg unless="$(arg stereo)" name="approx_sync"  default="$(arg depth)"/>     
[... 378 lines cut ...]
    </node>
  
  </group>
  
  <!-- Visualization RVIZ -->
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(arg rviz_cfg)"/>
  <node if="$(arg rviz)" pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_ros/point_cloud_xyzrgb" clear_params="$(arg clear_params)" output="$(arg output)">
    <remap if="$(arg stereo)" from="left/image"        to="$(arg left_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="right/image"       to="$(arg right_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="left/camera_info"  to="$(arg left_camera_info_topic)"/>
    <remap if="$(arg stereo)" from="right/camera_info" to="$(arg right_camera_info_topic)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/image"         to="$(arg rgb_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="depth/image"       to="$(arg depth_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/camera_info"   to="$(arg camera_info_topic)"/>
    <remap from="rgbd_image"        to="$(arg rgbd_topic_relay)"/>
    <remap from="cloud"             to="voxel_cloud" />

    <param name="decimation"  type="double" value="4"/>
    <param name="voxel_size"  type="double" value="0.0"/>
    <param name="approx_sync" type="bool"   value="$(arg approx_sync)"/>
    <param name="approx_sync_max_interval" type="double" value="$(arg approx_sync_max_interval)"/>
  </node>

</launch>
```

an99990, 2022-10-15:
looking at this http://wiki.ros.org/rtabmap_ros/Tutorials/RemoteMapping
I found the launch argument to be :
`roslaunch rtabmap_ros rtabmap.launch subscribe_rgbd:=true rgbd_topic:=/rgbd_image compressed:=true rtabmap_args:="--delete_db_on_start"`

however, when i try to give the /rgbd_image/compressed, it aks for a rgb_relay topic and if i used only rgb_image
i can warnings abou tfs

```
3) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.20098 timeout was 0.2."
[ WARN] [1665869101.650937227]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1665869100.748962) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.200409 timeout was 0.2."
[ WARN] [1665869101.853397163]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1665869101.082090) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.200388 timeout was 0.2."
[ WARN] [1665869102.056603413]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1665869101.282010) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.201294 timeout was 0.2."
[ WARN] [1665869102.260111045]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1665869101.481876) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.201836 timeout was 0.2."
[ WARN] [1665869102.466545671]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1665869101.681795) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.202081 timeout was 0.2."
```

matlabbe, 2022-10-23:
`rtabmap.launch` is using by default `frame_id:=camera_link`, if you don't have `camera_link` frame, just add `frame_id:=my_base_frame`. For oakd, in examples [here](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping), we set `frame_id:=oak-d_frame`