introlab/rtabmap_ros #826: rtabmap + depthacore issue

an99990, 2022-10-19:
Hi
I have built rtabmap with DEPTHAI_CORE successfully, and depthai_core is installed with opencv support as well.

[image]

However when i am launching rtabmap im getting this issue

`roslaunch rtabmap_ros rtabmap.launch subscribe_rgbd:=true rgbd_topic:=/rgbd_image compressed:=true rtabmap_args:="--delete_db_on_start"`

error log :

```
[rtabmap/rgbd_odometry-3] process has died [pid 3224, exit code 127, cmd /home/huynh-anh/catkin_ws/devel/lib/rtabmap_ros/rgbd_odometry --delete_db_on_start rgb/image:=/camera/rgb/image_rect_color_relay depth/image:=/camera/depth_registered/image_raw_relay rgb/camera_info:=/camera/rgb/camera_info rgbd_image:=/rgbd_image_relay odom:=odom imu:=/imu/data __name:=rgbd_odometry __log:=/home/huynh-anh/.ros/log/fc9b71e6-4ff3-11ed-9a83-74dfbf308a97/rtabmap-rgbd_odometry-3.log].
log file: /home/huynh-anh/.ros/log/fc9b71e6-4ff3-11ed-9a83-74dfbf308a97/rtabmap-rgbd_odometry-3*.log
[rtabmap/rtabmap-4] process has died [pid 3239, exit code 127, cmd /home/huynh-anh/catkin_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=/camera/rgb/image_rect_color_relay depth/image:=/camera/depth_registered/image_raw_relay rgb/camera_info:=/camera/rgb/camera_info rgbd_image:=/rgbd_image_relay left/image_rect:=/stereo_camera/left/image_rect_color_relay right/image_rect:=/stereo_camera/right/image_rect_relay left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud scan_descriptor:=/scan_descriptor user_data:=/user_data user_data_async:=/user_data_async gps/fix:=/gps/fix tag_detections:=/tag_detections fiducial_transforms:=/fiducial_transforms odom:=odom imu:=/imu/data __name:=rtabmap __log:=/home/huynh-anh/.ros/log/fc9b71e6-4ff3-11ed-9a83-74dfbf308a97/rtabmap-rtabmap-4.log].
log file: /home/huynh-anh/.ros/log/fc9b71e6-4ff3-11ed-9a83-74dfbf308a97/rtabmap-rtabmap-4*.log
[ERROR] [1666214486.093176240]: Failed to load nodelet [/rtabmap/republish_rgbd_image] of type [rtabmap_ros/rgbd_relay] even after refreshing the cache: Failed to load library /home/huynh-anh/catkin_ws/devel/lib//librtabmap_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libdepthai-opencv.so: cannot open shared object file: No such file or directory)
[ERROR] [1666214486.093281585]: The error before refreshing the cache was: Failed to load library /home/huynh-anh/catkin_ws/devel/lib//librtabmap_plugins.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libdepthai-opencv.so: cannot open shared object file: No such file or directory)
```

an99990, 2022-10-19:
After removing everything related to opencv3.2 and rebuilting my package i am encoutering issues while catkin_make.
does rtabmap suport opencv 4 ? @matlabbe 

`make[2]: *** No rule to make target '/usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.3.2.0', needed by '/home/huynh-anh/catkin_ws/devel/lib/libapriltag_ros_common.so'.  Stop.`

an99990, 2022-10-19:
solution : https://stackoverflow.com/questions/58484540/no-rule-to-make-target-libopencv-calib3d-so-3-2-0-but-opencv-3-4-1-installed-wh

but were back to the initial issue .. the issue is what ros installs a version with opencv 3.2 ..

an99990, 2022-10-21:
Update, upgraded my linux to 20.04. Installed ros noetic and opencv4.2 and i was able to succesfully install rtabmap with depthcore are=on AND rtabmap withdepthai=on as well. But i have this issue which is similair to this one : https://github.com/introlab/rtabmap/issues/742
But my message is slightly different.

```
rosrun rtabmap_ros rgbd_sync /rgb/image:=/stereo_inertial_publisher/color/image /depth/image:=/stereo_inertial_publisher/stereo/depth /rgb/camera_info:=/stereo_inertial_publisher/color/camera_info

roslaunch rtabmap_ros rtabmap.launch subscribe_rgbd:=true rgbd_topic:=/rgbd_image compressed:=true rtabmap_args:="--delete_db_on_start"

```

```

[ WARN] [1666386948.007141608]: /rtabmap/rgbd_sync: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rgbd_sync subscribed to (exact sync):
   /camera/rgb/image_rect_color_relay \
   /camera/depth_registered/image_raw_relay \
   /camera/rgb/camera_info
[ WARN] [1666386948.536583344]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1666386938.208084) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.201554 timeout was 0.2."
[ WARN] [1666386952.608202304]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1666386942.405908) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.201039 timeout was 0.2."
[ WARN] [1666386953.007540019]: /rtabmap/rgbd_sync: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rgbd_sync subscribed to (exact sync):
   /camera/rgb/image_rect_color_relay \
   /camera/depth_registered/image_raw_relay \
   /camera/rgb/camera_info
[ WARN] [1666386954.689534852]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1666386949.698168) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.201902 timeout was 0.2."
[ WARN] [1666386957.462905608]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1666386951.364040) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.201469 timeout was 0.2."
[ WARN] [1666386958.008208675]: /rtabmap/rgbd_sync: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rgbd_sync subscribed to (exact sync):
   /camera/rgb/image_rect_color_relay \
   /camera/depth_registered/image_raw_relay \
   /camera/rgb/camera_info
[ WARN] [1666386960.020094895]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1666386953.962526) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.201189 timeout was 0.2."
[ WARN] [1666386962.809159632]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1666386956.427963) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.20137 timeout was 0.2."
[ WARN] [1666386963.008807508]: /rtabmap/rgbd_sync: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rgbd_sync subscribed to (exact sync):
   /camera/rgb/image_rect_color_relay \
   /camera/depth_registered/image_raw_relay \
   /camera/rgb/camera_info
[ WARN] [1666386964.721851757]: odometry: Could not get transform from camera_link to oak_rgb_camera_optical_frame (stamp=1666386958.826662) after 0.200000 seconds ("wait_for_transform_duration"=0.200000)! Error="canTransform: target_frame camera_link does not exist.. canTransform returned after 0.200654
```

please help

matlabbe, 2022-10-23:
You don't have to use `rgbd_sync` and `rtabmap.launch` at the same time, because `rtabmap.lauch` can create a `rgbd_sync` inside if `rgbd_sync` argument is true:
```
rosrun rtabmap_ros rgbd_sync /rgb/image:=/stereo_inertial_publisher/color/image /depth/image:=/stereo_inertial_publisher/stereo/depth /rgb/camera_info:=/stereo_inertial_publisher/color/camera_info

roslaunch rtabmap_ros rtabmap.launch \
    subscribe_rgbd:=true \
    compressed:=true \
    rtabmap_args:="--delete_db_on_start" \
    rgbd_sync:=true \
    rgb_topic:=/stereo_inertial_publisher/color/image \
    depth_topic:=/stereo_inertial_publisher/stereo/depth \
    camera_info_topic:=/stereo_inertial_publisher/color/camera_info
```

If you are using depthai, you may also look at the depthai examples on this page http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping

closed 2022-10-24 by an99990