introlab/rtabmap_ros #827: Question: adding navigation

W4li8, 2022-10-20:
Hi all, 

I've been trying to implement navigation with the move_base package (first simply in 2d, later to be in 3d) for several days now and I'm starting to wonder if this is possible. The below pictures showcase my problem in the same area. The maps are generated from a bag, the 2d yaml/pgm version is then used separately and yields a valid path. Then, I'd like the same output trajectory (in 2d is ok at this stage) but from the environment generated by rtabmap and a realsense depth camera input (recorded in a rosbag). In the 3d case, the trajectory decides to go through a wall. Yet the move_base config are identical...

My question being, is it even possible to do what I'm trying to do with the chosen tools? Or what could be the problem here? Any idea is welcome :)

[image]

[image]

If this looks like it should work, please let me know what config files could help you help me.

Thanks!

matlabbe, 2022-10-23:
It is not really 3D in the second case, move_base is still using only the occupancy grid for planning (the point cloud is just for visualization). Maybe move_base is not subscribed to right map topic. You can show the global costmap generated from move_base to compare with the grid map you are visualizing in rviz. If the global costmap is empty, then it is likely that move_base will plan a straight line to goal.

If the bug is that "sometimes" it plan through the walls, this could be related to this issue: https://github.com/ros-planning/navigation/issues/320

matlabbe, 2022-11-26:
If you show only `move_base/global_costmap/costmap`, do you see something? You can change the colormap of `move_base/global_costmap/costmap` to `costmap` to see pink/blue/red colors over the white static map.

W4li8, 2022-11-28:
That is fine.

On a separate note, when using live camera input, any idea why the move_base costmaps have no publishers and rtabmap generates a map just fine? It runs in simulation

W4li8, 2022-11-30:
> 

Problem solved: I had wrongly configured and wrongly indented my yaml config files for move_base, probably combined with some topic mapping problems from previous tests.

closed 2022-11-30 by W4li8