introlab/rtabmap_ros #828: Question: GPS and IMU support, multiple RGB-D cameras

carlostojal, 2022-10-21:
Hi. First of all, sorry for asking this type of question here, given it's not properly an issue. I have some questions:

1) Checking noetic [documentation](http://wiki.ros.org/rtabmap_ros), I can't find references to any GPS and IMU topic. Was it deprecated? How did it impact mapping and graph generation and optimization? Which is the optimal way of feeding this data into RTAB-Map?

2) Which is the optimal method to use multiple RGB-D cameras on different orientations?

Thanks in advance.

matlabbe, 2022-10-23:
Hi, thx for the question, I've never updated the ros wiki for those topics. Just [did](http://wiki.ros.org/rtabmap_ros)! Here is what I added for imu and gps:

> imu ([sensor_msgs/Imu](http://docs.ros.org/en/api/sensor_msgs/html/msg/Imu.html))
> 
> Can be used to add gravity constraints in the graph. The whole map will then be aligned with gravity. The quaternion should be already computed (see [imu_filter_madgwick](http://wiki.ros.org/imu_filter_madgwick) to compute it).
> 
> gps/fix ([sensor_msgs/NavSatFix](http://docs.ros.org/en/api/sensor_msgs/html/msg/NavSatFix.html))
> 
>  Can be used to add GPS constraints in the graph. This could eventually be used to export poses in GPS format. With Optimizer/PriorsIgnored parameter set to false, the graph can be also optimized using the GPS values. See also this [page](https://github.com/introlab/rtabmap/wiki/Robust-Graph-Optimization#multi-session) on how GPS can be used to limit loop closure detection radius.

For multi-cameras, currently `rtabmap_ros` supports only multi stereo cameras or multi rgbd cameras, not multi mono cameras. If you have multiple RGB-D cameras or Stereo cameras, you should use `rgbd_sync` or `stereo_sync` nodes/nodelets to synchronize image topics per camera. You can then have multiple `rgbd_image` topics that can be connected to rtabmap node (set parameters `subscribe_rgbd` and `rgbd_cameras`). There is an example here: [demo_two_kinects.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/demo/demo_two_kinects.launch). To make it work properly, the TF between base link and the cameras should be accurate.

alexk1976, 2022-10-26:
@matlabbe Hello Mathieu, in previous forums you mentioned that  solvePnPRansac  (2D ->3D correspondence) and LBA are not supported when working with multiple RGBD cameras. Is it still the case or already supported? what should be implemented in order to support multi-cameras thoroughly?

matlabbe, 2022-11-07:
PnP for multi-camera is now supported in rtabmap. Just make sure rtabmap is built with [OpenGV](https://laurentkneip.github.io/opengv/) dependency.

alexk1976, 2022-11-11:
@matlabbe great and thank you for the amazing work! is your solution assuming all sensor clocks & images HW synced? are features/points/poses matched/solved concurrently or image from each camera matched/solved separately and HW sync is not critical?

matlabbe, 2022-11-13:
Assuming all cameras are hardware synchronized, but it should still work on systems where cameras are not hardware synchronized. I don't have a direct comparison though of how much accuracy is impacted.

alexk1976, 2022-12-01:
@matlabbe  When we build and run it with OpenGV - we get an immediate crash in rtabmap ([ERROR] [rtabmap-7]: process has died [pid 9805, exit code -6). Tried both with single and multiple cameras over ROS2/Foxy. Any direction?

matlabbe, 2022-12-04:
Can you run with gdb (section A of https://github.com/introlab/rtabmap_ros/issues/28#issue-99654759) and add "--udebug" argument to rtabmap node?

alexk1976, 2022-12-22:
Crash disappeared after we switched to use /tf instead of ros topic. Now dual-camera mode is working. Evaluating accuracy and will update.
thank you.

alexk1976, 2022-12-24:
@matlabbe can you explain in high-level multi-camera concept ? does rtabmap solve each camera image separately and fuse the result poses? or solving is done concurrently on both images and features from both cameras going to the solver ? thank you

matlabbe, 2022-12-30:
See "Non-central absolute pose" on this page: http://laurentkneip.github.io/opengv/page_how_to_use.html . So we assume `V_n` is fixed (assuming cameras trigger all at the exact same time). That will give the multi-camera "PnP" transform (like output of [cv::solvePnPRansac()](https://docs.opencv.org/4.x/d9/d0c/group__calib3d.html#ga50620f0e26e02caa2e9adc07b5fbf24e) for the single camera case), then bundle adjustment (g2o) with all cameras is applied to refine the transform (with `V_n` or `local_transform` fixed). We tried with multi-camera with approximate synchronization, the results were still good, but I don't have comparison results between a hardware synchronized cameras setup and a software synchronized setup.

dStanhill, 2023-01-01:
@matlabbe  As I understand it the above answer regarding multi-camera concept, refers mainly to visual odometry.
What is done in local occupancy grid creation in STM and in local loop closure?

matlabbe, 2023-01-09:
My comment https://github.com/introlab/rtabmap_ros/issues/828#issuecomment-1367978243 also applies for visual loop closure detection / local visual proximity detection. The local occupancy grid creation also support multi-cameras, each camera will be projected (with or without ray tracing) to create the local grids (this also applies for OctoMap). One caveat is that ray tracing will be done from the [average of viewpoints of the cameras](https://github.com/introlab/rtabmap/blob/98bf3fb184f89e8061b3d985f96a6c4f62ed6749/corelib/src/OccupancyGrid.cpp#L407-L411), not the camera themselves.

dStanhill, 2023-01-09:
I understand, so you mean that using all cameras can help declare certain cells as empty in the occupancy grid map, which will be left unknown when using the average camera?

matlabbe, 2023-01-15:
[image]
You can see on the right case that there will be wrongly empty cells ray traced.

alexk1976, 2023-01-17:
@matlabbe we just discovered that we run multiple camera launch file with EstimationType=0 (3D-3D estimation) following demo_two_kinects.launch. When we just tried to set EstimationType=1 (2D-3D estimation) but rtabmap crashed. can you confirm multi-camera was tested/supposed working with EstimationType=1? We'll try to debug. thanks

matlabbe, 2023-01-18:
2D->3D estimation should work with the changes from last summer (well, the integration of OpenGV was actually to do that). 3D->3D multi-camera was working before, it is using PCL.

If you have a bag and and corresponding launch file, I could try it here if I can see the same error.

alexk1976, 2023-01-19:
@matlabbe It looks crash happens sporadically on some recordings when working ok on others.
Attached crash reproduction with launch file and rosbag file.
instructions.txt specifies build params and the launch.

https://drive.google.com/drive/folders/1gBOcoA9rGioCVbXwQIH2xsP1TEEX9HtH?usp=sharing

Thank for for great support!

matlabbe, 2023-01-21:
I tried your instructions and I didn't have a crash.
[image]

Cleaning up the parameters (see [rtabmap_est_type_1_crash_param_cleanup.launch.zip](https://github.com/introlab/rtabmap_ros/files/10473285/rtabmap_est_type_1_crash_param_cleanup.launch.zip)):
```
ros2 launch rtabmap_est_type_1_crash_param_cleanup.launch.py \
    frame_id:=base_footprint \
    odom_topic:=/rtabmap/visual_odom_topic \
    qos:=1 \
    camera0:=camera0 \
    camera1:=camera1 \
    namespace:=rtabmap \
    rgbd_sync:=true \
    visual_odometry:=true \
    publish_tf_odom:=true \
    approx_sync:=true \
    rviz:=true \
    rtabmapviz:=true \
    args:="--delete_db_on_start \
        --Rtabmap/DetectionRate 1 \
        --Reg/Force3DoF true \
        --Grid/MaxObstacleHeight 1"
```

Default is using 2D->3D estimation. The biggest issues I see is the right camera, not sure if its orientation in TF is the right one, it gives a strange cloud at the beginning:
[image]

The tilt on the front camera's static TF seems not right too:
[image]

alexk1976, 2023-01-22:
hi @matlabbe , we took your params but still have sporadic crashes on this and some other clips. Verified on updated rtabmap and rtabmap_ros (latest humble-devel).
Did you also test it on humble-devel? 
Why do think our cameras TFs/orientation are wrong? Can you elaborate what to look/check?
thanks

matlabbe, 2023-01-22:
Oh you are on ubuntu 22.04. Humm, I'll need to setup github actions to generate the ros2 docker images so I can try it.

Look at my last [screenshot](https://github.com/introlab/rtabmap_ros/issues/828#issuecomment-1399351013) (taken with `rtabmap-databaseViewer` -> 3d View), this is a side view of the generate point cloud of the front camera in your robot base frame. We can clearly see that the camera is wrongly looking up, the bottom of the cloud should be aligned perfectly with the xy plane.

matlabbe, 2023-01-22:
Did you apply this [patch](https://gist.github.com/matlabbe/a412cf7c4627253874f81a00745a7fbb) on OpenGV to disable -march=native?

matlabbe, 2023-01-23:
Tried on Humble and still work without crashing. Make sure to apply the patch for opengv in my previous post.

alexk1976, 2023-01-25:
Hi @matlabbe , indeed this patch fixed the issue. Thank you!
Still we see a bit better accuracy when working with single camera.
Do you have any ideas what parameters to refine for dual-camera (Vis/MaxFeatures, DetectionRate - ?)

matlabbe, 2023-01-29:
Well, if you didn't fix the TF of your cameras

> We can clearly see that the camera is wrongly looking up, the bottom of the cloud should be aligned perfectly with the xy plane.

It will be worst for sure. Ideally, extrinsics calibration between all the cameras is required on multi-camera systems. Also, hardware synchronization between all cameras is preferable, but I have examples working fine even when using software approximate sync between the cameras. Otherwise, performance should be at least equal/often better/more robust to single camera, as the field of view is a lot larger.

julianraheema, 2023-03-14:
Hi @matlabbe, 
I have setup my ros noetic launch file for multiple cameras (rgbd). I have two cameras, the first one is located to rear of the robot and the second one is located to the side of the robot. In the rtabmap is showing only one camera pyramid, and not two as you are showing it in the screenshot, but once a while the pyramid relocate to the second position for a half second then go back to the rear camera position. Any ideas? Please see my screenshot.

the second question is how many rgbd cameras can the rtabmap synch?

[image]

Thank you

matlabbe, 2023-04-16:
If rtabmapviz is subscribing to odometry, the pyramids may not be correctly displayed.

With `rgbdx_sync`, up to 8 cameras.

alexk1976, 2023-05-13:
Hi @matlabbe , whether GBA and other post-processing methods were implemented over multiple-cameras?  thank you

matlabbe, 2023-05-19:
Would have to be tested, but `OptimizerG2O` supports multiple cameras. The post-processing functions may not have been tested/implemented with its support.

tehkillerbee, 2023-05-31:
> Hi, thx for the question, I've never updated the ros wiki for those topics. Just [did](http://wiki.ros.org/rtabmap_ros)! Here is what I added for imu and gps:
>
>
> For multi-cameras, currently [...]

FYI, The `gps/fix` and `imu` does not appear to be mentioned anywhere in the documentation listed here (http://wiki.ros.org/rtabmap_odom). Is this intentional? Has support been removed?

matlabbe, 2023-06-04:
Actually it has never been added to ros wiki for odometry part (it is for http://wiki.ros.org/rtabmap_slam). I just added it. http://wiki.ros.org/rtabmap_odom

tehkillerbee, 2023-06-06:
@matlabbe Thank you :)