introlab/rtabmap_ros #835: Could not transform IMU msg from frame "oak_imu_frame" to frame "oak_rgb_camera_optical_frame"

an99990, 2022-11-03:
Hi
I have recorded a rosbag in my rpi to do compile rtabmap on my pc. However I am encoutering issues when i launch rtabmap node.
I used the IMU filter suggested. This issue is not present in my rpi. I am thus wondering if there a topic missing or something that i need to record
[image]

```
roslaunch rtabmap_ros rtabmap.launch args:="--delete_db_on_start" \
rgb_topic:=/stereo_inertial_publisher/color/image \
camera_info_topic:=/stereo_inertial_publisher/color/camera_info \
depth_topic:=/stereo_inertial_publisher/stereo/depth \
imu_topic:=/stereo_inertial_publisher/imu/data \
frame_id:=oak_rgb_camera_optical_frame \
approx_sync:=true \
wait_imu_to_init:=true
```

matlabbe, 2022-11-07:
Make sure to record `/tf` and `/tf_static` topics in your bag.

an99990, 2022-11-07:
Hi @matlabbe i did record them. But the tf_static topic only has one message. And tf doesn't show up even if i record it ..please help

matlabbe, 2022-11-26:
Make sure all nodes are started before starting the bag so that the first tf_static message is not missed by those nodes:
```
roscore
rosparam set use_sim_time true

# Start in pause mode:
rosbag play --clock --pause mybag.bag

# launch all nodes here
...

# then go back on rosbag terminal, press "space" to start it.

```