introlab/rtabmap_ros #836: TORO,G2O,GTSAM support for Melodic

tugbakara, 2022-11-08:
Hi @matlabbe !
I have two main question which are about optimizers and realsense sdk.

**1.** When I am mapping with TORO optimizer I got warnings that's mentioned question from ros [page](https://answers.ros.org/question/343139/rtabmap-g2o-gtsam/). And also when I run command:
```
~$ rtabmap --version
RTAB-Map:              0.20.21
PCL:                     1.8.1
With VTK:                6.3.0
OpenCV:                  3.2.0
With OpenCV xfeatures2d: false
With OpenCV nonfree:     false
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With OpenGV:             false
With Madgwick:            true
With PDAL:               false
With TORO:                true
With g2o:                 true
With GTSAM:              false
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI2:             true
With Freenect:            true
With Freenect2:          false
With K4W2:               false
With K4A:                false
With DC1394:              true
With FlyCapture2:        false
With ZED:                false
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:         false
With MYNT EYE S:         false
With DepthAI:            false
With libpointmatcher:     true
With CCCoreLib:          false
With octomap:             true
With cpu-tsdf:           false
With open chisel:        false
With Alice Vision:       false
With LOAM:               false
With FLOAM:              false
With FOVIS:              false
With Viso2:              false
With DVO:                false
With ORB_SLAM:           false
With OKVIS:              false
With MSCKF_VIO:          false
With VINS-Fusion:        false
With OpenVINS:           false
```

I can see that there are TORO and g2o optimizers but when I open TORO optimizer, I saw that kinda WARN:
`
OptimizerTORO.cpp:151::optimize() TORO optimizer doesn't support prior or gravity links, use GTSAM or g2o optimizers (see parameter Optimizer/Strategy). Link 1 ignored...
`

When I pass the g2o , this warning disappears but another warning arises:
`OptimizerG2O.cpp:1211::optimize() Computing marginals: vertex 14 has negative hessian index (-1). Cannot compute last pose covariance.
`

**2.**  I installed realsense 2.50 SDK then I made cmake with specific flags in rtabmap build folder (first of all I deleted this folder ) but in rtabmap --version there isnot any realsense looks true. I don't know why rtabmap cannot see realsense?

matlabbe, 2022-11-27:
Can you share the database causing "OptimizerG2O.cpp:1211::optimize() Computing marginals: vertex 14 has negative hessian index (-1). Cannot compute last pose covariance. " to see if there is a bug? For TORO warning, it is fine as it cannot support priors.

CMake should be able to find `realsense2Config.cmake` to add to rtabmap dependency. You can explicitly set `-Drealsense2_DIR=/path/to/folder/containing/realsense2Config.cmake`., though normally if realsense2 is installed from system binaries or installed in `/usr/local`, cmake should be able to find it.

tugbakara, 2022-11-28:
Hi again!

> Can you share the database causing "OptimizerG2O.cpp:1211::optimize() Computing marginals: vertex 14 has negative hessian index (-1). Cannot compute last pose covariance. " to see if there is a bug (or just wrong input constraints)? For TORO warning, it is fine as it cannot support priors.

I didn't take these errors anymore, I reinstalled rtabmap three or four days ago, maybe the issue comes from binary installation now I don't have bin installation. But I got different error : `util3d_registration.cpp:186::transformFromXYZCorrespondences() RANSAC refineModel: Refinement failed: got an empty set of inliers!`
here is my [db](https://drive.google.com/file/d/1_hy4Plec9n4nqrQXuGRoCNnGO80pkEi3/view?usp=share_link).
> installed in `/usr/local`, cmake should be able to find it.
and SDK is in this path.

matlabbe, 2022-11-30:
In that database, you have **Identity** covariance for odometry, this won't work very well with gtsam or g2o. Probably the reason you got "Computing marginals" errors. If your robot doesn't have covariance set correctly, you may set `odom_frame_id` instead in  parameters of rtabmap node, and some arbitrary covariance will be set for you.

xinjuezou-whi, 2025-12-23:
> On ros1:
> 
> ```
> rosrun rtabmap_slam rtabmap _odom_frame_id:="odom" _odom_tf_linear_variance:=0.001 _odom_tf_angular_variance:=0.001 ...
> ```
> 
> On ros2:
> 
> ```
> ros2 run rtabmap_slam rtabmap --ros-args -p odom_frame_id:="odom" -p  odom_tf_linear_variance:=0.001 -p odom_tf_angular_variance:=0.001 ...
> ```

clear to me now. thank you

closed 2026-01-06 by matlabbe