introlab/rtabmap_ros #840: TF of received scan cloud at time

VictorJCS, 2022-11-14:
Hi,
I am using OS0-64-U (Lidar sensor) and Rtabmap package. Actually, in normal conditions the ouster node and Rtabmap node running without problem.
[image]

The odom and os_sensor frames looks like this:
[image]

But, I have problems when I try to do a test:

1. Ouster node and Rtabmap is running
2. I disconnect Lidar sensor
3. Ouster node and Rtabmap node continue running without data
4. After 2 minutes I connect Lidar sensor

The nodes reprocess Lidar data but I have the following error:

[ WARN] [1668430863.431583479]: odometry: Could not get transform from base_link to os_sensor (stamp=1668430841.862735) after 0.500000 seconds ("wait_for_transform_duration"=0.500000)! Error="Lookup would require extrapolation 21.063359355s into the past.  Requested time 1668430841.862734795 but the earliest data is at time 1668430862.926094294, when looking up transform from frame [os_sensor] to frame [base_link]. canTransform returned after 0.50219 timeout was 0.5."
[ERROR] [1668430863.431683692]: TF of received scan cloud at time 1668430841.862735s is not set, aborting rtabmap update.
[ WARN] (2022-11-14 14:01:03.564) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.797895/0.78 tr: 1.6155/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.797493,0.761473,1.929353)                                                                                                                                                                                                                                 
[ INFO] [1668430863.567594919]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.024402s
[ERROR] (2022-11-14 14:01:03.570) Rtabmap.cpp:1326::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 2115 is ignored!
[ INFO] [1668430863.570633656]: rtabmap (361): Rate=0.25s, Limit=0.000s, Conversion=0.0032s, RTAB-Map=0.0006s, Maps update=0.0000s pub=0.0000s (local map=29, WM=29)
[ WARN] (2022-11-14 14:01:03.736) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.87967/0.78 tr: 3.26331/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.798531,0.766531,1.927925)                                                                                                                                                                                                                                 
[ INFO] [1668430863.737157064]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.063009s
[ WARN] (2022-11-14 14:01:03.908) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.814037/0.78 tr: 0.926423/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.800887,0.775835,1.925019)                                                                                                                                                                                                                               
[ INFO] [1668430863.909349727]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.030619s
[ERROR] (2022-11-14 14:01:03.934) Rtabmap.cpp:1326::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 2118 is ignored!
[ INFO] [1668430863.934681013]: rtabmap (361): Rate=0.25s, Limit=0.000s, Conversion=0.0025s, RTAB-Map=0.0002s, Maps update=0.0000s pub=0.0000s (local map=29, WM=29)
[ WARN] (2022-11-14 14:01:04.065) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.793545/0.78 tr: 0.374791/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.803496,0.785809,1.921865)                                                                                                                                                                                                                               
[ INFO] [1668430864.070119826]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.028675s
[ WARN] (2022-11-14 14:01:04.238) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.948769/0.78 tr: 2.03662/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.806273,0.789994,1.921502)                                                                                                                                                                                                                                
[ERROR] (2022-11-14 14:01:04.241) Rtabmap.cpp:1326::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 2121 is ignored!
[ INFO] [1668430864.241144736]: rtabmap (361): Rate=0.25s, Limit=0.000s, Conversion=0.0012s, RTAB-Map=0.0001s, Maps update=0.0000s pub=0.0000s (local map=29, WM=29)
[ INFO] [1668430864.242279130]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.043019s
[ WARN] (2022-11-14 14:01:04.454) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.621641/0.78 tr: 3.01912/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.808487,0.799845,1.919126)                                                                                                                                                                                                                                
[ INFO] [1668430864.456853015]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.057996s
[ WARN] (2022-11-14 14:01:04.694) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.783879/0.78 tr: 2.61371/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.811433,0.809174,1.917065)                                                                                                                                                                                                                                
[ INFO] [1668430864.695184685]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.094303s
[ERROR] (2022-11-14 14:01:04.697) Rtabmap.cpp:1326::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 2125 is ignored!
[ INFO] [1668430864.697549149]: rtabmap (361): Rate=0.25s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.0002s, Maps update=0.0000s pub=0.0000s (local map=29, WM=29)
[ WARN] (2022-11-14 14:01:04.967) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.875851/0.78 tr: 3.36318/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.816697,0.823881,1.913866)                                                                                                                                                                                                                                
[ INFO] [1668430864.971425224]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.066398s
[ERROR] (2022-11-14 14:01:04.972) Rtabmap.cpp:1326::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 2128 is ignored!
[ INFO] [1668430864.972244719]: rtabmap (361): Rate=0.25s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.0001s, Maps update=0.0000s pub=0.0000s (local map=29, WM=29)
[ WARN] (2022-11-14 14:01:05.236) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.804123/0.78 tr: 1.37214/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.819398,0.834430,1.911678)                                                                                                                                                                                                                                
[ INFO] [1668430865.239351242]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.060249s
[ WARN] (2022-11-14 14:01:05.456) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.805016/0.78 tr: 2.96419/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.824529,0.850720,1.908587)                                                                                                                                                                                                                                
[... cut ...]
[ INFO] [1668430875.735581323]: rtabmap (365): Rate=0.25s, Limit=0.000s, Conversion=0.0044s, RTAB-Map=2.1147s, Maps update=0.0011s pub=0.0003s (local map=32, WM=32)

The point cloud looks like this:
[image]

The frames look like this:
[image]

At first I thought it was a synchronization problem with the sensor. [https://github.com/ouster-lidar/ouster-ros/issues/9](url)
I have adjusted the synchronization parameters and I am using new timestamp_mode TIME_FROM_ROS_TIME, but I still have the same problem.

Can you help me please?

matlabbe, 2022-11-19:
Make sure the sensor doesn't move while you disconnect it, otherwise icp_odometry will have a hard time to register correctly to odometry's local map so far. Based on your logs, it seems the sensor moved or the timestamps in new lidar scans are not 2 min after the disconnection (stamps reset?).

VictorJCS, 2022-11-21:
Hi @matlabbe 
Thats true, this error log does not correspond to 2 minutes. But the problem is for any disconnection time and when the sensor does not move. Here is another error log for 20 seconds (disconnection time).

[ INFO] [1669023725.629511988]: Odom: ratio=0.966178, std dev=0.007773m|0.002458rad, update time=0.154633s
[ INFO] [1669023725.837837988]: Odom: ratio=0.967638, std dev=0.007701m|0.002435rad, update time=0.122332s
[ INFO] [1669023725.963663835]: rtabmap (652): Rate=0.25s, Limit=0.000s, Conversion=0.0052s, RTAB-Map=0.3174s, Maps update=0.1167s pub=0.0002s (local map=31, WM=31)
[ INFO] [1669023726.042116792]: Odom: ratio=0.966606, std dev=0.007914m|0.002503rad, update time=0.121966s
[ INFO] [1669023726.218811549]: Odom: ratio=0.964495, std dev=0.007924m|0.002506rad, update time=0.097783s
[ INFO] [1669023726.254816203]: rtabmap (653): Rate=0.25s, Limit=0.000s, Conversion=0.0029s, RTAB-Map=0.2199s, Maps update=0.0592s pub=0.0002s (local map=31, WM=31)
[ INFO] [1669023726.497714477]: rtabmap (654): Rate=0.25s, Limit=0.000s, Conversion=0.0011s, RTAB-Map=0.1526s, Maps update=0.0670s pub=0.0001s (local map=31, WM=31)
[ INFO] [1669023726.723775142]: rtabmap (655): Rate=0.25s, Limit=0.000s, Conversion=0.0013s, RTAB-Map=0.1612s, Maps update=0.0610s pub=0.0002s (local map=31, WM=31)
[ INFO] [1669023726.957848961]: rtabmap (656): Rate=0.25s, Limit=0.000s, Conversion=0.0021s, RTAB-Map=0.1639s, Maps update=0.0672s pub=0.0002s (local map=31, WM=31)
[ WARN] [1669023748.768555410]: odometry: Could not get transform from base_link to os_sensor (stamp=1669023725.908860) after 0.500000 seconds ("wait_for_transform_duration"=0.500000)! Error="Lookup would require extrapolation 22.342833655s into the past.  Requested time 1669023725.908859968 but the earliest data is at time 1669023748.251693726, when looking up transform from frame [os_sensor] to frame [base_link]. canTransform returned after 0.500757 timeout was 0.5."                                                                                                                                                                                                                                           
[ERROR] [1669023748.768694839]: TF of received scan cloud at time 1669023725.908860s is not set, aborting rtabmap update.
[ WARN] (2022-11-21 10:42:28.858) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 1.0648/0.78 tr: 2.17465/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.899096,0.642949,1.943187)                                                                                                                                                                                                                                  
[ INFO] [1669023748.859388175]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.021517s
[ERROR] (2022-11-21 10:42:28.861) Rtabmap.cpp:1326::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 3092 is ignored!
[ INFO] [1669023748.861354924]: rtabmap (656): Rate=0.25s, Limit=0.000s, Conversion=0.0016s, RTAB-Map=0.0003s, Maps update=0.0000s pub=0.0000s (local map=31, WM=31)
[ WARN] (2022-11-21 10:42:29.006) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.358934/0.78 tr: 3.11069/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.899791,0.647377,1.942522)                                                                                                                                                                                                                                
[ INFO] [1669023749.006795702]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.046781s
[ WARN] (2022-11-21 10:42:29.121) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.82098/0.78 tr: 1.69965/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.900843,0.651815,1.941917)                                                                                                                                                                                                                                 
[ INFO] [1669023749.122830264]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.033041s
[ INFO] [1669023749.392978767]: Odom: ratio=0.052216, std dev=0.197382m|0.062418rad, update time=0.169400s
[ INFO] [1669023749.701774941]: Odom: ratio=0.979363, std dev=0.007664m|0.002423rad, update time=0.201533s
[ INFO] [1669023749.992555144]: Odom: ratio=0.984018, std dev=0.007769m|0.002457rad, update time=0.189718s
[ INFO] [1669023750.307430981]: Odom: ratio=0.979481, std dev=0.007477m|0.002364rad, update time=0.223137s
[ INFO] [1669023750.631522384]: Odom: ratio=0.980008, std dev=0.007900m|0.002498rad, update time=0.223261s
[ INFO] [1669023750.985732353]: Odom: ratio=0.979048, std dev=0.008153m|0.002578rad, update time=0.230489s
[ INFO] [1669023751.053143680]: rtabmap (657): Rate=0.25s, Limit=0.000s, Conversion=0.0022s, RTAB-Map=1.6394s, Maps update=0.0141s pub=0.0005s (local map=32, WM=32)
[ INFO] [1669023751.351315306]: Odom: ratio=0.974827, std dev=0.008091m|0.002559rad, update time=0.262401s
[ INFO] [1669023751.616232398]: Odom: ratio=0.972304, std dev=0.008098m|0.002561rad, update time=0.192796s
[... cut ...]
[ INFO] [1669023762.914268469]: Odom: ratio=0.972102, std dev=0.008243m|0.002607rad, update time=0.291697s
[ INFO] [1669023763.287813523]: Odom: ratio=0.969356, std dev=0.008184m|0.002588rad, update time=0.238913s
[ INFO] [1669023763.576288528]: Odom: ratio=0.965073, std dev=0.008090m|0.002558rad, update time=0.189178s
[ INFO] [1669023763.890849566]: Odom: ratio=0.972558, std dev=0.008364m|0.002645rad, update time=0.216973s
[ INFO] [1669023764.236384689]: Odom: ratio=0.972499, std dev=0.008284m|0.002620rad, update time=0.231730s

[image]

matlabbe, 2022-11-27:
I'll try to comment the log:

-------

[ INFO] [1669023726.957848961]: rtabmap (656): Rate=0.25s, Limit=0.000s, Conversion=0.0021s, RTAB-Map=0.1639s, Maps update=0.0672s pub=0.0002s (local map=31, WM=31)

**Here the connection is lost, no scans are published for 20 sec...**

**Odometry received a scan with stamp 22 sec in the past, so it is discarded as TF is now 22 seconds after the disconnection:**
[ WARN] [1669023748.768555410]: odometry: Could not get transform from base_link to os_sensor (stamp=1669023725.908860) after 0.500000 seconds ("wait_for_transform_duration"=0.500000)! Error="Lookup would require extrapolation 22.342833655s into the past. Requested time 1669023725.908859968 but the earliest data is at time 1669023748.251693726, when looking up transform from frame [os_sensor] to frame [base_link]. canTransform returned after 0.500757 timeout was 0.5."
[ERROR] [1669023748.768694839]: TF of received scan cloud at time 1669023725.908860s is not set, aborting rtabmap update.

**Odometry recevied another scan, is able to get TF and compute a transform. However it estimate that the robot moved more than 1.06 radians and 2.17 meters. Based on maximum expected rotation between consecutiev scans, the transform is rejected, ssuming ICP diverged for some reason:**
[ WARN] (2022-11-21 10:42:28.858) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 1.0648/0.78 tr: 2.17465/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.899096,0.642949,1.943187)

**Odometry is published as null, to notify rtabmap that odom could not be computed:**
[ INFO] [1669023748.859388175]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.021517s
[ERROR] (2022-11-21 10:42:28.861) Rtabmap.cpp:1326::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 3092 is ignored!
[ INFO] [1669023748.861354924]: rtabmap (656): Rate=0.25s, Limit=0.000s, Conversion=0.0016s, RTAB-Map=0.0003s, Maps update=0.0000s pub=0.0000s (local map=31, WM=31)

**Again odometry failed because transform is out of limits**
[ WARN] (2022-11-21 10:42:29.006) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.358934/0.78 tr: 3.11069/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.899791,0.647377,1.942522)
[ INFO] [1669023749.006795702]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.046781s
[ WARN] (2022-11-21 10:42:29.121) OdometryF2M.cpp:561::computeTransform() Registration failed: "libpointmatcher has failed: limit out of bounds: rot: 0.82098/0.78 tr: 1.69965/3" (guess=xyz=0.000000,0.000000,0.000000 rpy=2.900843,0.651815,1.941917)
[ INFO] [1669023749.122830264]: Odom: ratio=0.000000, std dev=0.000000m|0.000000rad, update time=0.033041s

**ICP has been accepted from now on: However, as you said the odometry is wrong...**
[ INFO] [1669023749.392978767]: Odom: ratio=0.052216, std dev=0.197382m|0.062418rad, update time=0.169400s
[ INFO] [1669023749.701774941]: Odom: ratio=0.979363, std dev=0.007664m|0.002423rad, update time=0.201533s

------

I would be interested to see a rosbag showing the disconnection. IF the robot really didn't move during that 22 sec, normally ICP should have been able to recover correctly.

VictorJCS, 2022-12-23:
Hi @matlabbe 

Sorry for the delay in replying, in the following link you will find two bag files:
[https://drive.google.com/drive/folders/19wkfugI2jNDvA7RUzfPTOq-xbGV06pCT?usp=share_link](url)

- Test_1.bag: Disconnection time 22 seconds.
- Test_2.bag: Disconnection time 38 seconds.

matlabbe, 2022-12-30:
Hi,

The problem seems coming from the imu filter node. Here is the orientation before and after the disconnection (bag 1):
```
rostopic hz /os_cloud_node/imu/data
header: 
  seq: 9193
  stamp: 
    secs: 1671797933
    nsecs: 259817905
  frame_id: "os_imu"
orientation: 
  x: -0.001659615278838697
  y: 0.014982743069379613
  z: -0.6650405625816423
  w: 0.7466550831597785
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.005326322180158476
  y: 0.0014647385995435807
  z: -0.0023968449810713144
angular_velocity_covariance: [0.0006, 0.0, 0.0, 0.0, 0.0006, 0.0, 0.0, 0.0, 0.0006]
linear_acceleration: 
  x: -0.04548983154296875
  y: -0.16280571289062498
  z: 10.019733947753906
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
--- NEXT MSG: 36 sec later...
header: 
  seq: 9194
  stamp: 
    secs: 1671797969
    nsecs: 864432684
  frame_id: "os_imu"
orientation: 
  x: -0.9311884565118846
  y: -0.24178901914841666
  z: -0.2058763182165815
  w: 0.17900019630113587
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.012250541014364496
  y: -0.0006657902725198095
  z: -0.0038615835806148956
angular_velocity_covariance: [0.0006, 0.0, 0.0, 0.0, 0.0006, 0.0, 0.0, 0.0, 0.0006]
linear_acceleration: 
  x: 0.062249243164062495
  y: -0.4165910888671875
  z: 9.947907897949218
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]

```

I think the imu filter node (are you using [madgwick](http://wiki.ros.org/imu_filter_madgwick) or [complementary](http://wiki.ros.org/imu_complementary_filter?distro=noetic) filter?) is expecting a constant flow of imu data without interruption, otherwise the filter would diverge like this. If you expect lidar interruptions, I'll suggest to use another IMU directly connected to the computer (that would never disconnect). Otherwise, there is a `constant_dt` parameter on those imu filtering nodes that could be used to avoid very large 36 sec `dt` between two consecutive imu topics. Here the imu topic recomputed with `constant_dt:=0.001`:
```
header: 
  seq: 1822
  stamp: 
    secs: 1671797933
    nsecs: 259817905
  frame_id: "os_imu"
orientation: 
  x: -0.0013279212593349868
  y: 0.015157634839991878
  z: -0.9196207326677529
  w: 0.39251266320822276
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.005326322180158476
  y: 0.0014647385995435807
  z: -0.0023968449810713144
angular_velocity_covariance: [0.0006, 0.0, 0.0, 0.0, 0.0006, 0.0, 0.0, 0.0, 0.0006]
linear_acceleration: 
  x: -0.04548983154296875
  y: -0.16280571289062498
  z: 10.019733947753906
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
--- NEXT MSG: 36 sec later...
header: 
  seq: 1823
  stamp: 
    secs: 1671797969
    nsecs: 864432684
  frame_id: "os_imu"
orientation: 
  x: -0.0014216748653109272
  y: 0.015180461128007295
  z: -0.9196216182752298
  w: 0.39250937777506173
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 0.012250541014364496
  y: -0.0006657902725198095
  z: -0.0038615835806148956
angular_velocity_covariance: [0.0006, 0.0, 0.0, 0.0, 0.0006, 0.0, 0.0, 0.0, 0.0006]
linear_acceleration: 
  x: 0.062249243164062495
  y: -0.4165910888671875
  z: 9.947907897949218
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
```
Note how in this case the orientation is the same than before the disconnection.

csvicbot-7, 2023-05-02:
Thank you so much @matlabbe , I will close this issue.

closed 2023-05-02 by csvicbot-7