introlab/rtabmap_ros #842: Turtlebot RGBD Odometry Problem. 3D Map Freezing After Start.

AddyPete, 2022-11-17:
Hi I followed this tutorial on turtlebot on RGBD Mapping.
[](https://github.com/introlab/rtabmap_ros/blob/ros2/launch/ros2/turtlebot3_rgbd.launch.py)
and it works really great! I like it. 

However instead of using wheel encoders as Odom source, **I want to use the RGBD Camera as an odom source**. However when I launched it, the 3D Map in RTABMap is working in the first second but then it freezes after. When I check TF, I see the Map->Odom is publishing but Odom->Base footprint (which is the base frame) is not connected.

[image]
I already removed the TF Publishing Odom->Basefootprint by the wheel encoders in the turtlebot waffle SDF.
so that the Odom->Base Footprint will be published by RTABMap. However it freezes. How to fix this?

[image]

Note that There are no error messages in the terminal, I can see that ODOM are publishing well in Console.

**Here is a youtube video of the turtlebot 3D Map Freezing on RTABMAP during start:** [](https://www.youtube.com/watch?v=YHRvt70nQEo)

------------------------------------------------------------------------

Additionally, I want to merge this since this is also a Similar 3D Map Error. 
Also, I tried this with a Kinect 360, the Odometry in RTABMap is working but the 3D Map in the RTABMap is Freezing RED all throughout.

[image]

The error messages of this are: 

1. [rtabmapviz-3]          at line 133 in /tmp/binarydeb/ros-foxy-tf2-0.13.13/src/buffer_core.cpp
[rtabmapviz-3] Warning: Invalid frame ID "odom" passed to canTransform argument target_frame - frame does not exist
2. [rgbd_odometry-1] [ WARN] (2022-11-17 15:27:43.493) OdometryF2M.cpp:1287::computeTransform() 20 visual features required to initialize the odometry (only 875 extracted).
3. [rtabmap-2] [ERROR] (2022-11-17 15:27:25.875) Rtabmap.cpp:1324::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!

Thank you very much!

matlabbe, 2022-11-19:
I would not try visual odometry in simulation in this kind of environment (too simple, too textureless). In comparison see this simulated environment: https://www.youtube.com/watch?v=A487ybS7E4E

For the xbox360 test, see this page: https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lost-odometry-red-screens

AddyPete, 2022-11-19:
I see. Is it possible to use RTABMap's Visual Odometry using Intel realsense stereo Camera in a Mining Site like this? With 

[image]

matlabbe, 2022-11-19:
Yes, in stereo odometry mode it should work. In large open space like this, having larger baseline between the left and right cameras could improve accuracy, though not required (as it needs sometime more work to get calibration and hardware time synchronization right).

AddyPete, 2022-11-19:
okay thanks for the info!

AddyPete, 2022-11-21:
> I would not try visual odometry in simulation in this kind of environment (too simple, too textureless). In comparison see this simulated environment: https://www.youtube.com/watch?v=A487ybS7E4E [...]

Hello again! I tried RTABMap with Kinect 360 using the RTABMap App. (without ROS2)
and **it worked perfectly.** Pic below

[image]

But when I did it with ROS2, **the odometry is lost on start** as what I've explained earlier. Pic below

[image]

Video on Odometry Lost on Start. (Using ROS2): https://youtu.be/Ip0ul12_uQM

I have transformations from from base_link to sensors and I've broadcasted the right messages which is rgb, depth and rgb camera info which is used of RGBD Odometry

This is my launch parameters: 

ros2 launch rtabmap_ros rtabmap.launch.py visual_odometry:=true frame_id:=base_link approx_sync:=true

matlabbe, 2022-11-21:
Maybe a sync problem with the topics, which camera node are you using?

AddyPete, 2022-11-21:
> Maybe a sync problem with the topics, which camera node are you using?

That's what I initially thought.

I used this package to broadcast the required topics for RTABMap.
https://github.com/fadlio/kinect_ros2

I used this because this is the only ros2 kinect 360 driver I could find.

matlabbe, 2022-11-21:
I tried the package, had to make those changes to get the right `stamp` and `frame_id` in rgb topic header and to get registered depth image:
```diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 296ed9d..f6112bf 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -5,15 +5,18 @@ if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
   add_compile_options(-Wall -Wextra -Wpedantic)
 endif()
 
+set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} ${CMAKE_CURRENT_SOURCE_DIR})
+
 include_directories(include)
 
+
 # find dependencies
 find_package(ament_cmake REQUIRED)
 find_package(rclcpp_components REQUIRED)
 find_package(rclcpp REQUIRED)
 # find_package(Threads REQUIRED)
 
-find_package(libfreenect REQUIRED)
+find_package(Freenect REQUIRED)
 find_package(OpenCV REQUIRED)
 
 find_package(sensor_msgs REQUIRED)
@@ -34,8 +37,9 @@ ament_target_dependencies(kinect_ros2_component
   camera_info_manager
   image_transport
   OpenCV)
+include_directories(${Freenect_INCLUDE_DIRS})
 target_link_libraries(kinect_ros2_component
-  freenect)
+  ${Freenect_LIBRARY})
 rclcpp_components_register_nodes(kinect_ros2_component "kinect_ros2::KinectRosComponent")
 
 add_executable(kinect_ros2_node
diff --git a/include/kinect_ros2/kinect_ros2_component.hpp b/include/kinect_ros2/kinect_ros2_component.hpp
index 3a9f234..73b8e9e 100644
--- a/include/kinect_ros2/kinect_ros2_component.hpp
+++ b/include/kinect_ros2/kinect_ros2_component.hpp
@@ -3,7 +3,7 @@
 
 extern "C"
 {
-  #include "libfreenect/libfreenect.h"
+  #include "libfreenect.h"
 }
 #include "rclcpp/rclcpp.hpp"
 #include "camera_info_manager/camera_info_manager.hpp"
diff --git a/src/kinect_ros2_component.cpp b/src/kinect_ros2_component.cpp
index 8dca9c6..8322605 100644
--- a/src/kinect_ros2_component.cpp
+++ b/src/kinect_ros2_component.cpp
@@ -1,4 +1,5 @@
 #include "kinect_ros2/kinect_ros2_component.hpp"
+#include <ament_index_cpp/get_package_share_directory.hpp>
 
 using namespace std::chrono_literals;
 
@@ -18,14 +19,17 @@ KinectRosComponent::KinectRosComponent(const rclcpp::NodeOptions & options)
 {
   timer_ = create_wall_timer(1ms, std::bind(&KinectRosComponent::timer_callback, this));
 
+  // may throw ament_index_cpp::PackageNotFoundError exception
+  std::string package_share_directory = ament_index_cpp::get_package_share_directory("kinect_ros2");
+
   //todo: use parameters
   depth_info_manager_ = std::make_shared<camera_info_manager::CameraInfoManager>(
     this, "kinect",
-    "file:///home/lio/tcc_ws/install/kinect_ros2/share/kinect_ros2/cfg/calibration_depth.yaml");
+    "file://" + package_share_directory + "/cfg/calibration_depth.yaml");
 
   rgb_info_manager_ = std::make_shared<camera_info_manager::CameraInfoManager>(
     this, "kinect",
-    "file:///home/lio/tcc_ws/install/kinect_ros2/share/kinect_ros2/cfg/calibration_rgb.yaml");
+    "file://" + package_share_directory + "/cfg/calibration_rgb.yaml");
 
   rgb_info_ = rgb_info_manager_->getCameraInfo();
   rgb_info_.header.frame_id = "kinect_rgb";
@@ -64,7 +68,7 @@ KinectRosComponent::KinectRosComponent(const rclcpp::NodeOptions & options)
     freenect_set_depth_mode(
     fn_dev_, freenect_find_depth_mode(
       FREENECT_RESOLUTION_MEDIUM,
-      FREENECT_DEPTH_MM));
+      FREENECT_DEPTH_REGISTERED));
   if (ret < 0) {
     freenect_shutdown(fn_ctx_);
     RCLCPP_ERROR(get_logger(), "FREENECT - ERROR SET DEPTH");
@@ -140,6 +144,7 @@ void KinectRosComponent::timer_callback()
   auto stamp = now();
   header.stamp = stamp;
   depth_info_.header.stamp = stamp;
+  rgb_info_.header.stamp = stamp;
 
   if (_depth_flag) {
     //convert 16bit to 8bit mono
@@ -155,7 +160,7 @@ void KinectRosComponent::timer_callback()
   }
 
   if (_rgb_flag) {
-    auto msg = cv_bridge::CvImage(std_msgs::msg::Header(), "rgb8", _rgb_image).toImageMsg();
+    auto msg = cv_bridge::CvImage(header, "rgb8", _rgb_image).toImageMsg();
     rgb_pub_.publish(*msg, rgb_info_);
 
     // cv::imshow("RGB", _rgb_image);
```
Note that as I wanted to use `libfreenect-dev` already installed in system (my kinect revision is 1414, works with freenect binaries), I copied this [FindFreenect.cmake](https://github.com/introlab/rtabmap/blob/master/cmake_modules/FindFreenect.cmake) directly in `kinect_ros2` directory (so it is why I modified the CMakeLists.txt).

Steps:
```bash
ros2 run kinect_ros2 kinect_ros2_node

ros2 launch rtabmap_ros rtabmap.launch.py \
   visual_odometry:=true \
   frame_id:=base_link \
   approx_sync:=true \
   rgb_topic:=/image_raw \
   camera_info_topic:=/camera_info \
   depth_topic:=/depth/image_raw \
   frame_id:=kinect_depth
```
It seems working correctly for the small test I did. However there were a lot of:
`
[rgbd_odometry-1] [WARN] [1669000724.217921035] [rtabmap.rgbd_odometry]: The time difference between rgb and depth frames is high (diff=0.014151s, rgb=1669000724.176265s, depth=1669000724.190416s). You may want to set approx_sync_max_interval lower than 0.01s to reject spurious bad synchronizations or use approx_sync=false if streams have all the exact same timestamp.
`
so [I increased a little](https://github.com/introlab/rtabmap_ros/commit/ed248b4dbd7d6a98f8bbc4140a0ec68185bd8be5) this threshold to avoid spamming warnings (same issue on ros1, also updated).

AddyPete, 2022-11-21:
Thank you for this. Really appreciate it. Will try this in a few minutes

AddyPete, 2022-11-21:
> I tried the package, had to make those changes to get the right `stamp` and `frame_id` in rgb topic header and to get registered depth image:
> 
> ```diff
> diff --git a/CMakeLists.txt b/CMakeLists.txt
> index 296ed9d..f6112bf 100644
> --- a/CMakeLists.txt
> +++ b/CMakeLists.txt
> @@ -5,15 +5,18 @@ if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
>    add_compile_options(-Wall -Wextra -Wpedantic)
>  endif()
>  
> +set(CMAKE_MODULE_PATH ${CMAKE_MODULE_PATH} ${CMAKE_CURRENT_SOURCE_DIR})
> +
>  include_directories(include)
>  
> +
>  # find dependencies
>  find_package(ament_cmake REQUIRED)
>  find_package(rclcpp_components REQUIRED)
>  find_package(rclcpp REQUIRED)
>  # find_package(Threads REQUIRED)
>  
> -find_package(libfreenect REQUIRED)
> +find_package(Freenect REQUIRED)
>  find_package(OpenCV REQUIRED)
>  
>  find_package(sensor_msgs REQUIRED)
> @@ -34,8 +37,9 @@ ament_target_dependencies(kinect_ros2_component
>    camera_info_manager
>    image_transport
>    OpenCV)
> +include_directories(${Freenect_INCLUDE_DIRS})
>  target_link_libraries(kinect_ros2_component
> -  freenect)
> +  ${Freenect_LIBRARY})
>  rclcpp_components_register_nodes(kinect_ros2_component "kinect_ros2::KinectRosComponent")
>  
>  add_executable(kinect_ros2_node
> diff --git a/include/kinect_ros2/kinect_ros2_component.hpp b/include/kinect_ros2/kinect_ros2_component.hpp
> index 3a9f234..73b8e9e 100644
> --- a/include/kinect_ros2/kinect_ros2_component.hpp
> +++ b/include/kinect_ros2/kinect_ros2_component.hpp
> @@ -3,7 +3,7 @@
>  
>  extern "C"
>  {
> -  #include "libfreenect/libfreenect.h"
> +  #include "libfreenect.h"
>  }
>  #include "rclcpp/rclcpp.hpp"
>  #include "camera_info_manager/camera_info_manager.hpp"
> diff --git a/src/kinect_ros2_component.cpp b/src/kinect_ros2_component.cpp
> index 8dca9c6..8322605 100644
> --- a/src/kinect_ros2_component.cpp
> +++ b/src/kinect_ros2_component.cpp
> @@ -1,4 +1,5 @@
>  #include "kinect_ros2/kinect_ros2_component.hpp"
> +#include <ament_index_cpp/get_package_share_directory.hpp>
>  
>  using namespace std::chrono_literals;
>  
> @@ -18,14 +19,17 @@ KinectRosComponent::KinectRosComponent(const rclcpp::NodeOptions & options)
>  {
>    timer_ = create_wall_timer(1ms, std::bind(&KinectRosComponent::timer_callback, this));
>  
> +  // may throw ament_index_cpp::PackageNotFoundError exception
> +  std::string package_share_directory = ament_index_cpp::get_package_share_directory("kinect_ros2");
> +
>    //todo: use parameters
>    depth_info_manager_ = std::make_shared<camera_info_manager::CameraInfoManager>(
>      this, "kinect",
> -    "file:///home/lio/tcc_ws/install/kinect_ros2/share/kinect_ros2/cfg/calibration_depth.yaml");
> +    "file://" + package_share_directory + "/cfg/calibration_depth.yaml");
>  
>    rgb_info_manager_ = std::make_shared<camera_info_manager::CameraInfoManager>(
>      this, "kinect",
> -    "file:///home/lio/tcc_ws/install/kinect_ros2/share/kinect_ros2/cfg/calibration_rgb.yaml");
> +    "file://" + package_share_directory + "/cfg/calibration_rgb.yaml");
>  
>    rgb_info_ = rgb_info_manager_->getCameraInfo();
>    rgb_info_.header.frame_id = "kinect_rgb";
> @@ -64,7 +68,7 @@ KinectRosComponent::KinectRosComponent(const rclcpp::NodeOptions & options)
>      freenect_set_depth_mode(
>      fn_dev_, freenect_find_depth_mode(
>        FREENECT_RESOLUTION_MEDIUM,
> -      FREENECT_DEPTH_MM));
> +      FREENECT_DEPTH_REGISTERED));
>    if (ret < 0) {
>      freenect_shutdown(fn_ctx_);
>      RCLCPP_ERROR(get_logger(), "FREENECT - ERROR SET DEPTH");
> @@ -140,6 +144,7 @@ void KinectRosComponent::timer_callback()
>    auto stamp = now();
>    header.stamp = stamp;
>    depth_info_.header.stamp = stamp;
> +  rgb_info_.header.stamp = stamp;
>  
>    if (_depth_flag) {
>      //convert 16bit to 8bit mono
> @@ -155,7 +160,7 @@ void KinectRosComponent::timer_callback()
>    }
>  
>    if (_rgb_flag) {
> -    auto msg = cv_bridge::CvImage(std_msgs::msg::Header(), "rgb8", _rgb_image).toImageMsg();
> +    auto msg = cv_bridge::CvImage(header, "rgb8", _rgb_image).toImageMsg();
>      rgb_pub_.publish(*msg, rgb_info_);
>  
>      // cv::imshow("RGB", _rgb_image);
> ```
> 
> Note that as I wanted to use `libfreenect-dev` already installed in system (my kinect revision is 1414, works with freenect binaries), I copied this [FindFreenect.cmake](https://github.com/introlab/rtabmap/blob/master/cmake_modules/FindFreenect.cmake) directly in `kinect_ros2` directory (so it is why I modified the CMakeLists.txt).
> 
> Steps:
> 
> ```shell
> ros2 run kinect_ros2 kinect_ros2_node
> 
> ros2 launch rtabmap_ros rtabmap.launch.py \
>    visual_odometry:=true \
>    frame_id:=base_link \
>    approx_sync:=true \
>    rgb_topic:=/image_raw \
>    camera_info_topic:=/camera_info \
>    depth_topic:=/depth/image_raw \
>    frame_id:=kinect_depth
> ```
> 
> It seems working correctly for the small test I did. However there were a lot of: `[rgbd_odometry-1] [WARN] [1669000724.217921035] [rtabmap.rgbd_odometry]: The time difference between rgb and depth frames is high (diff=0.014151s, rgb=1669000724.176265s, depth=1669000724.190416s). You may want to set approx_sync_max_interval lower than 0.01s to reject spurious bad synchronizations or use approx_sync=false if streams have all the exact same timestamp.` so [I increased a little](https://github.com/introlab/rtabmap_ros/commit/ed248b4dbd7d6a98f8bbc4140a0ec68185bd8be5) this threshold to avoid spamming warnings (same issue on ros1, also updated).

Hi there! just tried and it works like charm! Thank you for this, I really appreciate it.

Also, how to make the octomap mapping (/rtabmap/octomap_occupied_space) faster? I visualized it with rviz but want to make it faster.

AddyPete, 2022-11-21:
Also, the map generated in 3D Map on Rtabmap and Rviz is flipped, how to fix this? Thank you.

alvaropascuaI, 2022-11-21:
Hi @AddyPete, i would like to launch rtabmap with stereo cameras realsense, but i have a file for launch with 1 camera but i tried to launch with file rtabmap.launch.py, but dont update rviz, so with stereo cameras i cant see nothing. How can you play rtabmap with stereo cameras in ros2 ?

Thank you

AddyPete, 2022-11-22:
> Hi @AddyPete, i would like to launch rtabmap with stereo cameras realsense, but i have a file for launch with 1 camera but i tried to launch with file rtabmap.launch.py, but dont update rviz, so [...]

Hi unfortunately I didn't try stereo cameras yet. I will try it soon tho

matlabbe, 2022-11-27:
> Also, the map generated in 3D Map on Rtabmap and Rviz is flipped, how to fix this? Thank you.

Do you mean mirrored, upside down or on the side? I may have forgotten to put the static_transform_publisher to to the optical rotation between `base_link` and `kinect`, which would look like (didn't test):
```
ros2 run tf2_ros static_transform_publisher 0 0 0 -1.57 0 -1.57 base_link kinect
```

AddyPete, 2022-11-27:
> Do you mean mirrored, upside down or on the side? I may have forgotten to put the static_transform_publisher to to the optical rotation between `base_link` and `kinect`, which would look like [...]

Hello! I was able to fix this yesterday with TF rotation, just like what you suggested. Thank you very much for your help

closed 2022-11-27 by matlabbe