introlab/rtabmap_ros #844: Error on installation

pfcouto, 2022-11-19:
Hello, I am using a VM with Ubuntu 20.04, I have ROS2 Galactic installed.

I was trying to install your program, however I have some error on the installation process, namely, during the building stage.
Running the command `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release` after 8 minutes gave me the following errors. Can you help me out?

[image]
[image]

matlabbe, 2022-11-19:
You ran out of RAM, or compilation issue. This can be fixed by using less compilation threads at the same time, or just relaunching the compilation. From the [installation instructions](https://github.com/introlab/rtabmap_ros/tree/ros2#installation):
```
export MAKEFLAGS="-j6" # Can be ignored if you have a lot of RAM (>16GB)
```

pfcouto, 2022-11-19:
OK, I will try. Can you ansewr my other question? Will the ROS solution work with a realsense D435i? Will it make a SLAM with that camera? Thanks!

matlabbe, 2022-11-19:
There is a D400 example here: https://github.com/introlab/rtabmap_ros/blob/ros2/launch/ros2/realsense_d400.launch.py

To include IMU, you would have to compare with ROS1 example http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping and add required parameters/nodes.

pfcouto, 2022-11-19:
Even using that command it did not work as you can see. This VM has 8GB of RAM, I believe that RAM should not be the issue.

[image]