introlab/rtabmap_ros #845: depth camera and laser grid map

davidlmljll, 2022-11-22:
Hi
When using the depth camera and 2D laser to build a 2D grid map, some of the environmental information detected by the depth camera does not appear on the final grid map(the camera is mounted at a lower height than the laser)
depth  camera grid map：
 
[image]
laser grid map：
[image]
laser + depth camera map：
[image]

matlabbe, 2022-11-27:
Can I assume you set `Grid/Sensor=2`? If `Grid/3D` is false, the lidar may ray trace the cells from the camera. To help more, I would need the database.

davidlmljll, 2022-11-29:
Thank you very much for your reply! I re-checked the parameter settings and found that it was because I did not enable octomap library support. By default, General\Octomap = false. When I changed General\Octomap = true, 3D ray trace seemed to be correct and 2D grid maps are fine in the rtabmap interface, but with rtabMap-databaseViewer some of the depth image occupancy grids will disappear.
rtabmap interface:
[image]
rtabmap-databaseViewer interface:
[image]

matlabbe, 2022-11-30:
You can inspect local occupancy grid for each node in DatabaseViewer with 3D View, then click on Grid checkbox.

davidlmljll, 2022-11-30:
Thank you very much for your reply, I see the result I want in the rtabmap-databaseViewer interface!

matlabbe, 2022-12-13:
If you are loading the map in localization mode, the map in rviz is the optimized occupancy grid. You can compare the optimized one and the one created in GraphView in rtabmap-databaseViewer. For the optimized one, do File->Edit optimized 2d map.... Are both maps not identical?

matlabbe, 2022-12-16:
You would have to open the 3D View in database viewer, click on Grid checkbox and debug your local occupancy grids to see if the obstacle was really detected as obstacle, and if next nodes are ray tracing over it. You can also open Occupancy grid view. Play with number keys 1->5 on the keyboard in those views to change the colors to better see the obstacles.