introlab/rtabmap_ros #847: Unexpected result in mapping trajectory -> Trajectory goes towards right when vehicle goes straight

takahiko-hasegawa-hdjp, 2022-11-22:
We are using RTABMAP-ros with our own rosbag data and seems like the output trajectory goes towards right when the camera is going  straight. 

**Basic setup** 
- Camera: Zed2i (rgb and depth image)
- Zed2 sdk version: 3.7 
- Image size: 1280x720 
- Frequency: ~10Hz
- Image encoding: raw bgr8 (Original rosbag data included 8UC3 so modified the code in ros wrapper to do the conversion to bgr8 using cvBridge)
- ros topic used at this time: rgb image, depth image, sesor tf, camera_info 

**High level detail of experiment that we tried**

- Created our own launch file based on demo_robot_mapping.launch file(no significant changes but edited only topic remap section and added detection algorithm param).
- Tried changing DetectorStrategy and MaxFeature but not significant changes to the trajectory output.
- We are sure that using correct intrinsic and extrinsic parameters for the camera. We also tried changing the extrinsic params to some random values just to test the output but no apparent change to the output.

Are there any specific parameters that need to be modified/added in order to mitigate this problem?

Thanks in advance!

matlabbe, 2022-11-27:
Is odometry computed from ZED? or rtabmap's visual odometry node? Maybe an optical rotation is missing, what is your TF tree?

takahiko-hasegawa-hdjp, 2022-12-18:
Thanks for your input, i found out that it was using external odometry data to do the trajectory estimation. Since out odometry is not that great, the map created by using odometry was also bad. Thanks!

closed 2022-12-18 by takahiko-hasegawa-hdjp