introlab/rtabmap_ros #848: Still waiting for data on topics `/imu/data_raw` and `/imu/mag`

tsadarsh, 2022-11-27:
I am running ROS2 Humble on my Pop!_OS 22.04 LTS. I have installed `rtabmap_ros` as per the instruction [given](https://github.com/introlab/rtabmap_ros/tree/ros2#installation) in `ros2` branch. I am following the "[RGB-D Handheld Mapping](http://wiki.ros.org/rtabmap_ros/Tutorials/HandHeldMapping)" tutorial with an IntelRealsense D435i stereo camera. As the tutorial lists out the launch commands for ROS1, I am listing the commands that I used for running the nodes in ROS2.

To start the camera:
```
ros2 launch realsense2_camera rs_launch.py enable_accel:=true enable_gyro:=true unite_imu_method:=2 align_depth.enable:=true
```

To start `imu_filter_madgwick`: 
```
ros2 run imu_filter_madgwick imu_filter_madgwick_node --ros-args -p _use_mag:=false -p _publish_tf:=false -p _world_frame:="enu" -r /imu/data_raw:=/camera/imu -r /imu/data:=/rtabmap/imu
```
The above command gives the following output:
```
[INFO] [1669529567.519634979] [imu_filter_madgwick]: Starting ImuFilter
[INFO] [1669529567.519965058] [imu_filter_madgwick]: Using dt computed from message headers
[INFO] [1669529567.519979864] [imu_filter_madgwick]: The gravity vector is kept in the IMU message.
[INFO] [1669529567.520065282] [imu_filter_madgwick]: Imu filter gain set to 0.100000
[INFO] [1669529567.520080927] [imu_filter_madgwick]: Gyro drift bias set to 0.000000
[INFO] [1669529567.520092382] [imu_filter_madgwick]: Magnetometer bias values: 0.000000 0.000000 0.000000
[WARN] [1669529577.523446617] [imu_filter_madgwick]: Still waiting for data on topics /imu/data_raw and /imu/mag...
[WARN] [1669529587.523560413] [imu_filter_madgwick]: Still waiting for data on topics /imu/data_raw and /imu/mag...
```
The WARN message keeps repeating. Why does `/imu/mag` expects data even though I set `_use_mag:=false`? Is my `ros2 run` command wrong?

matlabbe, 2022-11-28:
Hi, 

just made two new example launch files for D435i. There is a color+depth and a infra+depth versions:
- https://github.com/introlab/rtabmap_ros/blob/ros2/launch/ros2/realsense_d435i_color.launch.py
    - For color+depth, because of this issue (https://github.com/IntelRealSense/realsense-ros/issues/2564), we have to regenerate the aligned depth on our side, it is why the launch is more complicated.
- https://github.com/introlab/rtabmap_ros/blob/ros2/launch/ros2/realsense_d435i_infra.launch.py
    - (Best odom quality/accuracy) Follow top instructions, don't forget to disable IR emitter.

tsadarsh, 2022-11-28:
Thank you @matlabbe for providing the example launch files. It works now :smile: 

This issue can be closed.

closed 2022-11-28 by matlabbe

Benjamin-86, 2024-07-29:
Hey guys, i am having the same issue as @tsadarsh. We use humble on ubuntu 22.04. The fun part is we had it working for a while using the launch file mentioned by @matlabbe, but out of no where this issue `[imu_filter_madgwick_node-5] [WARN] [1722261630.037924833] [imu_filter_madgwick]: Still waiting for data on topic imu/data_raw...
[stereo_odometry-3] [WARN] [1722261632.375946187] [stereo_odometry]: stereo_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
[stereo_odometry-3] stereo_odometry subscribed to (exact sync):
` merged. We have tried to update the whole system but nothing works. Hope you guys can help. Thanks in advanced.

matlabbe, 2024-08-01:
Do `rostopic hz` on the topics shown in the warning. Are they still published?

MeltedHyperion07, 2024-08-30:
I am getting the same issue for Depth-ai OAK-D RGB-D camera. I am using ros2 humble. Cna you tell me what i should do?
[image]

@matlabbe can you help me out

matlabbe, 2024-09-05:
Looks like the raw IMU is not published. You may check the input topic if it is published.

I added recently a example with depthai: https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/depthai.launch.py, you may try it to compare