introlab/rtabmap_ros #850: The sample  launch/ros2/turtlebot3_rgbd.launch.py outs put some queer navigation2 logs.

tosa-no-onchan, 2022-11-28:
I am trying to test  launch/ros2/turtlebot3_rgbd.launch.py,
but sometime navigation2 log outs put the following,
-------
 [planner_server-2] [INFO] [1669623367.896801572] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_scan' at time 267.814 for reason 'Unknown'
------
PC Ubuntu Mate 20.04
ROS2 Foxy
Gazebo Turtlebot3 House

How to test.
1) Edit turtlebot3_gazebo/models/turtlebot3_waffle/model.sdf according to turtlebot3_rgbd.launch.py.
2) ros2 launch turtlebot3_gazebo turtlebot3_house.launch.py
3) ros2 launch rtabmap_ros turtlebot3_rgbd.launch.py
4) ros2 launch nav2_bringup navigation_launch.py use_sim_time:=True
5) ros2 launch nav2_bringup rviz_launch.py
6) on Rviz2, set Controller off, drive waffle by "Navigation2 Goal" icon.
 Then, navigation2 out put next INFO messages.
-----
[planner_server-2] [INFO] [1669625122.612764386] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_scan' at time 306.616 for reason 'Unknown'
[planner_server-2] [INFO] [1669625122.813546408] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_scan' at time 306.816 for reason 'Unknown'
....

 I checked Rviz2 TF -> Frames -> base_footprint and base_scan.
Both values stop somtime for a while at a same time, and  begin to update its value again.

At the same time, navigation2 out above INFO logs.
I doubt,  rtabmap_ros doesn't publish tf-odom->base_footprint topic in a time.  

Are there any resolve way?

matlabbe, 2022-11-30:
If this doesn't happen often, maybe it is just simulation time issue with the stamp in the base scan topic. I do have sometimes this:
`
[planner_server-2] [INFO] [1669775197.377534105] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_scan' at time 467.631 for reason 'the timestamp on the message is earlier than all the data in the transform cache'
`
But the robot seems able to navigate.

tosa-no-onchan, 2022-11-30:
Thank you for your relpy.

I see.
I feel, it is on Gazebo side, tf odom -> base_footprint publish are missed some time, too.
Then, I am going to check a mismatch among OpenGL and graphic driver.

http://wiki.ros.org/rviz/Troubleshooting
https://answers.ros.org/question/11148/rviz-in-ros-electric/?answer=3627%3Fanswer%3D3627#3627